OctoPrint/OctoPrint-FirmwareUpdater #261: Flashing Bin and Assets folder

LucaTomei, 2021-07-10:
**Hardware Setup**
FlyingBear Ghost 5

**Describe the problem**
I have a FlyingBear Ghost 5 3D printer and I have two problem:
1) Which flashing method I have to choose?
2) Can I flash binary file and its related assets folder together ([see firmware here](https://github.com/flyingbear-club-ita/mks-robin-nano-1.x/tree/master/ghost_5_tmc2209))?

benlye, 2021-07-10:
1. No idea, sorry. What type of MCU (processor) did it have?
2. No.

LucaTomei, 2021-07-10:
Where can I see the MCU type of my 3D printer?

benlye, 2021-07-10:
Post a hi res picture of the board and I'll try to tell you.

LucaTomei, 2021-07-10:
This is the board
[image]

benlye, 2021-07-10:
[image]

Need a close up of the component highlighted in the red box.

It looks like some kind of STM32 and Google says it's probably an STM32F103, which means it *might* work with dfu-util, but it will really depend on whether or not they have put a bootloader on it as I don't think that the STM32F103 has a native DFU mode like other STM32 MCUs do.

Your best bet is probably to find a user forum for the board / printer and ask there if anyone has figured it out.

closed 2021-07-29 by github-actions[bot]