OctoPrint/OctoPrint-FirmwareUpdater #262: Connection closed before reset command is sent.

ManuelMcLure, 2021-07-12:
**Hardware Setup**
Re-ARM/RAMPS (LPC1768)

**Describe the problem**
If either or both of "Don't wait for reset" or "Don't wait for restart" are set, Firmware Updater closes the connection to the printer before the reset command is set so the reset doesn't occur.

**Log Files**
[serial.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/6798390/serial.log)
[octoprint.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/6798391/octoprint.log)
[plugin_firmwareupdater_console.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/6798392/plugin_firmwareupdater_console.log)

**Additional context**
I use those options because resetting my board turns off the `PS_ON` pin so the board will not power back up after the `M997` command is sent. If I disable those settings then the reset occurs as expected, but then I get error messages because the board doesn't come back. This is not a high priority issue, though - I can just walk over to my printer (which I need to do anyway to boot it back up) and reset it manually with the kill button.

benlye, 2021-07-12:
The printer is always disconnected before the flash starts, and never reconnected until after it resets.  The `M997` command to reset the printer is sent out-of-band using a direct command on the serial port, so it doesn't matter what state the printer connection is in.

I looked at your logs and the first flash attempt in octoprint.log looks fine (2021-07-11 20:17:56) - you had 'Don't wait for reset' enabled so the plugin didn't wait for the reset to occur and reported flashing successful.  What was the problem you observed during that flash?

ManuelMcLure, 2021-07-12:
It appears that the printer never actually received the `M997` because it did not power down. My printer set up is somewhat uncommon - I use `PS_ON` to control the power supply, but if the printer is reset or there is an error it will _not_ reboot but rather stay unpowered until I press a physical button on the printer itself to "bootstrap" the printer.

Upon further testing, it appears that the printer is never getting the `M997` from Firmware Updater and is not resetting as expected. This is even the case if I do not select any of the options listed. However, an `M997` from the OctoPrint terminal does work just fine.

I thought that this might be interference from the PortLister plugin but it appears to happen even with that plugin disabled. I also tested with all non-bundled plugins disabled except for Firmware Updater and still don't see a reset happening.

As I mentioned, since I need to go to the printer anyway to reboot it by pressing a physical button, it not a huge deal for me, but there's definitely some weirdness going on here.

One thing I didn't mention is that I am not on OctoPi - I use a Celeron J1900 mini PC running Debian. Please let me know if there's any more debugging I can do to help you fix this.

benlye, 2021-07-12:
These are the commands that the plugin sends to reset the board - what happens if you issue them in a shell on your host?
```
stty -F /dev/ttyS0 speed 115200 -echo > /dev/null
echo M997 >> /dev/ttyS0 
```

The baud rate is hard-coded to 115200, so maybe try changing that to 250000?  If that fixes it I'll have to update the plugin to use the actual baud rate (or make it configurable).

ManuelMcLure, 2021-07-12:
Never mind - it was PEBKAC (or maybe an ID10T error). I had the wrong serial port selected because I thought "LPC doesn't use the serial port". Once I changed it to `ttyACM0` it magically started working. Sorry for the trouble.
Perhaps there should be an "Auto" setting that would default to the port OctoPrint is using to connect to the printer?

closed 2021-07-12 by ManuelMcLure