OctoPrint/OctoPrint-FirmwareUpdater #265: Firmware update on Ender 5 Plus

NKeyEdge, 2021-07-16:
**Hardware Setup**
Please board type and printer make and model.

**Describe the problem**
Please provide a clear and concise description of what the problem is, including details of any error messages you receive (screenshots are helpful).

**Log Files**
Attach the OctoPrint log and the Firmware Updater log Log files, which can be accessed under **Settings** -> **Logging** in OctoPrint, or in `~/.octoprint/logs/` on the host.  Both `octoprint.log` and `~/.octoprint/logs/plugin_firmwareupdater_console.log` should be included.

**Additional context**
Add any other context about the problem here.

benlye, 2021-07-19:
Not much I can do without log file.  You can always disable verification (but I wouldn't recommend it).

NKeyEdge, 2021-07-20:
Okay I think I got something now. I'm guessing this is what you need but not fully sure as it does not script a update in process.

Connection closed, closing down monitor
Changing monitoring state from "Operational" to "Offline"
Changing monitoring state from "Offline" to "Opening serial connection"
Connecting to port /dev/ttyUSB0, baudrate 115200
Changing monitoring state from "Opening serial connection" to "Connecting"
Connected to: Serial<id=0xa3aa0cb0, open=True>(port='/dev/ttyUSB0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Send: N0 M110 N0*125
Send: N0 M110 N0*125
Recv: start
Recv: echo: External Reset
Send: N0 M110 N0*125
Recv: Marlin Ver 1.70.2 BL
Recv: 
Recv: echo: Last Updated: 2019-09-04 | Author: Ender-5 Plus
Recv: echo:Compiled: Jan 17 2020
Recv: echo: Free Memory: 1460  PlannerBufferBytes: 1232
Recv: echo:V41 stored settings retrieved (526 bytes; crc 59647)
Recv: echo:  G21    ; Units in mm
Recv: echo:  M149 C ; Units in Celsius
Recv:

NKeyEdge, 2021-07-20:
I have also tried updating with verification disabled, but the update did not go through

benlye, 2021-07-20:
No, I need to see the actual log files - `octoprint.log` and `plugin_firmwareupdater_console.log`

You can get them from Settings -> Logging -> Logs.

https://github.com/OctoPrint/OctoPrint-FirmwareUpdater#troubleshooting

NKeyEdge, 2021-07-20:
0 16:25:26,373 /usr/bin/avrdude -v -q -p m2560 -c wiring -P /dev/ttyUSB0 -D -U flash:w:/tmp/tmp6zsw0j4q:i
2021-07-20 16:25:27,509 > 
2021-07-20 16:25:27,510 > avrdude: Version 6.3-20171130
2021-07-20 16:25:27,511 >          Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
2021-07-20 16:25:27,511 >          Copyright (c) 2007-2014 Joerg Wunsch
2021-07-20 16:25:27,512 > 
2021-07-20 16:25:27,513 >          System wide configuration file is "/etc/avrdude.conf"
2021-07-20 16:25:27,513 >          User configuration file is "/home/pi/.avrduderc"
2021-07-20 16:25:27,514 >          User configuration file does not exist or is not a regular file, skipping
2021-07-20 16:25:27,514 > 
2021-07-20 16:25:27,515 >          Using Port                    : /dev/ttyUSB0
2021-07-20 16:25:27,516 >          Using Programmer              : wiring
2021-07-20 16:25:27,516 >          AVR Part                      : ATmega2560
2021-07-20 16:25:27,517 >          Chip Erase delay              : 9000 us
2021-07-20 16:25:27,517 >          PAGEL                         : PD7
2021-07-20 16:25:27,518 >          BS2                           : PA0
2021-07-20 16:25:27,519 >          RESET disposition             : dedicated
2021-07-20 16:25:27,519 >          RETRY pulse                   : SCK
2021-07-20 16:25:27,520 >          serial program mode           : yes
2021-07-20 16:25:27,520 >          parallel program mode         : yes
2021-07-20 16:25:27,521 >          Timeout                       : 200
2021-07-20 16:25:27,522 >          StabDelay                     : 100
2021-07-20 16:25:27,522 >          CmdexeDelay                   : 25
2021-07-20 16:25:27,523 >          SyncLoops                     : 32
2021-07-20 16:25:27,523 >          ByteDelay                     : 0
2021-07-20 16:25:27,524 >          PollIndex                     : 3
2021-07-20 16:25:27,524 >          PollValue                     : 0x53
2021-07-20 16:25:27,525 >          Memory Detail                 :
2021-07-20 16:25:27,526 > 
2021-07-20 16:25:27,526 >                                   Block Poll               Page                       Polled
2021-07-20 16:25:27,527 >            Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
2021-07-20 16:25:27,527 >            ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
2021-07-20 16:25:27,528 >            eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
2021-07-20 16:25:27,529 >            flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
2021-07-20 16:25:27,529 >            lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
2021-07-20 16:25:27,530 >            hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
2021-07-20 16:25:27,530 >            efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
2021-07-20 16:25:27,531 >            lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
2021-07-20 16:25:27,532 >            calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
2021-07-20 16:25:27,532 >            signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
2021-07-20 16:25:27,533 > 
2021-07-20 16:25:27,533 >          Programmer Type : Wiring
2021-07-20 16:25:27,534 >          Description     : Wiring
2021-07-20 16:25:27,535 >          Programmer Model: AVRISP
2021-07-20 16:25:27,535 >          Hardware Version: 15
2021-07-20 16:25:27,536 >          Firmware Version Master : 2.10
2021-07-20 16:25:27,536 >          Vtarget         : 0.0 V
2021-07-20 16:25:27,537 >          SCK period      : 0.1 us
2021-07-20 16:25:27,540 > 
2021-07-20 16:25:27,541 > avrdude: AVR device initialized and ready to accept instructions
2021-07-20 16:25:27,541 > avrdude: Device signature = 0x1e9801 (probably m2560)
2021-07-20 16:25:27,542 > avrdude: reading input file "/tmp/tmp6zsw0j4q"
2021-07-20 16:25:27,543 > avrdude: writing flash (127344 bytes):
2021-07-20 16:25:27,543 > 
2021-07-20 16:25:51,447 > avrdude: 127344 bytes of flash written
2021-07-20 16:25:51,447 > avrdude: verifying flash memory against /tmp/tmp6zsw0j4q:
2021-07-20 16:25:51,448 > avrdude: load data flash data from input file /tmp/tmp6zsw0j4q:
2021-07-20 16:25:51,449 > avrdude: input file /tmp/tmp6zsw0j4q contains 127344 bytes
2021-07-20 16:25:51,449 > avrdude: reading on-chip flash data:
2021-07-20 16:25:51,450 > 
2021-07-20 16:26:18,851 > avrdude: verifying ...
2021-07-20 16:26:18,852 > avrdude: verification error, first mismatch at byte 0x0000
2021-07-20 16:26:18,852 >          0x14 != 0x0c
2021-07-20 16:26:18,853 > avrdude: verification error; content mismatch
2021-07-20 16:26:18,854 > 
2021-07-20 16:26:18,854 > avrdude done.  Thank you.

benlye, 2021-07-20:
Yup, verification is failing.  Don't know what to tell you - I don't know of any reason why that would happen, but I would suspect that maybe your board/chip is bad.  Not really anything that I can do about that.  

I don't know why you would be having problems with verification disabled though - can you post a log from an attempt with the Disable Verification option enabled?

NKeyEdge, 2021-07-20:
Ya, it says it was successful but the printer is still saying it has the stock firmware that was originally on it.

0 16:32:50,722 /usr/bin/avrdude -v -q -p m2560 -c wiring -P /dev/ttyUSB0 -D -V -U flash:w:/tmp/tmpt7mdn5zt:i
2021-07-20 16:32:51,794 > 
2021-07-20 16:32:51,795 > avrdude: Version 6.3-20171130
2021-07-20 16:32:51,796 >          Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
2021-07-20 16:32:51,797 >          Copyright (c) 2007-2014 Joerg Wunsch
2021-07-20 16:32:51,797 > 
2021-07-20 16:32:51,798 >          System wide configuration file is "/etc/avrdude.conf"
2021-07-20 16:32:51,798 >          User configuration file is "/home/pi/.avrduderc"
2021-07-20 16:32:51,799 >          User configuration file does not exist or is not a regular file, skipping
2021-07-20 16:32:51,800 > 
2021-07-20 16:32:51,800 >          Using Port                    : /dev/ttyUSB0
2021-07-20 16:32:51,801 >          Using Programmer              : wiring
2021-07-20 16:32:51,801 >          AVR Part                      : ATmega2560
2021-07-20 16:32:51,802 >          Chip Erase delay              : 9000 us
2021-07-20 16:32:51,803 >          PAGEL                         : PD7
2021-07-20 16:32:51,803 >          BS2                           : PA0
2021-07-20 16:32:51,804 >          RESET disposition             : dedicated
2021-07-20 16:32:51,804 >          RETRY pulse                   : SCK
2021-07-20 16:32:51,805 >          serial program mode           : yes
2021-07-20 16:32:51,806 >          parallel program mode         : yes
2021-07-20 16:32:51,806 >          Timeout                       : 200
2021-07-20 16:32:51,808 >          StabDelay                     : 100
2021-07-20 16:32:51,810 >          CmdexeDelay                   : 25
2021-07-20 16:32:51,811 >          SyncLoops                     : 32
2021-07-20 16:32:51,811 >          ByteDelay                     : 0
2021-07-20 16:32:51,812 >          PollIndex                     : 3
2021-07-20 16:32:51,813 >          PollValue                     : 0x53
2021-07-20 16:32:51,813 >          Memory Detail                 :
2021-07-20 16:32:51,814 > 
2021-07-20 16:32:51,814 >                                   Block Poll               Page                       Polled
2021-07-20 16:32:51,815 >            Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
2021-07-20 16:32:51,816 >            ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
2021-07-20 16:32:51,816 >            eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
2021-07-20 16:32:51,817 >            flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
2021-07-20 16:32:51,817 >            lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
2021-07-20 16:32:51,818 >            hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
2021-07-20 16:32:51,819 >            efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
2021-07-20 16:32:51,820 >            lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
2021-07-20 16:32:51,820 >            calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
2021-07-20 16:32:51,821 >            signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
2021-07-20 16:32:51,821 > 
2021-07-20 16:32:51,822 >          Programmer Type : Wiring
2021-07-20 16:32:51,822 >          Description     : Wiring
2021-07-20 16:32:51,823 >          Programmer Model: AVRISP
2021-07-20 16:32:51,824 >          Hardware Version: 15
2021-07-20 16:32:51,825 >          Firmware Version Master : 2.10
2021-07-20 16:32:51,825 >          Vtarget         : 0.0 V
2021-07-20 16:32:51,826 >          SCK period      : 0.1 us
2021-07-20 16:32:51,826 > 
2021-07-20 16:32:51,827 > avrdude: AVR device initialized and ready to accept instructions
2021-07-20 16:32:51,828 > avrdude: Device signature = 0x1e9801 (probably m2560)
2021-07-20 16:32:51,828 > avrdude: reading input file "/tmp/tmpt7mdn5zt"
2021-07-20 16:32:51,829 > avrdude: writing flash (127344 bytes):
2021-07-20 16:32:51,829 > 
2021-07-20 16:33:15,720 > avrdude: 127344 bytes of flash written
2021-07-20 16:33:15,721 > 
2021-07-20 16:33:15,722 > avrdude done.  Thank you.
2021-07-20 16:33:15,723 > 
2021-07-20 16:33:15,724 > 
2021-07-20 16:33:15,731 Flashing successful.

benlye, 2021-07-20:
Yeah, I don't think I can help you with it.  The plugin definitely works flashing other ATmega2560 boards (I flash a RAMPS board with it), and your log says that everything worked fine so I can't explain why it would still have the stock firmware on it.

Bottom line - I have no idea what might be special about your board that would be causing the problems you have.  Maybe there's an Ender user group you can ask in?

NKeyEdge, 2021-07-20:
Well I know the ender 5 plus is a pain in the rear to work with, especially with the firmware. Thank you for your help, I'll find a solution.

closed 2021-07-20 by NKeyEdge