OctoPrint/OctoPrint-FirmwareUpdater #268: Close the selected port.

SinisterRj, 2021-08-01:
I'm using the plugin to flash another arduino (not the printer) connected to the raspberry pi and I realized that it disconected the printer (instead of the serial port that I selected to flash) when I hit "flash from file" button. IMO it should disconnect the selected serial port.

benlye, 2021-08-02:
In your scneario the plugin can't know what process has the selected port open in order to ask to kill/close it.  The expectation is that it's the printer OctoPrint is connected to, so that's what gets closed.

You can do this yourself by adding a pre-flash system command to close/kill whatever has the port open.

closed 2021-08-02 by benlye

SinisterRj, 2021-08-04:
In this case the plugin can send a warning to user, asking permission to close the port. Considering that you need a exclusive access to the port to write a firmware, the plugin must close that port before starting flashing no matter if it is the printer or anything else. 
In other hand, everytime that user changes the physical connection, physical port can change its name (i.e. changing from ttyUSB0 to ttyUSB1) and the user will need to change the pre-flash command everytime.
So please, I ask you again to change this.

benlye, 2021-08-04:
You're missing the point. The plugin can't close the port if OctoPrint doesn't have it open. I cannot know which process has it open so cannot know how to close it, even if the user gave permission. What you're asking simply isn't possible.