OctoPrint/OctoPrint-FirmwareUpdater #272: MMU Flashing error

fila612, 2021-10-04:
**Hardware Setup**
Prusa MK3s & MMUS (Atmel ATmega 8-bit MCUs)

**Describe the problem**
start flashing firmware MMU => get an unknown error

**Log Files**
avrdude: Version 6.3-20171130
2021-10-04 11:55:08,579 >          Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
2021-10-04 11:55:08,580 >          Copyright (c) 2007-2014 Joerg Wunsch
2021-10-04 11:55:08,581 > 
2021-10-04 11:55:08,582 >          System wide configuration file is "/etc/avrdude.conf"
2021-10-04 11:55:08,583 >          User configuration file is "/home/pi/.avrduderc"
2021-10-04 11:55:08,583 >          User configuration file does not exist or is not a regular file, skipping
2021-10-04 11:55:08,584 > 
2021-10-04 11:55:08,585 >          Using Port                    : /dev/ttyACM0
2021-10-04 11:55:08,586 >          Using Programmer              : avr109
2021-10-04 11:55:08,586 >          Overriding Baud Rate          : 57600
2021-10-04 11:55:08,587 >          AVR Part                      : ATmega32U4
2021-10-04 11:55:08,588 >          Chip Erase delay              : 9000 us
2021-10-04 11:55:08,589 >          PAGEL                         : PD7
2021-10-04 11:55:08,590 >          BS2                           : PA0
2021-10-04 11:55:08,590 >          RESET disposition             : dedicated
2021-10-04 11:55:08,591 >          RETRY pulse                   : SCK
2021-10-04 11:55:08,592 >          serial program mode           : yes
2021-10-04 11:55:08,593 >          parallel program mode         : yes
2021-10-04 11:55:08,593 >          Timeout                       : 200
2021-10-04 11:55:08,594 >          StabDelay                     : 100
2021-10-04 11:55:08,595 >          CmdexeDelay                   : 25
2021-10-04 11:55:08,596 >          SyncLoops                     : 32
2021-10-04 11:55:08,596 >          ByteDelay                     : 0
2021-10-04 11:55:08,597 >          PollIndex                     : 3
2021-10-04 11:55:08,598 >          PollValue                     : 0x53
2021-10-04 11:55:08,599 >          Memory Detail                 :
2021-10-04 11:55:08,600 > 
2021-10-04 11:55:08,600 >                                   Block Poll               Page                       Polled
2021-10-04 11:55:08,601 >            Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
2021-10-04 11:55:08,608 >            ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
2021-10-04 11:55:08,611 >            eeprom        65    20     4    0 no       1024    4      0  9000  9000 0x00 0x00
2021-10-04 11:55:08,612 >            flash     
[plugin_firmwareupdater_console-2.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/7277226/plugin_firmwareupdater_console-2.log)
    65     6   128    0 yes     32768  128    256  4500  4500 0x00 0x00
2021-10-04 11:55:08,612 >            lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
2021-10-04 11:55:08,613 >            hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
2021-10-04 11:55:08,614 >            efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
2021-10-04 11:55:08,615 >            lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
2021-10-04 11:55:08,615 >            calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
2021-10-04 11:55:08,616 >            signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
2021-10-04 11:55:08,617 > 
2021-10-04 11:55:08,618 >          Programmer Type : butterfly
2021-10-04 11:55:08,619 >          Description     : Atmel AppNote AVR109 Boot Loader
2021-10-04 11:55:08,620 > 
2021-10-04 11:55:08,620 > 
2021-10-04 11:55:13,831 > Connecting to programmer: .avrdude: butterfly_recv(): programmer is not responding
2021-10-04 11:55:13,832 > 
2021-10-04 11:55:13,833 > 
2021-10-04 11:55:18,836 > avrdude: butterfly_recv(): programmer is not responding
2021-10-04 11:55:18,837 > 
2021-10-04 11:55:24,092 > avrdude: butterfly_recv(): programmer is not responding
2021-10-04 11:55:24,093 > 
2021-10-04 11:55:29,097 > avrdude: butterfly_recv(): programmer is not responding
2021-10-04 11:55:29,098 > 
2021-10-04 11:55:34,102 > avrdude: butterfly_recv(): programmer is not responding
2021-10-04 11:55:34,109 > 

**Additional context**
Add any other context about the problem here.

benlye, 2021-10-04:
@Samaelson can you help here?  I don't have this hardware or any experience with flashing an MMU.

Samaelson, 2021-10-04:
@fila612: did you choose the right com port? Guess you have both, MK3s and MMU2s board connected via USB, same as I have. So make sure you' ve chosen the correct port.

fila612, 2021-10-04:
if you mean the ttyACM*-port, then yes, I've tried all ports. and also disconnected all except the emu, so there were also one port to choose.

Samaelson, 2021-10-04:
What firmware are you running on MMU2s Board? What are PID&VID MMU2s board enumerates when connected via USB? Is MMU2s beard detected by Prusa Slic3r?

Samaelson, 2021-10-04:
One more comment: Prusa uses a special initialization sequence to enter bootloader on MMU2s board. I assume no changes have been made to the code part I submitted. So what you also could check is to manually push the reset button on the MMU2s after firmware upload was triggered. Please let me know if this was helpful.

fila612, 2021-10-05:
> What firmware are you running on MMU2s Board? What are PID&VID MMU2s board enumerates when connected via USB? Is MMU2s beard detected by Prusa Slic3r?

currently there is the TZB Rel. 3.2.0 on the MMU2s. The Update from original Prusa to TZB was done via this Plugin. My Problem is, that the printer is located in an IKEA Platsa and therefore ist not so easy for me to connect via USB, cause I've only an iMac :(
if I run "ls -la /dev/serial/by-id" on my raspberry I got following result:
`
pi@octopi:~ $ ls -la /dev/serial/by-id
total 0
drwxr-xr-x 2 root root 80 Oct  5 06:53 .
drwxr-xr-x 4 root root 80 Oct  4 11:41 ..
lrwxrwxrwx 1 root root 13 Oct  5 06:53 usb-Arduino_LLC_Arduino_Leonardo-if00 -> ../../ttyACM0
lrwxrwxrwx 1 root root 13 Oct  4 12:05 usb-Prusa_Research__prusa3d.com__Original_Prusa_i3_MK3____________________-if00 -> ../../ttyACM1
`
=> so, for me the ACM0 is the MMU and the ACM1 is the MK3S

fila612, 2021-10-05:
> One more comment: Prusa uses a special initialization sequence to enter bootloader on MMU2s board. I assume no changes have been made to the code part I submitted. So what you also could check is to [...]

Thanks for that hit, but unfortunately that doesn't helps. also the sequence makes no difference: 1st reset and than start fw-upload or vice versa..

Samaelson, 2021-10-05:
TZB Firmware breaks Prusa's init routine. Tried this in the past on my own MMU2s board. So after flashing TZB firmware also regular Prusa Firmware Updater in Slic3r is was no more working on my side

fila612, 2021-10-05:
hmmm, that sounds no so good. does that means, that the board is now only usable with the installed firmware 3.2.0?

Samaelson, 2021-10-05:
Let me check how to add support for MMU2s running custom TZB firmware. Pid&Vid TZB Firmware enumerates is necessary for that

fila612, 2021-10-05:
okay, how can I help you? where can I check the PID&VID?

Samaelson, 2021-10-05:
@benlye : a hint should be added stating only stock firmware is supported for now for Prusa's MMU2s and CW1

Samaelson, 2021-10-05:
@fila612 : open terminal and simply type lsusb

fila612, 2021-10-05:
pi@octopi:/tmp $ lsusb
Bus 001 Device 006: ID 05a3:9331 ARC International Camera
Bus 001 Device 007: ID 2c99:0002  
Bus 001 Device 005: ID 0424:7800 Standard Microsystems Corp. 
Bus 001 Device 003: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

fila612, 2021-10-05:
pi@octopi:/tmp $ lsusb
Bus 001 Device 021: ID 2341:8036 Arduino SA Leonardo (CDC ACM, HID)
Bus 001 Device 006: ID 05a3:9331 ARC International Camera
Bus 001 Device 007: ID 2c99:0002  
Bus 001 Device 005: ID 0424:7800 Standard Microsystems Corp. 
Bus 001 Device 003: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Samaelson, 2021-10-05:
Yes it is. Will work on support for non stock firmware flash procedure

fila612, 2021-10-05:
I don't no if its necessary, but the update from TZB 3.2.0 to 3.3.0 on my MK3S was working.

Samaelson, 2021-10-05:
Not important als MK3s board uses different firmware update approach.

Samaelson, 2021-10-06:
@fila612 : could you check if MMU2s enumerates correctly after pushing reset button on MMU2s. 2c99:0002 is MK3s board and MMU2s shall be 2c99:0003 when in bootloader

fila612, 2021-10-06:
@Samaelson: seems so:
1st:
pi@octopi:~ $ lsusb
Bus 001 Device 027: ID 2341:8036 Arduino SA Leonardo (CDC ACM, HID)
Bus 001 Device 007: ID 2c99:0002  
Bus 001 Device 005: ID 0424:7800 Standard Microsystems Corp. 
Bus 001 Device 003: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

pushing reset:
pi@octopi:~ $ lsusb
Bus 001 Device 028: ID 2c99:0003  
Bus 001 Device 007: ID 2c99:0002  
Bus 001 Device 005: ID 0424:7800 Standard Microsystems Corp. 
Bus 001 Device 003: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

after a few sec:
pi@octopi:~ $ lsusb
Bus 001 Device 029: ID 2341:8036 Arduino SA Leonardo (CDC ACM, HID)
Bus 001 Device 007: ID 2c99:0002  
Bus 001 Device 005: ID 0424:7800 Standard Microsystems Corp. 
Bus 001 Device 003: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Samaelson, 2021-10-06:
Ok, thanks.

Samaelson, 2021-10-10:
>   okay thanks. for my understanding this will avoid further errors like that, but what do you think can I do to resolve this issue on my MMU? Should I try to update via Laptop/USB directly or can I update the original Prusa-FW via OctoprintFWU? If this cannot be solved via this issue then this can be closed....

In the past I used a Laptop, USB cable and Prusa Slic3r to Flash back to original. You need to push reset button on MMU to enter bootloader. Btw. you could compile TZB firmware including the UART0 init by your own and then use FirmwareUpdater to flash to MMU. Tried this on my own and it works fine.--Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.