OctoPrint/OctoPrint-FirmwareUpdater #282: Flashing Prusa MMU2

BRoadifer, 2021-10-30:
I asked this on Facebook, but no answers as of yet.  I was able to install this plugin and updated my firmware on my MK3S firmware via Octoprint Firmware Updater plugin without issue.  

I'm trying to update my MMU2 firmware via OctoPrint Firmware.  The MMU is connected via USB to the Pi and the updater sees it as /dev/tty/ACM1. The updater fails, but I think that might be due to me not knowing what AVR MCU type to use for the MMU2. I tried 2560, but that fails.  

Is there something else I'm missing?

benlye, 2021-10-30:
@Samaelson is the MMU flashing expert.  I believe you need to have a flashing profile set up with ATmega32u4 as the MCU and avr109 as the programmer type.  After that I don't know.

There are some [old issues](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/issues?q=is%3Aissue+mmu) which might be helpful to read.

Samaelson, 2021-10-30:
Hi,

Profile is as follows:

Flash methode: avrdude (Atmel AVR Family)
AVR MCU: ATmega32u4
AVR Programmer Type: avr109

benlye, 2021-10-30:
Thanks @Samaelson, I'll add that to the documentation.

BRoadifer, 2021-11-02:
Thanks guys.  Worked perfectly. Thanks for the quick response!

closed 2021-11-02 by benlye