OctoPrint/OctoPrint-FirmwareUpdater #283: Wait for Reset time is simply too short and so it times out shortly before the comm port resets, no log/indication of the original name of firmware file in use

gpread, 2021-11-01:
BTT GTRV1
Octoprint 0.16.1

It seems that in my case the wait for the printer to reset is simply not long enough, from a verbose log inspections it seems like the code waits 10secs ?

From my own crude timing it looks like my board takes 12-15secs for the comm port to reset (time taken to load the new firmware from the SD card and then start to boot it) after the reset/reload command is received.

Can you not add an option to the config screed to allow the wait for reset time to be adjusted by the user (with default of the current 10s) ?

I can get around it with the "Don't wait for reset" option but clearly that's not as robust as detection of the reset, and it seems it would work fine if the wait time was simply longer.

Also I would like to see something in the log which allows one to know which firmware file (name and path) was selected for the update.  Currently it only reports (which is the binary transfer module anyway I suspect) the temporary file name which is used to perform the transfer.

benlye, 2021-11-01:
I can add the timeout as a configurable option.  I cannot log the full name and path of the file you choose to upload as the browser does not actually expose that to the code - it's a security/privacy thing that the browser enforces and I have no control over it.

benlye, 2021-11-01:
v1.12.1b1 is released in the Development branch with configurable timeouts for the printer reset when using Marlin BFT.

Please test it and let me know if it meets your expectations.  If you're happy I'll release it as a GA version.

gpread, 2021-11-02:
Hi, so what I see you've implemented is what I expected/requested.

However, after increasing the wait for reset time, the detection of the reset still fails, although when it does report the failure then the board is successfully fully reset and running the updated firmware.

It seems that the way board and Marlin (I am currently using 2.0.9.2) reacts is not consistent with what octoprint FirmwareUpdater is expecting.

To verify further what is happening I turned on serial logging within octoprint and using the terminal I sent M997 to the board.

The serial log shows that ~13secs after the M997 is sent, there is a msg containing "start" followed by ~1sec later another containing "Software Reset" (followed by a few others regarding the firmware just booted and its configs.

So there is no hard reset of serial communications between the board and octoprint but there are msgs to indicate there was a successful reset.

Actual entries from the octoprint serial log as follows:
2021-11-02 00:30:17,121 - Recv: Not SD printing
2021-11-02 00:30:18,122 - Recv: Not SD printing
2021-11-02 00:30:19,121 - Recv: Not SD printing
2021-11-02 00:30:19,552 - Send: M997
2021-11-02 00:30:32,547 - Recv: start
2021-11-02 00:30:32,549 - Printer sent 'start' while already operational. External reset? Resetting line numbers to be on the safe side
2021-11-02 00:30:32,576 - Send: N0 M110 N0*125
2021-11-02 00:30:33,213 - Recv:  Software Reset
2021-11-02 00:30:33,217 - Recv: Marlin 2.0.9.2-GPR
2021-11-02 00:30:33,218 - Recv: echo: Last Updated: 2021-09-03 | Author: GPR
2021-11-02 00:30:33,219 - Recv: echo: Compiled: Nov  1 2021
2021-11-02 00:30:33,221 - Recv: echo: Free Memory: 97435  PlannerBufferBytes: 6912
2021-11-02 00:30:33,585 - Recv: //action:notification BTT_GTRV1:CR10V3-GPR Ready.
2021-11-02 00:30:35,429 - Recv: echo:V85 stored settings retrieved (643 bytes; crc 14547)
2021-11-02 00:30:35,435 - Recv: Unified Bed Leveling System v1.01 inactive
2021-11-02 00:30:35,479 - Recv: UBL reset
2021-11-02 00:30:36,891 - Recv: //action:prompt_end
2021-11-02 00:30:36,901 - Recv: Testing X connection... OK
2021-11-02 00:30:36,907 - Recv: Testing Y connection... OK
2021-11-02 00:30:36,911 - Recv: Testing Z connection... OK
2021-11-02 00:30:36,914 - Recv: Testing Z2 connection... OK
2021-11-02 00:30:36,916 - Recv: Testing E connection... OK
2021-11-02 00:30:38,088 - Recv: echo:SD card ok
2021-11-02 00:30:38,165 - Recv: ok N0 P63 B31
2021-11-02 00:30:38,169 - Send: N0 M110 N0*125
2021-11-02 00:30:38,174 - Recv: ok N0 P63 B31

gpread, 2021-11-02:
So it would appear (at least in the case of more recent versions of Marlin, I have not tested with older versions), the FirmwareUpdater in marlinbft mode, at a minimum needs to be listening for the "Software Reset" message from the board to reliably detect a successful firmware update and reset.

Alternatively the "start" msg could perhaps be considered to indicate successful firmware flash and reset, with a subsequent  "Software Reset" msg considered an indication of a successful restart. 

Hence also that the configuration options (for marlinbft mode) might need to be modified according to version of Marlin being used (for support of older/wider range of versions?) or  at least changed to be intuitive/consistent with the recent Marlin versions.

M997 behaviour is sequentially and conversationally identical when there is no new firmware to be loaded from SD Card, just the time period from M997 send to "start" msg received is much shorter, in my case ~2secs. This does suggest/confirm that in truth the "restart" phase begins and the "reset" phase completes before receipt of the "start" msg but there is nothing more definitive available to use as state change/success/failure indicators.

benlye, 2021-11-02:
If only it were that simple.  The 'problem' here is that your board (like some Creality boards) reacts to M997 differently to most boards; it has nothing to do with the 'recent version of Marlin'.

My board (a BTT 1.4 Turbo) fully resets on M997 and the serial port goes away and then comes back and this is the behavior that the plugin looks for.  The reset has started when the serial port is no longer present and the restart is over when it comes back.

The reason I cannot simply detect the 'software reset' and 'start' messages is that the plugin must disconnect OctoPrint from the serial port in order to do the binary file transfer, and once that happens it is no longer possible to detect messages on the port.

It *might* be possible to figure out how to accommodate boards which behave like yours and the Creality ones, but I don't have such a board, and there is really not much I can do without one to test on.

The best you can do with the plugin at this point is to check the 'Don't wait for reset' box.

gpread, 2021-11-02:
ok, but surely there is a connection established in order to send the M997 once the transfer has been completed?

And if that is the case why can't those messages be detected using that connection and only after serial comm glitch, timeout or start/reset msg, the octoprint connection be re-established ?

What I appear to see is that, when "do not wait for set" is enabled Octoprint is reconnected immediately after the M997 is sent and itself detects the "start" and "software reset" msgs.

Hence, similarly/alternatively isn't there another potential solution where octoprint is reconnectedearlier and used to pick up the M997 responses (and possibly even issue the M997 in the first place) ?

Given the simplicity of the electronics required for serial connections with the majority of STM32 processors (there are no external componentry required at all). I suspect the difference in behaviours you observe is as much or perhaps always a function of how the board has been integrated (including running firmware config settings).

If the octoprint connection is via a USB interface on the board, then I believe it will always reset (when an M997 is requested) and hence you would detect a serial comm reset (because of how USB is implemented within STM32 or that for some versions external USB interface componentry is required).

If the octoprint connection is via one of the serial interfaces on STM32 (in my case there are several and I am using the Pi labelled one and not the USB-B connection), then I think the only time a comm reset would be detected is if, either the board manufacturer for some reason added some interface circuitry for the serial connection (but for cost, power and size reasons that's not so likely for majority of boards), or they way the board is connected and configured for power results in a power reset (PS_ON signal effects and the comm reset is more a secondary effect).

Regardless in general it seems to me that the less appropriate/desirable behaviour from a printer/board is for a hard reset (comm loss) to occur since by definition an M997 is a "soft reset".

gpread, 2021-11-02:
Unfortunately I don't have any other highly relevant boards to offer up or test and report the behaviours of myself.  I am in the process of replacing a Creality V2.5.2 (ATMega 2560 based) and this remote software update ability was my last pre-kinematics integration test.

benlye, 2021-11-02:
The plugin sends M997 directly to the serial port, without using OctoPrint's serial protocol.  The current implementation is based on what has been seen to work with multiple boards.  I can tell you for sure that when I send M997 to my board it really does reset the entire thing - it is not a 'soft' reset.

From the console:

```
Send: M997
Unexpected error while reading serial port, please consult octoprint.log for details: SerialException: 'read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:3940
Please see https://faq.octoprint.org/serialerror for possible reasons of this.
Changing monitoring state from "Operational" to "Offline after error"
Connection closed, closing down monitor
```

Without the hardware to test I can't (won't) try to solve this as there is already a workaround.  If you really want it to work the way you think it should then you have two options:
1. Figure it out for yourself and propose a code change via a pull request (but don't break things for boards which don't behave the way yours does)
2. Send me a board and I'll do my best

gpread, 2021-11-10:
Thanks, for sure I do/did believe that some boards do cause a comm reset when they receive M997.

I'm using an STM32 based board and for sure with that an M997 will cause the USB connection to reset but none of the serial ports do (TFT, WiFi, Pi, etc). 

Your BTT SKR 1.4 Turbo is LPC1769 based, you didn't indicate if you are connecting using the USB port or not, nor if you are using Marlin.  Perhaps those MCU's UART implementations differ and/or perhaps the Firmware HAL is different and so their UARTs might reset when they bootstrap.

I believe you would see what I have showed/described when octoprint is connected to a serial port (not the USB) on the vast majority, if not all, of STM32 based boards (for sure the more recent Creality Boards are also STM32 based). 

Later when I have more time (right now I need to just finalise my electronics upgrade and be able to print again) I may try to produce modified code myself and propose detailed change.

jezzaaa, 2021-11-11:
Thanks for the plug-in.

FYI I have the same problem with an Ender 3 v4.2.2 running Marlin 2.0.x from TH3D's 2.35 build (I think it's Marlin 2.0.9.1), and FU plugin v1.12.1b1. I've enabled "Don't wait for reset" for now, and I'm hoping that someday someone is able to add a fix.

Some people have worked out a way to connect their Pi directly to the UART chip on the board, bypassing the USB altogether. Is this configuration likely to be allow the plugin to correctly detect the reset?

benlye, 2021-11-11:
I've picked up an Ender 3 4.2.2 board cheaply on eBay.  When it arrives I'll be able to see if this can be solved using the USB port.

benlye, 2021-11-25:
Finally received my v4.2.2 board today.  Will be able to look at this at some point soon.

jezzaaa, 2021-11-25:
@benlye do you have a Paypal account or similar I can send some money for the 4.2.2 board?

benlye, 2021-11-26:
Give v1.13.0b2 a go (you can install it from the Devel branch).  It has an option to use OctoPrint's comms protocol to send the `M997` command and intercept the `start` response.  It seems to work well on my Ender 3 V4.2.2 board.

[image]

For an Ender 3 board or similar you want 'Use timestamp filenames' and 'Use alternative reset' both checked.

You'll notice that I moved some settings about to defrag the UI a bit.  I left the settings for adjusting the timeouts in but moved them to the new "Advanced 2" settings tab.

I also found and fixed a but where if you flashed firmware and then saved a setting change in the plugin without refreshing the web UI the next flash would fail because the last-flashed filename was getting corrupted.  Fixing that was more work than implementing the Ender reset detection!

Please let me know if this works for you, and report any weirdness.

@jezzaaa yes, you can Paypal me at the email address on my [GitHub profile](https://github.com/benlye).  A couple of bucks to help cover the cost of the board would be much appreciated :-)

jezzaaa, 2021-11-28:
Thanks for the work on this.

Alas, no luck with the 1.13.0b2 version. After the firmware transfer was completed, and the plugin said "Resetting the board" there was almost no delay (maybe 1 second) before it showed the pop-up error "Flashing failed/firmware update failed" even though the boot-up logo ("TH3D", cos that's where I fetched the Marlin source) was still on the printer's display at that time of the error message. It seems to be that a pause is required. I tried setting "post flash delay" to 3 seconds, but it didn't seem to wait that long.

Also, before setting "post flash delay" the printer remained disconnected from Octoprint, with a communication error (too many retries, I think) and I had to manually connect. After setting "post flash delay" to 3s, this wasn't a problem.

jezzaaa, 2021-11-28:
[firmware-updater.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/7613580/firmware-updater.log)
attaching logfile

benlye, 2021-11-28:
Can you put the SD card in a computer and delete any bin files on it? The main reason I've seen for the board resetting too quickly is when it doesn't flash the new FW file because the last one is still on the card.

benlye, 2021-11-28:
https://github.com/MarlinFirmware/Marlin/issues/23217

jezzaaa, 2021-11-29:
> Can you put the SD card in a computer and delete any bin files on it? The main reason I've seen for the board resetting too quickly is when it doesn't flash the new FW file because the last one is still on the card.

[image]

It works! I tried 3 times, just to be sure. Yes, it did take an extra 10 seconds (showing "resetting..." on the display) before the reboot happened, which I didn't see before deleting the .bin files.

Awesome! Thanks very much.

One thing I did notice is that the "Flashing successful" pop-up message hangs around forever. The second time I flashed, I thought your plugin had finished checking prematurely, before the board had even reset. But then I realised that the pop-up was from the previous flash. I wonder if it would be a good idea to close any lingering pop-up message as soon as a "flash from..." button is clicked?

benlye, 2021-11-29:
> It works! I tried 3 times, just to be sure. Yes, it did take an extra 10 seconds (showing "resetting..." on the display) before the reboot happened, which I didn't see before deleting the .bin files.
> 

Excellent!

> One thing I did notice is that the "Flashing successful" pop-up message hangs around forever. The second time I flashed, I thought your plugin had finished checking prematurely, before the board had [...]

Done, give 1.13.0b3 a try!

jezzaaa, 2021-11-29:
Perfect. Thanks very much for your work.

gpread, 2021-11-30:
I have just tried 1.13.0b3, and I can confirm it does now seem to work perfectly for my STM32 based board (BTT GTRV1) with Octoprint connected to its RPi serial port (TX3/RX3).

I am using Marlin 2.0.9.2 with BINARY_FILE_TRANSFER capability enabled (M115 shows "Cap:BINARY_FILE_TRANSFER:1")

I believe this will now mean that a large number of boards which utilise the firmware load from SDcard during bootstrap will be able to have their firmware updates reliably managed from Octoprint via the [Marlin Binary File Transfer Protocol](https://github.com/MarlinFirmware/Marlin/pull/14817)

Here is a fragment from my Octoprint log where I successfully performed a firmware update:
[octoprint_log_firmwareupdater_1_13_0b3.txt](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/7625964/octoprint_log_firmwareupdater_1_13_0b3.txt)

Thanks for all the great work here! 👌👍😁

gpread, 2021-11-30:
I have occasionally noticed firmware update failures where the full firmware.bin does not seem to be have been successfully transferred to the SDcard.  I can't tell if that is due to corruption as a result of my serial connectivity and/or some failing in Marlin BFT.  

Currently the only way I can recover after that is to manually remove the corrupt firmware.bin from SDcard, replace with good one and restart the printer board.  Since it otherwise leaves the printer board without valid firmware and hence no way to interact with it remotely.

I don't know if a binary file read is possible with Marlin BFT?  A nice and robust enhancement if it is possible would be to perhaps perform a read after the write followed by a compare, to validate the file on SDcard before the M997 is issued.

Alternatively (but its not as good nor equivalent to what most firmware update programs do prior to bootstrap) the binary file info could be listed (via M20) after the BFT transfer and the sizes compared.  I am aware Marlin's M20 does not currently allow listing of binary files and support your request for [Marlin to enhance M20 for that](https://github.com/MarlinFirmware/Marlin/issues/23217) as there are benefits regardless.

benlye, 2021-12-10:
v1.13.0 has been released.