OctoPrint/OctoPrint-FirmwareUpdater #284: Error during installation: 'Distribution' object has no attribute 'convert_2to3_doctests'

dogmd, 2021-11-07:
**Hardware Setup**
Raspberry Pi 4b, Octoprint 1.7.2

**Describe the problem**
I get a python error when trying to install this package from the Plugin Manager.

**Log Files**
[plugin_pluginmanager_console.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/7491861/plugin_pluginmanager_console.log)
[octoprint.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/7491862/octoprint.log)

**Additional context**
I'd really appreciate any help you can provide with this!

benlye, 2021-11-08:
I can't reproduce this.  I've tried the latest version of the `setuptools` package (which is where the error seems to be coming from) and the same version you have.  Are you able to install other plugins using Plugin Manager?

dogmd, 2021-11-08:
Apologies, I should have attempted that first. The issue is with all plugin manager installs.

closed 2021-11-08 by dogmd