OctoPrint/OctoPrint-FirmwareUpdater #285: Problem programming ATMega2560

MrBill1234, 2021-11-27:
**Hardware Setup**
Anycubic Mega. ATMega2560 (Trigorilla)

**Describe the problem**
Get verification failed message.  Looks like it is not actually programming anything because it is still on the old firmware - Marlin 1.1.9 when i reboot the board.

Trying to flash Marlin 2.x

I decided to take the avrdude command line I found in syslog and run it manually (I added extra debug):

/usr/bin/avrdude -vv -q -p m2560 -c wiring -P /dev/ttyUSB0 -D -U flash:w:/root/MEGA_v1.4.0-b3.hex:i

**Log Files**
avrdude: Version 6.3-20171130
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/etc/avrdude.conf"
         User configuration file is "/root/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/ttyUSB0
         Using Programmer              : wiring
avrdude: wiring_open(): releasing DTR/RTS
avrdude: wiring_open(): asserting DTR/RTS
         AVR Part                      : ATmega2560
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
           flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Wiring
         Description     : Wiring
         Programmer Model: AVRISP
         Hardware Version: 15
         Firmware Version Master : 2.10
         Vtarget         : 0.0 V
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions
avrdude: stk500isp_read_byte(.., signature, 0x0, ...)
avrdude: stk500isp_read_byte(): Sending read memory command: avrdude: stk500isp_read_byte(.., signature, 0x1, ...)
avrdude: stk500isp_read_byte(): Sending read memory command: avrdude: stk500isp_read_byte(.., signature, 0x2, ...)
avrdude: stk500isp_read_byte(): Sending read memory command: avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: stk500isp_read_byte(.., lfuse, 0x0, ...)
avrdude: stk500isp_read_byte(): Sending read memory command: avrdude: stk500isp_read_byte(.., lfuse, 0x0, ...)
avrdude: stk500isp_read_byte(): Sending read memory command: avrdude: stk500isp_read_byte(.., lfuse, 0x0, ...)
avrdude: stk500isp_read_byte(): Sending read memory command: avrdude: safemode: lfuse reads as FF
avrdude: stk500isp_read_byte(.., hfuse, 0x0, ...)
avrdude: stk500isp_read_byte(): Sending read memory command: avrdude: stk500isp_read_byte(.., hfuse, 0x0, ...)
avrdude: stk500isp_read_byte(): Sending read memory command: avrdude: stk500isp_read_byte(.., hfuse, 0x0, ...)
avrdude: stk500isp_read_byte(): Sending read memory command: avrdude: safemode: hfuse reads as D8
avrdude: stk500isp_read_byte(.., efuse, 0x0, ...)
avrdude: stk500isp_read_byte(): Sending read memory command: avrdude: stk500isp_read_byte(.., efuse, 0x0, ...)
avrdude: stk500isp_read_byte(): Sending read memory command: avrdude: stk500isp_read_byte(.., efuse, 0x0, ...)
avrdude: stk500isp_read_byte(): Sending read memory command: avrdude: safemode: efuse reads as FD
avrdude: reading input file "/root/MEGA_v1.4.0-b3.hex"
avrdude: writing flash (134990 bytes):
avrdude: 134990 bytes of flash written
avrdude: verifying flash memory against /root/MEGA_v1.4.0-b3.hex:
avrdude: load data flash data from input file /root/MEGA_v1.4.0-b3.hex:
avrdude: input file /root/MEGA_v1.4.0-b3.hex contains 134990 bytes
avrdude: reading on-chip flash data:
avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0000
         0x14 != 0x0c
avrdude: verification error; content mismatch

avrdude: stk500isp_read_byte(.., lfuse, 0x0, ...)
avrdude: stk500isp_read_byte(): Sending read memory command: avrdude: stk500isp_read_byte(.., lfuse, 0x0, ...)
avrdude: stk500isp_read_byte(): Sending read memory command: avrdude: stk500isp_read_byte(.., lfuse, 0x0, ...)
avrdude: stk500isp_read_byte(): Sending read memory command: avrdude: safemode: lfuse reads as FF
avrdude: stk500isp_read_byte(.., hfuse, 0x0, ...)
avrdude: stk500isp_read_byte(): Sending read memory command: avrdude: stk500isp_read_byte(.., hfuse, 0x0, ...)
avrdude: stk500isp_read_byte(): Sending read memory command: avrdude: stk500isp_read_byte(.., hfuse, 0x0, ...)
avrdude: stk500isp_read_byte(): Sending read memory command: avrdude: safemode: hfuse reads as D8
avrdude: stk500isp_read_byte(.., efuse, 0x0, ...)
avrdude: stk500isp_read_byte(): Sending read memory command: avrdude: stk500isp_read_byte(.., efuse, 0x0, ...)
avrdude: stk500isp_read_byte(): Sending read memory command: avrdude: stk500isp_read_byte(.., efuse, 0x0, ...)
avrdude: stk500isp_read_byte(): Sending read memory command: avrdude: safemode: efuse reads as FD
avrdude: safemode: Fuses OK (E:FD, H:D8, L:FF)

avrdude done.  Thank you.

**Additional context**

I have flashed this before (so it must work) - but I don't remember how.  I _think_ I did it direct from cura using a cable from a PC - but i'd really like to do it this way - much more elegant :)

benlye, 2021-11-27:
You're getting a verification error but that should not stop it from updating. 

It's not clear if you're saying whether or not the board updates when you flash it from the command line. If flashing from the command line isn't working than there's not really a lot I can do to help. 

If flashing from the command line is working but the plugin isn't, you need to post logs from the plugin.

MrBill1234, 2021-11-27:
Hi

It first failed in the plug-in and is failing on the command line too.  I’ve been reading about lots of issues with some of the clone ATMEGA2560’s that ship with some 3D printers.

I suspect it is not your problem - but if you have any suggestions, I’m all ears.

benlye, 2021-11-27:
Like I said - the only error in the output above is verification - it completes the write successfully.  If the plugin log contains only the same error (need to see the log to be sure) then you can just disable the verification check.

You said it is still running Marlin 1.1.9 after the update - are you 100% sure you are flashing a Marlin 2.x hex file?  Have you triple-checked that you are using the correct file?

MrBill1234, 2021-11-27:
100% sure.  It is as if the write isn’t happening.

Perhaps I’ll try direct from a PC.

Where is the plug-in log stored?

benlye, 2021-11-27:
Well, the verification not matching does support the FW not actually being written to the chip.  Might need to take a look at your fuses to see if they are doing something weird, but I'd expect avrdude to give an error.  More likely to be a broken or buggy bootloader, I think.

The plugin is usually in /home/pi/OctoPrint/venv3/lib/python3.7/site-packages/octoprint_firmwareupdater/ if you're using venv.

closed 2021-12-17 by github-actions[bot]