OctoPrint/OctoPrint-FirmwareUpdater #289: SD card not showing up on Pi with SKR Mini E3 V2

mortenmoulder, 2021-12-26:
**Hardware Setup**
Board: SKR Mini E3 V2
Printer: Ender 3 Pro
Marlin 2.0

**Describe the problem**
I cannot get my Raspberry Pi 4 to list the SD card no matter what I do. Usbmount is [configured correctly](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/blob/master/doc/lpc176x.md#usbmount) and I have rebooted the Pi.

I have the firmware correctly updated:

```
Configuration.h:
#define SDSUPPORT

Configuration_adv.h:
#define SDCARD_CONNECTION ONBOARD
```

My SD card I use to flash the firmware with works manually. It changes the .bin extension to .CUR after flashing. However, when I run a M20 command, I just see:

```
Send: M20
Recv: Begin file list
Recv: End file list
Recv: ok
```

even though I also placed a "test.txt" file on the SD card in the same directory as the firmware.cur file.

When I run M22, wait for a few minutes, then run `blkid /dev/sd*` I get nothing. `ls /media/usb` returns nothing. 

I can run M21 and M22 all day long. Not sure if M21 works if the Pi has mounted the SD card, but I doubt it will.

dmesg doesn't show any new messages after running M22.

The printer shows "No Media" after running M22, so I know that M22 works.

**Log Files**
` ~/.octoprint/logs/plugin_firmwareupdater_console.log` is empty.

` ~/.octoprint/logs/octoprint.log` shows nothing of interest.

benlye, 2021-12-27:
M20 only ever returns g-code files.  Neither .bin nor .txt files will be shown in the output.

mortenmoulder, 2021-12-27:
@benlye Yeah I figured that.

benlye, 2021-12-27:
If you plug your board into a PC does it show up the SD card as storage device?

mortenmoulder, 2021-12-27:
UPDATE: Plugging the board into a Windows PC via USB doesn't show the SD card in Windows Explorer either.

benlye, 2021-12-27:
Then you need to:
* Confirm that your board can actually expose the SD card (some can't)
* Figure out how to configure the FW to do so

mortenmoulder, 2021-12-27:
@benlye My SKR Mini E3 V2 should be able to expose the SD card according to what other people report.

The firmware has been configured according to the page I linked to: https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/blob/master/doc/lpc176x.md#marlin-firmware-configuration

benlye, 2021-12-27:
If other people say it works than you need to talk to them to find out what you've done differently.  I don't have the board so can't help you.  Once you have the SD card showing up in Windows there is no reason why it shouldn't show up when connected to the Pi as long as usbmount is properly configured.

mortenmoulder, 2021-12-27:
@benlye I've visited every issue (about 5 or so) with the LPC176x not being able to connect. Most people got it working by updating the firmware to support the SD card stuff (as mentioned in the documentation), so that's not much help.

I'll wait and see if somebody knows what to do. I've done everything the documentation says, but it's still not showing up in Linux or Windows as an USB device (the SD card, that is).

[image]

OctoPrint can easily see the contents of the SD card, though. I can also upload files, but I can't upload a firmware file to the SD card, as it only accepts gcode files.

mortenmoulder, 2021-12-27:
Does it matter which firmware I use? I use this one: https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V2.0/Marlin-2.0.8.2.x-SKR-mini-E3-V2.0

Should I be using something else? I have a TFT35 V3.0 as well.

benlye, 2021-12-27:
Like I said - I don't have the board and don't know any more than is in the documentation or reported in previous issues.  Your best bet is finding a user group or forum for this board and asking there.

benlye, 2021-12-27:
Or you could use Binary File Transfer instead, which doesn't need the SD card mounted on the host.  It's a bit more effort to set up but it works pretty well.

mortenmoulder, 2021-12-27:
Okay, I actually got it working. It was a firmware error on my part.

I was using STM32F103RC_btt_maple specified in platformio.ini because I was using this firmware: https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3

I changed it to STM32F103RC_btt_512K_USB (which makes sense, as _USB allows USB host.. as far as I can read), compiled the firmware, and after issuing a M22 I can mount /dev/sda1 to /media/usb/ with no issues.

Should/could the documentation include that the environment should include _USB, otherwise it won't work? Maybe linking to this issue could help other people as well.

closed 2021-12-27 by benlye