OctoPrint/OctoPrint-FirmwareUpdater #294: Avrdude returned code 1

Volanaro, 2022-01-10:
Cr10s Pro, flashed new firmware, was fine, then decided to go with a diffrent one now im getting that error code, not changed anything. but did notice that the Baud rate had changed since the last firmware, was at 250000 now 115200 when connecting to the printer 

`2022-01-09 22:01:00,735 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2022-01-09 22:01:00,738 - octoprint.server - INFO - --- Log roll over detected ---------------------------------------------------
2022-01-09 22:01:00,738 - octoprint.server - INFO - OctoPrint 1.7.2
2022-01-09 22:01:00,740 - octoprint.plugin.core - INFO - 40 plugin(s) registered with the system:
|  Action Command Notification Support (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/action_command_notification
|  Action Command Prompt Support (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/announcements
|  Anonymous Usage Tracking (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/tracking
|  Application Keys Plugin (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/appkeys
|  Backup & Restore (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/backup
|  Bed Leveling Wizard (0.2.4) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_bedlevelingwizard
|  Bed Visualizer (1.1.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_bedlevelvisualizer
|  BLTouch Plugin (0.3.4) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_BLTouch
|  Core Wizard (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/corewizard
|  Custom Control Editor (0.2.2) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_customControl
|  Discovery (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/discovery
|  DisplayLayerProgress Plugin (1.27.2) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_DisplayLayerProgress
|  DisplayZ (0.2.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_displayz
|  Dragon Order (0.1.7) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_dragon_order
|  Error Tracking (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/errortracking
|  Event Manager (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/eventmanager
| !Filament Manager (1.9.1) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_filamentmanager
|  File Check (2021.2.23) (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_file_check
|  FileManager (0.1.6) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_filemanager
|  Firmware Check (2021.10.11) (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_firmware_check
|  Firmware Updater (1.13.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_firmwareupdater
|  GCode Viewer (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/gcodeviewer
|  ipOnConnect (0.2.4) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_ipOnConnect
|  Logging (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/logging
|  M117NavBar (0.1.1) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_M117NavBar
|  Marlin EEPROM Editor (3.1.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_eeprom_marlin
|  OctoEverywhere! (1.5.1) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_octoeverywhere
|  Octolapse (0.4.1) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse
|  Pi Support Plugin (2021.10.28) (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_pi_support
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/pluginmanager
|  PrintTimeGenius Plugin (2.2.8) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius
|  Resource Monitor (0.3.6) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_resource_monitor
| *SlicerSettingsTab (1.0.3) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_SlicerSettingsTab
|  Software Update (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/softwareupdate
|  System Command Editor (0.3.4) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_systemcommandeditor
| *Terminal Commands (0.1.6) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_TerminalCommands
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/virtual_printer
|  WS281x LED Status (0.7.2) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_ws281x_led_status
| *Yamlpatcher (0.1.1) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_yamlpatcher
Prefix legend: ! = disabled, # = blacklisted, * = incompatible
2022-01-09 22:01:00,744 - octoprint.environment - INFO - Detected environment is Python 3.7.3 under Linux (linux). Details:
|  hardware:
|    cores: 4
|    freq: 1500.0
|    ram: 4041732096
|  os:
|    bits: 32
|    id: linux
|    platform: linux
|  plugins:
|    pi_support:
|      model: Raspberry Pi 4 Model B Rev 1.2
|      octopi_version: 0.17.0
|      throttle_state: '0x0'
|  python:
|    pip: 21.3.1
|    version: 3.7.3
|    virtualenv: /home/pi/oprint
2022-01-09 22:01:00,744 - octoprint.server - INFO - ------------------------------------------------------------------------------
2022-01-09 22:16:00,736 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2022-01-09 22:31:00,737 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2022-01-09 22:46:00,738 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2022-01-09 22:56:02,197 - octoprint.plugins.octoeverywhere - INFO - NotificationsHandler successfully sent 'progress'; ETA: 23071
[... 713 lines cut ...]
2022-01-10 06:55:39,436 - octoprint.plugins.firmwareupdater - INFO - Flash method: avrdude
2022-01-10 06:55:39,452 - octoprint.plugins.firmwareupdater - INFO - Firmware update started
2022-01-10 06:55:39,461 - octoprint.plugins.firmwareupdater - INFO - Disconnecting from printer
2022-01-10 06:55:39,548 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Offline"
2022-01-10 06:55:39,579 - octoprint.plugins.firmwareupdater - INFO - Running '/usr/bin/avrdude -v -q -p m2560 -c wiring -P /dev/ttyUSB0 -D -b 115200 -U flash:w:/tmp/tmpkdbjc5bn:i' in /usr/bin
2022-01-10 06:55:39,584 - octoprint.plugins.action_command_notification - INFO - Notifications cleared
2022-01-10 06:55:40,772 - octoprint.plugins.firmwareupdater - ERROR - Flashing failed. Avrdude returned code 1.
2022-01-10 06:55:40,781 - octoprint.plugins.firmwareupdater - INFO - Reconnecting to printer: port=/dev/ttyUSB0, baudrate=115200, profile={'axes': {'e': {'inverted': False, 'speed': 100}, 'x': {'inverted': False, 'speed': 6000}, 'y': {'inverted': False, 'speed': 6000}, 'z': {'inverted': False, 'speed': 200}}, 'color': 'default', 'extruder': {'count': 1, 'nozzleDiameter': 0.4, 'offsets': [(0.0, 0.0)], 'sharedNozzle': True}, 'heatedBed': True, 'heatedChamber': False, 'id': '_default', 'model': 'cr10s', 'name': 'cr10s', 'volume': {'custom_box': False, 'depth': 300.0, 'formFactor': 'rectangular', 'height': 400.0, 'origin': 'center', 'width': 300.0}}
2022-01-10 06:55:40,881 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Opening serial connection"
2022-01-10 06:55:40,884 - octoprint.util.comm - INFO - Connecting to port /dev/ttyUSB0, baudrate 115200
2022-01-10 06:55:40,931 - octoprint.util.comm - INFO - Changing monitoring state from "Opening serial connection" to "Connecting"
2022-01-10 06:55:40,935 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2022-01-10 06:55:40,940 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2022-01-10 06:55:41,837 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2022-01-10 06:55:46,322 - octoprint.plugins.action_command_notification - INFO - Got a notification: TH3D BLTouch Ready.
2022-01-10 06:55:48,326 - octoprint.util.comm - INFO - Changing monitoring state from "Connecting" to "Operational"
2022-01-10 06:55:48,352 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2022-01-10 06:55:48,401 - octoprint.plugins.firmwareupdater - INFO - Got CONNECTED event
2022-01-10 06:55:48,404 - octoprint.plugins.firmwareupdater - INFO - Run postflash flag is not set
[... cut ...] n: M117 192.168.0.104
2022-01-10 06:55:48,493 - octoprint.util.comm - INFO - Printer reports firmware name "Marlin TH3D UFW 2.36 (Jan 10 2022 05:26:30)"
2022-01-10 06:55:48,495 - octoprint.plugins.firmwareupdater - INFO - Setting BINARY_FILE_TRANSFER capability to False
2022-01-10 06:55:48,504 - octoprint.util.comm - INFO - Firmware states that it supports temperature autoreporting
2022-01-10 06:55:50,976 - octoprint.plugins.action_command_notification - INFO - Got a notification: SD Init Fail
`

benlye, 2022-01-10:
Printer baud rate is set in the firmware so presumably the firmware you were able to install had a lower baud than your original.

Baud rate for AVRDUDE is unrelated to the printer baud rate so you should set the plugin's setting for avrdude baud back to 'Default'.

Not sure I'll be able to tell you what the problem is but I need to see the plugin's log (plugin_firmwareupdater_console.log)

Volanaro, 2022-01-10:
i tried all of the Baud rates and it dose not seem to help :/ , and ok where do i find that? sorry

benlye, 2022-01-10:
Put the avrdude baud setting back to default.

Logs can be accessed under **Settings** -> **Logging** in OctoPrint, or in `~/.octoprint/logs/` on the host.  Both `octoprint.log` and `~/.octoprint/logs/plugin_firmwareupdater_console.log` should be included.

closed 2022-01-30 by github-actions[bot]