OctoPrint/OctoPrint-FirmwareUpdater #297: [Informational] Can the plugin flash both Marlin and Klipper?

jsdokos, 2022-02-14:
**Hardware Setup**
Ender 3 (heavily modified)
BTT SKR Mini E3 V2.0

**Describe the problem**
I am using the [lpc176x](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/blob/master/doc/lpc176x.md) method to flash my board now. I have looked around, and I can't tell if I can flash a generic firmware, or it has to be marlin. I went through old issues, and it seems like yes, but I'd like to not brick my board if possible.

**Log Files**
N/A

**Additional context**
None at this time

benlye, 2022-02-17:
I've no idea, I've never used Klipper. It works great with Marlin.

jsdokos, 2022-02-17:
I've been using it for Marlin for a while, so I may just need to bite the bullet and try it. I'll try and report back, but close this for now.

closed 2022-02-17 by jsdokos