OctoPrint/OctoPrint-FirmwareUpdater #298: Error installing `avrdude` on OctoPi

adilinden, 2022-02-17:
I am getting the following error installing `avrdude` to flash the Creality Melzi V1 board in an Ender-3.

```
pi@octopi-e3:~ $ sudo add-apt-repository ppa:pmjdebruijn/avrdude-release
sudo: add-apt-repository: command not found
```

I had to run `sudo apt install software-properties-common` in order to have the `sudo add-apt-repository ppa:pmjdebruijn/avrdude-release` no longer fail with the above error. But now I am greeted with:

```
pi@octopi-e3:~ $ sudo add-apt-repository ppa:pmjdebruijn/avrdude-release
Traceback (most recent call last):
  File "/usr/bin/add-apt-repository", line 95, in <module>
    sp = SoftwareProperties(options=options)
  File "/usr/lib/python3/dist-packages/softwareproperties/SoftwareProperties.py", line 109, in __init__
    self.reload_sourceslist()
  File "/usr/lib/python3/dist-packages/softwareproperties/SoftwareProperties.py", line 599, in reload_sourceslist
    self.distro.get_sources(self.sourceslist)
  File "/usr/lib/python3/dist-packages/aptsources/distro.py", line 93, in get_sources
    (self.id, self.codename))
aptsources.distro.NoDistroTemplateException: Error: could not find a distribution template for Raspbian/buster
```

I will keep digging, but that's how far I got....

benlye, 2022-02-17:
You're following instructions for Ubuntu. You should be following the instructions for Raspberry Pi.

adilinden, 2022-02-17:
Awesome! That worked!

THanks!

closed 2022-02-17 by adilinden