OctoPrint/OctoPrint-FirmwareUpdater #299: Error installing marlin-binary-protocol on 64-bit os (heatshrink fails)

rerickson1, 2022-02-24:
**Hardware Setup**
Raspberry Pi 4 running Raspberry Pi OS 64-bit lite

**Describe the problem**
Can't install heatshrink dependency.

**Log Files**
```text
(venv) pi@octopi:~/OctoPrint $ pip3 install marlin-binary-protocol
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting marlin-binary-protocol
  Downloading https://www.piwheels.org/simple/marlin-binary-protocol/marlin_binary_protocol-0.0.7-py3-none-any.whl (7.2 kB)
Collecting heatshrink2>=0.9
  Downloading heatshrink2-0.11.0.tar.gz (110 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 110.6/110.6 KB 1.5 MB/s eta 0:00:00
  Preparing metadata (setup.py) ... done
Requirement already satisfied: pyserial>=3.4 in ./venv/lib/python3.9/site-packages (from marlin-binary-protocol) (3.5)
Collecting heatshrink>=0.3.2
  Using cached Heatshrink-0.3.2.tar.gz (280 kB)
  Preparing metadata (setup.py) ... done
Using legacy 'setup.py install' for heatshrink, since package 'wheel' is not installed.
Using legacy 'setup.py install' for heatshrink2, since package 'wheel' is not installed.
Installing collected packages: heatshrink2, heatshrink, marlin-binary-protocol
  Running setup.py install for heatshrink2 ... done
  Running setup.py install for heatshrink ... error
  error: subprocess-exited-with-error
  
  × Running setup.py install for heatshrink did not run successfully.
  │ exit code: 1
  ╰─> [246 lines of output]
      running install
      running build
      running build_py
      creating build
      creating build/lib.linux-aarch64-3.9
      creating build/lib.linux-aarch64-3.9/heatshrink
      copying heatshrink/streams.py -> build/lib.linux-aarch64-3.9/heatshrink
      copying heatshrink/__init__.py -> build/lib.linux-aarch64-3.9/heatshrink
      running build_ext
      building 'heatshrink.core' extension
      creating build/temp.linux-aarch64-3.9
      creating build/temp.linux-aarch64-3.9/heatshrink
      creating build/temp.linux-aarch64-3.9/heatshrink/_heatshrink
      aarch64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-PN012d/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/build/python3.9-PN012d/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I. -I./heatshrink/_heatshrink -I/home/pi/OctoPrint/venv/include -I/usr/include/python3.9 -c heatshrink/_heatshrink/heatshrink_decoder.c -o build/temp.linux-aarch64-3.9/heatshrink/_heatshrink/heatshrink_decoder.o -std=c99
      aarch64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-PN012d/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/build/python3.9-PN012d/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I. -I./heatshrink/_heatshrink -I/home/pi/OctoPrint/venv/include -I/usr/include/python3.9 -c heatshrink/_heatshrink/heatshrink_encoder.c -o build/temp.linux-aarch64-3.9/heatshrink/_heatshrink/heatshrink_encoder.o -std=c99
      aarch64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-PN012d/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/build/python3.9-PN012d/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I. -I./heatshrink/_heatshrink -I/home/pi/OctoPrint/venv/include -I/usr/include/python3.9 -c heatshrink/core.c -o build/temp.linux-aarch64-3.9/heatshrink/core.o -std=c99
      heatshrink/core.c: In function ‘__pyx_pf_10heatshrink_4core_7Encoder_7fill’:
      heatshrink/core.c:4508:38: warning: comparison of integer expressions of different signedness: ‘size_t’ {aka ‘long unsigned int’} and ‘Py_ssize_t’ {aka ‘long int’} [-Wsign-compare]
       4508 |     __pyx_t_5 = ((__pyx_v_total_sunk < __pyx_t_9) != 0);
            |                                      ^
      heatshrink/core.c: In function ‘PyInit_core’:
      heatshrink/core.c:7279:39: error: ‘PyTypeObject’ {aka ‘struct _typeobject’} has no member named ‘tp_print’
       7279 |   __pyx_type_10heatshrink_4core_Writer.tp_print = 0;
            |                                       ^
      heatshrink/core.c:7283:39: error: ‘PyTypeObject’ {aka ‘struct _typeobject’} has no member named ‘tp_print’
       7283 |   __pyx_type_10heatshrink_4core_Reader.tp_print = 0;
            |                                       ^
      heatshrink/core.c:7287:59: error: ‘PyTypeObject’ {aka ‘struct _typeobject’} has no member named ‘tp_print’
       7287 |   __pyx_type_10heatshrink_4core___pyx_scope_struct___drain.tp_print = 0;
            |                                                           ^
      heatshrink/core.c: In function ‘__Pyx_ParseOptionalKeywords’:
      heatshrink/core.c:7712:21: warning: ‘_PyUnicode_get_wstr_length’ is deprecated [-Wdeprecated-declarations]
       7712 |                     (PyUnicode_GET_SIZE(**name) != PyUnicode_GET_SIZE(key)) ? 1 :
            |                     ^
      In file included from /usr/include/python3.9/unicodeobject.h:1026,
                       from /usr/include/python3.9/Python.h:97,
                       from heatshrink/core.c:24:
      /usr/include/python3.9/cpython/unicodeobject.h:446:26: note: declared here
        446 | static inline Py_ssize_t _PyUnicode_get_wstr_length(PyObject *op) {
            |                          ^~~~~~~~~~~~~~~~~~~~~~~~~~
[... 192 lines cut ...]
            |                      curexc_traceback
      heatshrink/core.c:11168:13: error: ‘PyThreadState’ {aka ‘struct _ts’} has no member named ‘exc_type’; did you mean ‘curexc_type’?
      11168 |     tstate->exc_type = *type;
            |             ^~~~~~~~
            |             curexc_type
      heatshrink/core.c:11169:13: error: ‘PyThreadState’ {aka ‘struct _ts’} has no member named ‘exc_value’; did you mean ‘curexc_value’?
      11169 |     tstate->exc_value = *value;
            |             ^~~~~~~~~
            |             curexc_value
      heatshrink/core.c:11170:13: error: ‘PyThreadState’ {aka ‘struct _ts’} has no member named ‘exc_traceback’; did you mean ‘curexc_traceback’?
      11170 |     tstate->exc_traceback = *tb;
            |             ^~~~~~~~~~~~~
            |             curexc_traceback
      error: command '/usr/bin/aarch64-linux-gnu-gcc' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> heatshrink

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.
```

benlye, 2022-02-24:
This is a known problem with the Marlin BFT package - the PyPi pacakge for Heatshrink is not compatible with Python 3 so a different version must first be installed manually, as explained in the **Python 3** instructions:
https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/blob/master/doc/marlinbft.md#python-3

rerickson1, 2022-02-24:
Ah yes, I tried the 0.3.3 version and got a similar error.

benlye, 2022-02-24:
Not much I can do.  You'll need to ask the owner of the Heatshrink package if they can create 64-bit compatible version.

rerickson1, 2022-02-24:
Thanks, I just wanted to make sure I wasn't missing anything.

rerickson1, 2022-02-28:
Reopening this. Looks like the heatshrink package is not longer maintained and its now heatshrink2. Can you change marlin-binary-protocl to not depend on the heatshrink package and just use heatshrink2?

rerickson1, 2022-02-28:
sorry, this issue actually belongs to https://github.com/trippwill/marlin-binary-protocol

closed 2022-02-28 by rerickson1