OctoPrint/OctoPrint-FirmwareUpdater #300: Support for Artillery Ruby / Sidewinder X2

synapsengymnastik, 2022-02-25:
Hello !
i want to ask, if this plug in supports the Artillery Ruby board with an STM32F401RC ?

or if not, if i can do something to get it work ?

thank you !

synapsengymnastik, 2022-02-26:
ok, i found a way !
with dfu util !

thank you for this plug in :)

benlye, 2022-03-02:
> × Encountered error while trying to install package.
╰─> Heatshrink

Assuming you're on Python 3, are you following the correct (Python 3) instructions, or trying to follow the Python 2 instructions?  Also, heatshrink only works on 32-bit, not 64-bit.

Sonicboom247, 2022-03-02:
Hello, and thanks for your reply, i just verified, I am on Python 3.10.2
2022-03-02 08:01:30,041 - octoprint.plugins.tracking - INFO - Sent tracking event startup, payload: {'version': '1.7.3', 'os': 'windows', 'bits': 64, 'python': '3.10.2', 'pip': '22.0.3', 'cores': 8, 'freq': 3200.0, 'ram': 33215954944}
2022-03-02 08:01:35,581 - octoprint.server.preemptive_cache - INFO - ... done in 6.98s

going through directions once more

Sonicboom247, 2022-03-02:
it "looks" compatible with 64 bit but i could be very wrong, this is the output
[50 lines of output]
      running install
      running build
      running build_py
      creating build
      creating build\lib.win-amd64-3.10
      creating build\lib.win-amd64-3.10\heatshrink
      copying heatshrink\streams.py -> build\lib.win-amd64-3.10\heatshrink
      copying heatshrink\__init__.py -> build\lib.win-amd64-3.10\heatshrink
      running build_ext
      building 'heatshrink.core' extension
      creating build\temp.win-amd64-3.10
      creating build\temp.win-amd64-3.10\Release
      creating build\temp.win-amd64-3.10\Release\heatshrink
      creating build\temp.win-amd64-3.10\Release\heatshrink\_heatshrink
      C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.28.29910\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -I. -I./heatshrink/_heatshrink -Ic:\octoprint\new\include -Ic:\program files\python310\include -Ic:\program files\python310\Include -IC:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.28.29910\include -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\winrt -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\cppwinrt /Tcheatshrink/_heatshrink/heatshrink_decoder.c /Fobuild\temp.win-amd64-3.10\Release\heatshrink/_heatshrink/heatshrink_decoder.obj -std=c99
      cl : Command line warning D9002 : ignoring unknown option '-std=c99'
      heatshrink_decoder.c
      heatshrink/_heatshrink/heatshrink_decoder.c(81): warning C4334: '<<': result of 32-bit shift implicitly converted to 64 bits (was 64-bit shift intended?)
      heatshrink/_heatshrink/heatshrink_decoder.c(111): warning C4267: '+=': conversion from 'size_t' to 'uint16_t', possible loss of data
      heatshrink/_heatshrink/heatshrink_decoder.c(282): warning C4267: '-=': conversion from 'size_t' to 'uint16_t', possible loss of data
      C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.28.29910\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -I. -I./heatshrink/_heatshrink -Ic:\octoprint\new\include -Ic:\program files\python310\include -Ic:\program files\python310\Include -IC:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.28.29910\include -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\winrt -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\cppwinrt /Tcheatshrink/_heatshrink/heatshrink_encoder.c /Fobuild\temp.win-amd64-3.10\Release\heatshrink/_heatshrink/heatshrink_encoder.obj -std=c99
      cl : Command line warning D9002 : ignoring unknown option '-std=c99'
      heatshrink_encoder.c
      heatshrink/_heatshrink/heatshrink_encoder.c(98): warning C4267: '=': conversion from 'size_t' to 'uint16_t', possible loss of data
      heatshrink/_heatshrink/heatshrink_encoder.c(152): warning C4267: 'initializing': conversion from 'size_t' to 'uint16_t', possible loss of data
      heatshrink/_heatshrink/heatshrink_encoder.c(536): warning C4244: '=': conversion from 'uint16_t' to 'uint8_t', possible loss of data
      heatshrink/_heatshrink/heatshrink_encoder.c(603): warning C4267: '-=': conversion from 'size_t' to 'uint16_t', possible loss of data
      heatshrink/_heatshrink/heatshrink_encoder.c(595): warning C4267: 'initializing': conversion from 'size_t' to 'uint16_t', possible loss of data
      heatshrink/_heatshrink/heatshrink_encoder.c(596): warning C4267: 'initializing': conversion from 'size_t' to 'uint16_t', possible loss of data
      C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.28.29910\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -I. -I./heatshrink/_heatshrink -Ic:\octoprint\new\include -Ic:\program files\python310\include -Ic:\program files\python310\Include -IC:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.28.29910\include -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\winrt -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\cppwinrt /Tcheatshrink/core.c /Fobuild\temp.win-amd64-3.10\Release\heatshrink/core.obj -std=c99
      cl : Command line warning D9002 : ignoring unknown option '-std=c99'
      core.c
      heatshrink/core.c(5226): warning C4018: '<': signed/unsigned mismatch
      heatshrink/core.c(7176): error C2105: '++' needs l-value
      heatshrink/core.c(7178): error C2105: '--' needs l-value
      heatshrink/core.c(7286): error C2105: '++' needs l-value
      heatshrink/core.c(7288): error C2105: '--' needs l-value
      heatshrink/core.c(7929): error C2039: 'tp_print': is not a member of '_typeobject'
      c:\program files\python310\include\cpython/object.h(191): note: see declaration of '_typeobject'
      heatshrink/core.c(7937): error C2039: 'tp_print': is not a member of '_typeobject'
[... cut ...]
      heatshrink/core.c(13329): warning C4047: '=': 'PyObject *' differs in levels of indirection from 'int'
      heatshrink/core.c(13413): warning C4047: '=': 'PyObject *' differs in levels of indirection from 'int'
      error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\BuildTools\\VC\\Tools\\MSVC\\14.28.29910\\bin\\HostX86\\x64\\cl.exe' failed with exit code 2
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> Heatshrink

closed 2022-03-22 by github-actions[bot]