OctoPrint/OctoPrint-FirmwareUpdater #301: MBFT cant connect

00ihsan, 2022-03-01:
First time it worked on my Ender 3 Jyers fw. now i get this in log:

2022-03-01 02:36:08,080 - octoprint.plugins.firmwareupdater - INFO - Connecting to printer at '/dev/ttyUSB0' using Marlin BFT protocol
2022-03-01 02:36:08,081 - octoprint.plugins.firmwareupdater - INFO - Connecting: Switching Marlin to binary protocol...
2022-03-01 02:36:08,096 - octoprint.plugins.firmwareupdater - INFO - Connection synced [0], binary protocol version 0.1.0, 96 byte payload buffer
2022-03-01 02:36:08,096 - octoprint.plugins.firmwareupdater - INFO - Transfering file to printer using Marlin BFT '/tmp/tmpnyq86hql' -> /firmware.bin
2022-03-01 02:36:09,352 - octoprint.plugins.firmwareupdater - ERROR - Flashing failed. Unable to transfer file.
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_firmwareupdater/methods/marlinbft.py", line 105, in _flash_marlinbft
    filetransfer.copy(firmware, target, True, False)
  File "/home/pi/oprint/lib/python3.7/site-packages/binproto2/protocols.py", line 407, in copy
    self.open(dest_filename, compression_support, dummy)
  File "/home/pi/oprint/lib/python3.7/site-packages/binproto2/protocols.py", line 372, in open
    raise Exception("Cannot open file on client")
Exception: Cannot open file on client
2022-03-01 02:36:10,361 - octoprint.plugins.firmwareupdater - INFO - Reconnecting to printer: port=/dev/ttyUSB0, baudrate=115200, profile={'axes': {'e': {'inverted': False, 'speed': 300}, 'x': {'inverted': False, 'speed': 6000}, 'y': {'inverted': False, 'speed': 6000}, 'z': {'inverted': False, 'speed': 200}}, 'color': 'default', 'extruder': {'count': 1, 'nozzleDiameter': 0.4, 'offsets': [(0.0, 0.0)], 'sharedNozzle': False}, 'heatedBed': True, 'heatedChamber': False, 'id': '_default', 'model': 'Generic RepRap Printer', 'name': 'E3V2', 'volume': {'custom_box': False, 'depth': 220.0, 'formFactor': 'rectangular', 'height': 250.0, 'origin': 'lowerleft', 'width': 220.0}}

benlye, 2022-03-01:
You'll need to open an issue with the developers of the binary file transfer protocol or your particular firmware.

00ihsan, 2022-03-02:
but it worked and i dod not change the fw?

benlye, 2022-03-02:
> but it worked and i dod not change the fw?

If it worked you must have changed the firmware...

In any case, you're getting an error from the protocol, which is not my work and I'm not the one to support it.