OctoPrint/OctoPrint-FirmwareUpdater #302: What firmware?

serget2, 2022-03-02:
What is this firmware update that I keep getting? What firmware does this update? That of my 3D-printer? My ender 3 pro with a BL-touch, an SKR Mini v 1.3 and a Swiss micro direct drive? You want me to update my firmware and mess all those settings up that I carefully put there?  Or is this some other firmware? Like what device besides my 3D-printer and my PC require "Firmware", you do realise that that term is reserved for steering hardware, otherwise it would be software

benlye, 2022-03-02:
I have literally no idea what you're on about or what the point of this issue is.

serget2, 2022-03-03:
The fact that every time Octoprint starts and  shows messages with an update of both Octoprint and one called "Firmware", when I click on read more I end up on this page, which explains nothing, zilch zero niente. WHAT does this firmware do? What does it update?

benlye, 2022-03-03:
It sounds like you have an older version of the Firmware Updater plugin installed and OctoPrint is asking if you want to update to the latest version of the plugin.  That's just a software update within OctoPrint.

This project does not publish or provide any firmware for your printer or anything else - it provides an OctoPrint plugin that you can use to update the firmware, but you have to get the firmware from somewhere else (or compile it yourself).

If you don't want to use the plugin just disable it.

serget2, 2022-03-04:
OK that clears up a lot, I was just annoyed that the message kept popping up but was titled "New firmware" and not "New Firmware UPDATER", so naturally I was scared that it would actually try to update the firmware, but it is only the Software to update, well thank you for clearing that up, Sorry to waste your time, I just become ver anxious whenever it pertains Firmware because I am not a programmer and it already cost me a lot of time to get the firmware working with every upgrade I have.

closed 2022-03-04 by benlye