OctoPrint/OctoPrint-FirmwareUpdater #303: Major Fault in Logging

CmdrCody51, 2022-03-02:
**Hardware Setup**
N/A

**Describe the problem**
Firmware updater id logging identical logs as Software updater!

**Log Files**
[octoprint-logs.zip](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/8171095/octoprint-logs.zip)

**Additional context**
Add any other context about the problem here.
https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/blob/ad6bd0134ab48494807ceb40e668ba9a24bd0596/octoprint_firmwareupdater/__init__.py#L76

benlye, 2022-03-02:
Oops.  That's been there a loooong time - since before I inherited the plugin.  Will be fixed in the next release.
https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/commit/a0fb8cc93851e89733f2a24b14d1ed89e3323b0d

benlye, 2022-03-18:
Fixed in v1.13.2

closed 2022-03-18 by benlye