OctoPrint/OctoPrint-FirmwareUpdater #305: Error when configuring Firmware Updater

andraisr, 2022-03-05:
**Hardware Setup**
ATmega2560

**Describe the problem**
Path to avrdude "/usr/bin/avrdude" comes up with the following error "The path doesn't exist"

**Log Files**
Attach the OctoPrint log and the Firmware Updater log Log files, which can be accessed under **Settings** -> **Logging** in OctoPrint, or in `~/.octoprint/logs/` on the host.  Both `octoprint.log` and `~/.octoprint/logs/plugin_firmwareupdater_console.log` should be included.

**I attached the OctoPrint log. The Firmware updater log had nothing in it.**

**Additional context**
Add any other context about the problem here.
[octoprint-logs.zip](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/8189884/octoprint-logs.zip)
[image]

andraisr, 2022-03-05:
This is on a new install of the Octoprint to move to Python 3 from the image file.

benlye, 2022-03-05:
Did you follow the instructions and install avrdude?

https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/blob/master/doc/avrdude.md#avrdude-installation

closed 2022-03-24 by github-actions[bot]