OctoPrint/OctoPrint-FirmwareUpdater #307: Timeout communicating with programmer

Muloo-r, 2022-03-11:
**Hardware Setup**
Prusa MK3S+
**Pi Zero 2 connected on the Etsy with pins (not USB)**

**Describe the problem**
**Flashing failed: Timeout communicating with programmer**

[image]

**Log Files**
2022-03-03 20:21:40,774   /home/pi/oprint/bin/python3 -m pip --disable-pip-version-check install https://github.com/OctoPrint/OctoPrint-PiSupport/archive/2022.3.1.zip --no-cache-dir
2022-03-03 20:21:45,020 > Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
2022-03-03 20:21:45,021 > Collecting https://github.com/OctoPrint/OctoPrint-PiSupport/archive/2022.3.1.zip
2022-03-03 20:21:45,023 > Downloading https://github.com/OctoPrint/OctoPrint-PiSupport/archive/2022.3.1.zip (57 kB)
2022-03-03 20:21:47,179 > Building wheels for collected packages: OctoPrint-PiSupport
2022-03-03 20:21:47,180 > Building wheel for OctoPrint-PiSupport (setup.py): started
2022-03-03 20:21:49,322 > Building wheel for OctoPrint-PiSupport (setup.py): finished with status 'done'
2022-03-03 20:21:49,323 > Created wheel for OctoPrint-PiSupport: filename=OctoPrint_PiSupport-2022.3.1-py2.py3-none-any.whl size=33056 sha256=ad826788188f042cd7b4da2ecc7e2c87e97d977f62f4bff5f2f005222a9df2ec
2022-03-03 20:21:49,325 > Stored in directory: /tmp/pip-ephem-wheel-cache-l7e24las/wheels/6c/ea/61/59c6f38b61b12f483f89c558f510ff4c7efb0475c0671966ab
2022-03-03 20:21:49,325 > Successfully built OctoPrint-PiSupport
2022-03-03 20:21:51,414 > Installing collected packages: OctoPrint-PiSupport
2022-03-03 20:21:51,415 > Attempting uninstall: OctoPrint-PiSupport
2022-03-03 20:21:51,416 > Found existing installation: OctoPrint-PiSupport 2021.10.28
2022-03-03 20:21:51,416 > Uninstalling OctoPrint-PiSupport-2021.10.28:
2022-03-03 20:21:51,416 > Successfully uninstalled OctoPrint-PiSupport-2021.10.28
2022-03-03 20:21:51,417 > Successfully installed OctoPrint-PiSupport-2022.3.1
2022-03-11 20:55:11,666 
2022-03-11 20:55:11,677 /usr/bin/avrdude -v -q -p m2560 -c wiring -P /dev/ttyAMA0 -D -U flash:w:/tmp/tmpnl6nvhu3:i
2022-03-11 20:55:12,421 > 
2022-03-11 20:55:12,422 > avrdude: Version 6.3-20171130
2022-03-11 20:55:12,423 >          Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
2022-03-11 20:55:12,424 >          Copyright (c) 2007-2014 Joerg Wunsch
2022-03-11 20:55:12,425 > 
2022-03-11 20:55:12,426 >          System wide configuration file is "/etc/avrdude.conf"
2022-03-11 20:55:12,427 >          User configuration file is "/home/pi/.avrduderc"
2022-03-11 20:55:12,427 >          User configuration file does not exist or is not a regular file, skipping
2022-03-11 20:55:12,428 > 
2022-03-11 20:55:12,429 >          Using Port                    : /dev/ttyAMA0
2022-03-11 20:55:12,429 >          Using Programmer              : wiring
2022-03-11 20:55:12,430 > 
2022-03-11 20:55:16,126 > avrdude: stk500v2_ReceiveMessage(): timeout
2022-03-11 20:55:16,135 > 
2022-03-11 20:55:20,331 > avrdude: stk500v2_ReceiveMessage(): timeout
2022-03-11 20:55:20,333 > 
2022-03-11 20:55:24,525 > avrdude: stk500v2_ReceiveMessage(): timeout
2022-03-11 20:55:24,527 > 
2022-03-11 20:55:28,720 > avrdude: stk500v2_ReceiveMessage(): timeout
2022-03-11 20:55:28,726 > 
2022-03-11 20:55:32,914 > avrdude: stk500v2_ReceiveMessage(): timeout
2022-03-11 20:55:32,916 > 
2022-03-11 20:55:37,109 > avrdude: stk500v2_ReceiveMessage(): timeout
2022-03-11 20:55:37,110 > avrdude: stk500v2_getsync(): timeout communicating with programmer
2022-03-11 20:55:37,111 > 
2022-03-11 20:55:37,112 > avrdude done.  Thank you.

---------------------------------------------------------------------------------------------------------------------------------

2022-03-11 20:55:11,161 - octoprint.plugins.firmwareupdater - INFO - Firmware update profile index: 0
2022-03-11 20:55:11,170 - octoprint.plugins.firmwareupdater - INFO - Firmware update profile name: Default
2022-03-11 20:55:11,171 - octoprint.plugins.firmwareupdater - INFO - Printer port: /dev/ttyAMA0
2022-03-11 20:55:11,179 - octoprint.plugins.firmwareupdater - INFO - Flash method: avrdude
2022-03-11 20:55:11,530 - octoprint.plugins.firmwareupdater - INFO - Firmware update started
2022-03-11 20:55:11,531 - octoprint.plugins.firmwareupdater - INFO - Disconnecting from printer
2022-03-11 20:55:11,607 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Offline"
2022-03-11 20:55:11,637 - octoprint.plugins.action_command_notification - INFO - Notifications cleared
2022-03-11 20:55:11,665 - octoprint.plugins.firmwareupdater - INFO - Running '/usr/bin/avrdude -v -q -p m2560 -c wiring -P /dev/ttyAMA0 -D -U flash:w:/tmp/tmpnl6nvhu3:i' in /usr/bin
2022-03-11 20:55:12,038 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:10.0.0.208
2022-03-11 20:55:12,250 - octoprint.server.util.flask - INFO - Passively logging in user muloo from ::ffff:10.0.0.208
2022-03-11 20:55:12,251 - octoprint.access.users - INFO - Logged in user: muloo
2022-03-11 20:55:12,439 - octoprint.server.util.sockjs - INFO - User muloo logged in on the socket from client ::ffff:10.0.0.208
2022-03-11 20:55:16,451 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:10.0.0.208
2022-03-11 20:55:37,114 - octoprint.plugins.firmwareupdater - ERROR - Flashing failed. Timeout communicating with programmer.
2022-03-11 20:55:37,125 - octoprint.plugins.firmwareupdater - INFO - Reconnecting to printer: port=/dev/ttyAMA0, baudrate=115200, profile={'axes': {'e': {'inverted': False, 'speed': 300}, 'x': {'inverted': False, 'speed': 6000}, 'y': {'inverted': False, 'speed': 6000}, 'z': {'inverted': False, 'speed': 200}}, 'color': 'default', 'extruder': {'count': 2, 'nozzleDiameter': 0.4, 'offsets': [(0.0, 0.0)], 'sharedNozzle': True}, 'heatedBed': True, 'heatedChamber': False, 'id': 'prusa_duo', 'model': 'MK3S', 'name': 'Prusa_duo', 'volume': {'custom_box': False, 'depth': 210.0, 'formFactor': 'rectangular', 'height': 210.0, 'origin': 'lowerleft', 'width': 250.0}}
2022-03-11 20:55:37,276 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Opening serial connection"
2022-03-11 20:55:37,278 - octoprint.util.comm - INFO - Connecting to port /dev/ttyAMA0, baudrate 115200
2022-03-11 20:55:37,305 - octoprint.util.comm - INFO - Changing monitoring state from "Opening serial connection" to "Connecting"
2022-03-11 20:55:37,315 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2022-03-11 20:55:37,328 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2022-03-11 20:55:37,338 - octoprint.util.comm - INFO - Changing monitoring state from "Connecting" to "Operational"
2022-03-11 20:55:37,406 - octoprint.plugins.firmwareupdater - INFO - Got CONNECTED event
2022-03-11 20:55:37,409 - octoprint.plugins.firmwareupdater - INFO - Run postflash flag is not set
2022-03-11 20:55:37,421 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2022-03-11 20:55:37,454 - octoprint.util.comm - INFO - Printer reports firmware name "Prusa-Firmware 3.10.0 based on Marlin"
2022-03-11 20:55:37,454 - octoprint.util.comm - INFO - Detected Prusa firmware, enabling relevant features for issue free communication
2022-03-11 20:55:37,461 - octoprint.util.comm - INFO - Firmware states that it supports temperature autoreporting
2022-03-11 20:55:56,035 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:10.0.0.25

**Additional context**
Tested different baud rate in advance setting without success

benlye, 2022-03-12:
I don't know your board at all so I'm going to have to make a few guesses.  Can you explain exactly how it's wired?

If you're not using the USB port then I suspect that you will need to change the programmer type from `wiring` to something else, but I really don't know what will work for you.  You're probably better off asking for help in a user group for that board/printer.

Muloo-r, 2022-03-12:
HI, which board don't you know, the Prusa MK3 board or the Raspberry PI ZERO 2 ? 
It was my guess also that I needed some other type instead of "wiring" I think I've tested them all without luck

On the Pi these connections are used

[image]

And on the Prusa board, not sure what connections they are.

[image]

But now that I look closer to the description I see the Prusa has the **EINSY RAMBo board** so I guess I have to check the settings for the RAMBo?

Muloo-r, 2022-03-12:
A test with AVR programmer set to : avr109 gives this error:
Unexpected error while connecting to serial port /dev/ttyAMA0, baudrate 115200 from hook default: SerialException: '[Errno 11] Could not exclusively lock port /dev/ttyAMA0: [Errno 11] Resource temporarily unavailable' @ comm.py:_open_serial:3779

cyb3rdog, 2022-03-12:
**+ following...**

I have the same setup, Prusa MK3S+, Einsy 10a Rambo board, with Rpi Zero 2W over GPIO, Plugin Version: 1.13.1
and I have very same issue - timeout communicating with programmer.

Following are log extracts from my trials:

*wiring, with overriding baudrate*
```
/usr/bin/avrdude -v -q -p m2560 -c wiring -P /dev/ttyAMA0 -D -b 115200 -U flash:w:/tmp/tmpgpphvqcc:i 
>          Using Programmer              : wiring
>          Overriding Baud Rate          : 115200
> 
> avrdude: stk500v2_ReceiveMessage(): timeout 
> avrdude: stk500v2_getsync(): timeout communicating with programmer 
```

*using linuxgpio*
```
/usr/bin/avrdude -v -q -p m2560 -c linuxgpio -P /dev/ttyAMA0 -D -U flash:w:/tmp/tmpgv2n67xh:i
> avrdude: Can't find programmer id "linuxgpio"
``` 

During flashing in the plugin, octoprint disconnects and re-connects its serial over GPIO just fine:
[image]

Any ideas?

benlye, 2022-03-12:
While you figure out what settings you need, you're probably better off copying a firmware file to the Pi and then using an SSH shell to run avrdude directly, taking OctoPrint and the firmware updater plugin out of the loop.

Once you have a command line which works it will be very easy to configure the plugin with the appropriate settings.

What I'm saying is that the problems you're having are really nothing to do with this plugin.  You will most likely have better luck getting help in a user forum for your printer / board.

cyb3rdog, 2022-03-12:
Many thanks for your reply Ben, and yes, I completely agree, it actually has nothing to do with this plugin.

I just thought, that having the solution in this thread might be of great help to folk like us. 
So, appologies for spamming your repo.

The main point for me, seems to be, that when try that over the USB with ```avrdude-slic3r -v -p atmega2560 -c wiring -P COM6 -b 115200 -D``` it works right on...
On the other hand, when I do the same thing from Rpi over GPIO with "avrdude -v -p atmega2560 -c wiring -P /dev/ttyAMA0 -b 115200 -D" it timeouts...
Otherwise, the octoprint itself connect over the /dev/ttyAMA0 just fine, and everything else also works fine, so the gpio, serial, disabled bluetooth and terminal settings are all also ok.

So I guess that i'll just need to have to deep dive in to why the avrdude itself fails... :)

Thanks and sorry again for an 'unrelated' spam.
Have a great day!

closed 2022-04-01 by github-actions[bot]