OctoPrint/OctoPrint-FirmwareUpdater #309: Flash Prusa MK3S firmware with custom method

Vortex-TH, 2022-04-18:
I also have the problem that other languages than English are not flashed to my Prusa MK3S #253

As mentioned in #139, a second flash command is necessary that could potentially be done through a 'custom' flash method.

Has this been implemented? I'm not sure where I should enter custom commands. What is the recommended way? 

Not sure if I should append it somehow to this customize command line setting:
{avrdude} -v -q -p {mcu} -c {programmer} -P {port} -D -C {conffile} -b {baudrate} {disableverify} -U flash:w:{firmware}:i

It didn't work so far..

benlye, 2022-04-19:
I never got around to implementing the custom command method.  I'll see what I can come up with - I'm thinking maybe just a checkbox for Prusa MK3S which would tell the plugin to run both commands.

arekm, 2022-07-27:
Or autodetect, octoprint fires event when it finds firmware information early (in response to M115 command)

````
   eventManager().fire(
                        Events.FIRMWARE_DATA, {"name": firmware_name, "data": data}
                    )
````

and uses such condition for autodetection of prusa:

````
 elif "prusa-firmware" in firmware_name.lower():
````

For reference, commands used by current prusaslicer (on windows) when upgrading firmware:

````
avrdude-slic3r -v -p atmega2560 -c wiring -P COM5 -b 115200 -D \
   -U flash:w:0:C:\prusa3d_fw_MK3S_MK3S+_3_11_1-RC1_4987.hex:i

avrdude-slic3r -v -p atmega2560 -c arduino -P COM5 -b 115200 -D \
   -u -U flash:w:1:C:\prusa3d_fw_MK3S_MK3S+_3_11_1-RC1_4987.hex:i
````

arekm, 2022-08-17:
Implemented as https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/pull/325

Unfortunately stock avrdude is not being able to flash languages part of firmware.  Modified variant is needed.

https://github.com/arekm/avrdude-prusa-mk3

It uses special serial command to init external flash mode on MK3 hardware and also has workarounds
for broken Prusa USB to serial converters.