OctoPrint/OctoPrint-FirmwareUpdater #311: arvdude?

dgoldenholz, 2022-05-05:
**Hardware Setup**
Please board type and printer make and model.
Prusa MK3+ i3
Octoprint  1.7.3
Firmware-Updater 13.1.2
Raspberry pi 4

**Describe the problem**
Please provide a clear and concise description of what the problem is, including details of any error messages you receive (screenshots are helpful).
I first did sudo apt-get update
that worked fine. 
but then...

 sudo apt-get install arvdude
[sudo] password for pi: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package arvdude

**Log Files**
not relevant

**Additional context**
any ideas?
I ran this:
sudo apt-get autoremove && sudo apt-get -f install && sudo apt-get update && sudo apt-get upgrade -y
and it did not help

benlye, 2022-05-11:
You misspelt avrdude.

closed 2022-05-11 by benlye

dgoldenholz, 2022-05-11:
thank you @benlye ! dyslexics are people too.