OctoPrint/OctoPrint-FirmwareUpdater #312: Update reports successful, yet firmware is not updated.

Papadoma, 2022-05-08:
**Hardware Setup**
MKS Gen-L 32bit, Anet A8 (modified), Marlin 2.0.9.3

**Describe the problem**
I tried updating the firmware using the updater plugin. Even though the popup reports successful, the new firmware with the changes is not applied and M114 does not report the newly built version. I've checked the SD card and it does not contain the uploaded firmware. I've tried both versions "lpc176x" and "marlinbft". They get configured correctly but the end result is the same. The only way I can update the firmware is to put it on the SD card as firmware.bin and restart the printer. 

**Log Files**
[octoprint-logs.zip](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/8647828/octoprint-logs.zip)

**Additional context**
I've noticed that dmesg reports the following:

`
[ 1101.290382] sd 0:0:0:0: [sda] 15431680 512-byte logical blocks: (7.90 GB/7.36 GiB)
[ 1101.307318]  sda: sda1
[ 1102.954344] sd 0:0:0:0: [sda] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=0x00 cmd_age=1s
[ 1102.954376] sd 0:0:0:0: [sda] tag#0 CDB: opcode=0x28 28 00 00 00 00 48 00 00 30 00
[ 1102.954400] blk_update_request: I/O error, dev sda, sector 72 op 0x0:(READ) flags 0x80700 phys_seg 6 prio class 0
[ 1102.954687] sd 0:0:0:0: [sda] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=0x00 cmd_age=0s
[ 1102.954709] sd 0:0:0:0: [sda] tag#0 CDB: opcode=0x28 28 00 00 00 00 88 00 00 78 00
[ 1102.954730] blk_update_request: I/O error, dev sda, sector 136 op 0x0:(READ) flags 0x80700 phys_seg 15 prio class 0
[ 1102.955073] sd 0:0:0:0: [sda] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=0x00 cmd_age=0s
[ 1102.955094] sd 0:0:0:0: [sda] tag#0 CDB: opcode=0x28 28 00 00 00 01 08 00 00 08 00
[ 1102.955115] blk_update_request: I/O error, dev sda, sector 264 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 0
[ 1102.955289] usb 1-1.4: USB disconnect, device number 6
[ 1102.983950] scsi 0:0:0:0: rejecting I/O to dead device
[ 1102.983991] blk_update_request: I/O error, dev sda, sector 72 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[ 1102.984015] Buffer I/O error on dev sda, logical block 9, async page read
[ 1102.984324] Buffer I/O error on dev sda, logical block 2, async page read
[ 1109.553985] usb 1-1.4: new full-speed USB device number 7 using dwc_otg
[ 1109.687686] usb 1-1.4: New USB device found, idVendor=1d50, idProduct=6029, bcdDevice= 1.00
[ 1109.687712] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1109.687728] usb 1-1.4: Product: Marlin USB Device
[ 1109.687742] usb 1-1.4: Manufacturer: marlinfw.org
[ 1109.687757] usb 1-1.4: SerialNumber: 14FFC00BAF38A0685BEDE29EF50020C4
[ 1109.689063] cdc_acm 1-1.4:1.0: ttyACM0: USB ACM device
[ 1109.690813] usb-storage 1-1.4:1.2: USB Mass Storage device detected
[ 1109.691864] scsi host0: usb-storage 1-1.4:1.2
[ 1110.726544] scsi 0:0:0:0: Direct-Access     Marlin   SDCard 01        1.0  PQ: 0 ANSI: 0 CCS
[ 1110.729010] sd 0:0:0:0: [sda] 15431680 512-byte logical blocks: (7.90 GB/7.36 GiB)
[ 1110.731107] sd 0:0:0:0: [sda] Write Protect is off
[ 1110.731130] sd 0:0:0:0: [sda] Mode Sense: 00 00 00 00
[ 1110.731545] sd 0:0:0:0: Attached scsi generic sg0 type 0
[ 1110.733141] sd 0:0:0:0: [sda] Asking for cache data failed
[ 1110.733163] sd 0:0:0:0: [sda] Assuming drive cache: write through
[ 1115.330189]  sda: sda1
[ 1115.338529] sd 0:0:0:0: [sda] Attached SCSI removable disk
[ 1116.073065] sd 0:0:0:0: [sda] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x00 driverbyte=0x08 cmd_age=0s
[ 1116.073083] sd 0:0:0:0: [sda] tag#0 Sense Key : 0x2 [current]
[ 1116.073093] sd 0:0:0:0: [sda] tag#0 ASC=0x3a ASCQ=0x0
[ 1116.073106] sd 0:0:0:0: [sda] tag#0 CDB: opcode=0x28 28 00 00 00 00 40 00 00 08 00
[ 1116.073118] blk_update_request: I/O error, dev sda, sector 64 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 0
[ 1116.074323] sd 0:0:0:0: [sda] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x00 driverbyte=0x08 cmd_age=0s
[ 1116.074340] sd 0:0:0:0: [sda] tag#0 Sense Key : 0x2 [current]
[ 1116.074349] sd 0:0:0:0: [sda] tag#0 ASC=0x3a ASCQ=0x3
[ 1116.074363] sd 0:0:0:0: [sda] tag#0 CDB: opcode=0x28 28 00 00 00 00 40 00 00 08 00
[ 1116.074376] blk_update_request: I/O error, dev sda, sector 64 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[ 1116.074389] Buffer I/O error on dev sda, logical block 8, async page read
`

I've tried a different SD card, thinking that may be the issue, no luck.

benlye, 2022-05-11:
If the plugin is reporting success it must be putting the firmware.bin file _somewhere_.  Both methods are known to work for others so this is something in your setup.  

I would expect the MarlinBFT method to have a higher chance of success as it can be easy to mess up the SD card mounting for the LPC176x method.  The logs don't give any pointers other than to show that the BFT method is successfully transferring the file to the printer and it's being stored somewhere - is there any chance your printer has two SD cards?

I'm afraid you're probably going to have to do more troubleshooting yourself as I have no knowledge of your board.  All I can say is both methods are working well for quite a few people.

Papadoma, 2022-05-21:
I figured it out.
Turning off the option "Reset before flashing" resolved the issue, at least with the LPC176x method. 
Now the board updates successfully every time.

closed 2022-05-21 by Papadoma