OctoPrint/OctoPrint-FirmwareUpdater #314: After update to Octoprint 1.8

mriscoc, 2022-05-26:


benlye, 2022-05-26:
Most likely cause is that filename that the plugin has for the last-flashed file is out of sync with the files on the SD card.  Can you try manually removing all old firmware files from the card and then flashing a couple more times?

If it still doesn't work I need to see the logs from the plugin, spanning a couple of flash attempts.

mriscoc, 2022-05-26:
Hi, I already tried deleting all the bin files, also, I can remove the files manually using the M30 G-code.  The plugin can rename the bin files to FW######.bin.
Maybe there could be a permission access problem to write the old used filename. I will check the logs.

benlye, 2022-05-26:
I think I've found the problem.  Will have a new version for you to test shortly.

benlye, 2022-05-26:
Please try v1.13.3b1 from the development release channel.

mriscoc, 2022-05-26:
It is ok now, thank you.

benlye, 2022-05-27:
v1.13.3 released

closed 2022-05-27 by benlye