OctoPrint/OctoPrint-FirmwareUpdater #315: LONG_FILENAME_WRITE_SUPPORT support

gryzlov, 2022-06-11:
Please, add LONG_FILENAME_WRITE_SUPPORT for FB Reborn updating

benlye, 2022-06-13:
Not 100% sure what you need me to do - just allow long file names in the custom file name option?  I don't know if the underlying MarlinBFT protocol will accept them, and that is outside of my control.

gryzlov, 2022-06-13:
Yes, can you change regexp to support long file names?

benlye, 2022-06-14:
Yes, I can do that.  In the meantime, if you set the custom filename option to anything other than `firmware.bin` the setting will be added to OctoPrint's YAML configuration file, where you can then manually change it to anything you like.  You just won't be able to edit it in the UI anymore.

gryzlov, 2022-06-16:
I removed the code

```
if (self.getProfileSetting("flash_method") == "marlinbft" && self.getProfileSetting("marlinbft_use_custom_filename") && !self.checkMarlinBftCustomFileName()) {
                 alert = gettext("The target filename is invalid. Filename must be 8dot3 format.");
             }
```

and i can confirm that long names (in my case - Robin_nano35.bin) are supported and the firmware is successfully updated.

benlye, 2022-06-16:
Glad you figured it out, but lets keep this open until I've had a chance to add this option.

O1Wcreations, 2022-07-25:
how do i go about updating my ender 6, bother the firmware and the cr touch firmware? not sure if im suppose to use a flash method or not and, just flash the file to the serial port?

benlye, 2022-07-26:
> how do i go about updating my ender 6, bother the firmware and the cr touch firmware? not sure if im suppose to use a flash method or not and, just flash the file to the serial port?

I haven't got a clue - I don't have the board. Your best bet is to find a user group for your printer and ask there.

benlye, 2022-11-10:
Will be addressed in 1.14.0 (available in the devel branch for testing now).

jneilliii, 2023-02-02:
Should probably update the UI side of things so it doesn't appear as error when specifying long filename and `Cap:LFN_WRITE:1` in capabilities report. I just tested the latest beta version and the process worked flawlessly for me now. Thanks so much for integrating this change.

closed 2023-03-20 by benlye