OctoPrint/OctoPrint-FirmwareUpdater #316: Auto-closing issues is harmful and causes friction

ghostsquad, 2022-06-29:
There's a much broader conversation about this happening here: https://github.com/kubernetes/kubernetes/issues/103151

I see issues that appear to be totally valid being closed by a stale bot in this repo for the simple reason of being open without comments for too long. I don't see why an active conversation is a requirement to the legitimacy of a bug report or feature request.

I was just browsing here after installing the plugin, found #309 which mentioned #253 which was closed and locked without resolution. The method of closing/locking issues used here also provides no warning that it's about to happen, essentially completely roadblocking an open issue.

benlye, 2022-06-29:
Issues in this repo are only closed automatically if I ask for more information, help, or input and never receive it. Confirmed bugs or qualified enhancement requests are tagged so that they are ignored by the bot and are kept open until they are fixed or implemented.

There *is* a warning posted 7 days before the issue is closed. It literally tells anyone on the issue to make a comment to keep it open. Furthermore, even a closed issue can still be reopened or commented on once its closed, for some time before it gets locked. If people can't be bothered to come back and comment at that point, why should I keep their issue open or invest my time any further in it? 

I'm not going to let the repo get cluttered up with stale issues where people can't be bothered to respond and I can't do anything without more information or cooperation. It's a policy that's worked pretty well for me so far and I'm quite happy with it. 

#309 will be addressed when I get time. It's tagged as an enhancement so won't be closed until its implemented. You're welcome to contribute to solving it if you want to. I don't have the hardware to test with, so it's a slightly more difficult one for me to address without engaged testers.

closed 2022-06-29 by benlye

ghostsquad, 2022-06-30:
Responding to questions can take awhile, sometimes weeks or months. Personally, I'm part of many projects and communities, and having an issue closed just because I haven't commented on it recently makes it appear as though the issue is not valid.