OctoPrint/OctoPrint-FirmwareUpdater #317: Error in BFT after update Octoprint

mriscoc, 2022-07-04:
After update Octoprint using
```bash
sudo apt upgrade
```

BFT is not working and these errors are written into the log file:
```
2022-07-03 00:25:28,129 - octoprint.plugins.firmwareupdater - INFO - Initializing Marlin BFT protocol
2022-07-03 00:25:28,131 - octoprint.plugins.firmwareupdater - INFO - pySerial Version: 3.5
2022-07-03 00:25:28,148 - octoprint.plugins.firmwareupdater - INFO - Attempting to delete previous firmware file /fw083722.bin
2022-07-03 00:25:28,638 - octoprint.plugins.firmwareupdater - INFO - Connecting to printer at '/dev/ttyUSB0' using Marlin BFT protocol
2022-07-03 00:25:28,639 - octoprint.plugins.firmwareupdater - INFO - Connecting: Switching Marlin to binary protocol...
2022-07-03 00:25:29,388 - octoprint.plugins.firmwareupdater - INFO - Connection synced [0], binary protocol version 0.1.0, 96 byte payload buffer
2022-07-03 00:25:29,389 - octoprint.plugins.firmwareupdater - INFO - Transfering file to printer using Marlin BFT '/tmp/tmpfu3od3fh' -> /fw002529.bin
2022-07-03 00:25:35,244 - octoprint.plugins.firmwareupdater - ERROR - Flashing failed. Unable to transfer file.
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_firmwareupdater/methods/marlinbft.py", line 105, in _flash_marlinbft
    filetransfer.copy(firmware, target, True, False)
  File "/home/pi/oprint/lib/python3.7/site-packages/binproto2/protocols.py", line 422, in copy
    self.write(data[start:end])
  File "/home/pi/oprint/lib/python3.7/site-packages/binproto2/protocols.py", line 378, in write
    self.protocol._send(FileTransferProtocol.protocol_id, FileTransferProtocol.Packet.WRITE, data)
  File "/home/pi/oprint/lib/python3.7/site-packages/binproto2/protocols.py", line 223, in _send
    self._await_response()
  File "/home/pi/oprint/lib/python3.7/site-packages/binproto2/protocols.py", line 243, in _await_response
    switch[token](data)
  File "/home/pi/oprint/lib/python3.7/site-packages/binproto2/protocols.py", line 292, in _response_ok
    raise SynchronizationError()
binproto2.exceptions.SynchronizationError
```

benlye, 2022-07-04:
I think you'll have to take that up with the BFT protocol developer.

mriscoc, 2022-07-04:
Oh sorry, my mistake, I was thinking that I was in his Github.

mriscoc, 2022-07-04:
New attempt

```
2022-07-04 15:19:28,155 - octoprint.plugins.firmwareupdater - INFO - Current Baud: 250000
2022-07-04 15:19:28,156 - octoprint.plugins.firmwareupdater - INFO - Initializing Marlin BFT protocol
2022-07-04 15:19:28,157 - octoprint.plugins.firmwareupdater - INFO - pySerial Version: 3.5
2022-07-04 15:19:28,196 - octoprint.plugins.firmwareupdater - INFO - Attempting to delete previous firmware file /fw083722.bin
2022-07-04 15:19:29,871 - octoprint.plugins.firmwareupdater - INFO - Connecting to printer at '/dev/ttyUSB0' using Marlin BFT protocol
2022-07-04 15:19:29,874 - octoprint.plugins.firmwareupdater - INFO - Connecting: Switching Marlin to binary protocol...
2022-07-04 15:19:29,890 - octoprint.plugins.firmwareupdater - INFO - Connection synced [0], binary protocol version 0.1.0, 96 byte payload buffer
2022-07-04 15:19:29,890 - octoprint.plugins.firmwareupdater - INFO - Transfering file to printer using Marlin BFT '/tmp/tmpxqj7m0t3' -> /fw151929.bin
2022-07-04 15:19:29,900 - octoprint.plugins.firmwareupdater - INFO - File Transfer version: 0.1.0, compression: heatshrink
2022-07-04 15:19:29,951 - octoprint.plugins.firmwareupdater - INFO - Opened file: fw151929.bin
2022-07-04 15:19:30,033 - octoprint.plugins.firmwareupdater - INFO - PROGRESS: 0.00% 9.62KiB/s [11.36KiB/s] Errors: 0
2022-07-04 15:19:30,042 - octoprint.plugins.firmwareupdater - INFO - PROGRESS: 0.05% 10.07KiB/s [11.89KiB/s] Errors: 0
2022-07-04 15:19:30,051 - octoprint.plugins.firmwareupdater - INFO - PROGRESS: 0.11% 10.25KiB/s [12.10KiB/s] Errors: 0
2022-07-04 15:19:30,061 - octoprint.plugins.firmwareupdater - INFO - PROGRESS: 0.16% 10.15KiB/s [11.98KiB/s] Errors: 0
2022-07-04 15:19:30,072 - octoprint.plugins.firmwareupdater - INFO - PROGRESS: 0.21% 9.72KiB/s [11.47KiB/s] Errors: 0
2022-07-04 15:19:30,082 - octoprint.plugins.firmwareupdater - INFO - PROGRESS: 0.26% 9.64KiB/s [11.39KiB/s] Errors: 0
2022-07-04 15:19:30,090 - octoprint.plugins.firmwareupdater - INFO - PROGRESS: 0.32% 9.88KiB/s [11.66KiB/s] Errors: 0
2022-07-04 15:19:30,100 - octoprint.plugins.firmwareupdater - INFO - PROGRESS: 0.37% 9.84KiB/s [11.62KiB/s] Errors: 0
2022-07-04 15:19:30,109 - octoprint.plugins.firmwareupdater - INFO - PROGRESS: 0.42% 9.86KiB/s [11.64KiB/s] Errors: 0
2022-07-04 15:19:30,118 - octoprint.plugins.firmwareupdater - INFO - PROGRESS: 0.47% 9.92KiB/s [11.72KiB/s] Errors: 0
2022-07-04 15:19:30,127 - octoprint.plugins.firmwareupdater - INFO - PROGRESS: 0.53% 9.98KiB/s [11.78KiB/s] Errors: 0
2022-07-04 15:19:30,136 - octoprint.plugins.firmwareupdater - INFO - PROGRESS: 0.58% 10.02KiB/s [11.83KiB/s] Errors: 0
2022-07-04 15:19:30,145 - octoprint.plugins.firmwareupdater - INFO - PROGRESS: 0.63% 10.07KiB/s [11.89KiB/s] Errors: 0
2022-07-04 15:19:30,154 - octoprint.plugins.firmwareupdater - INFO - PROGRESS: 0.68% 10.10KiB/s [11.92KiB/s] Errors: 0
2022-07-04 15:19:30,162 - octoprint.plugins.firmwareupdater - INFO - PROGRESS: 0.74% 10.13KiB/s [11.96KiB/s] Errors: 0
2022-07-04 15:19:30,171 - octoprint.plugins.firmwareupdater - INFO - PROGRESS: 0.79% 10.15KiB/s [11.99KiB/s] Errors: 0
2022-07-04 15:19:30,180 - octoprint.plugins.firmwareupdater - INFO - PROGRESS: 0.84% 10.17KiB/s [12.01KiB/s] Errors: 0
2022-07-04 15:19:30,189 - octoprint.plugins.firmwareupdater - INFO - PROGRESS: 0.89% 10.20KiB/s [12.04KiB/s] Errors: 0
2022-07-04 15:19:30,198 - octoprint.plugins.firmwareupdater - INFO - PROGRESS: 0.95% 10.21KiB/s [12.06KiB/s] Errors: 0
2022-07-04 15:19:30,207 - octoprint.plugins.firmwareupdater - INFO - PROGRESS: 1.00% 10.23KiB/s [12.08KiB/s] Errors: 0
2022-07-04 15:19:30,216 - octoprint.plugins.firmwareupdater - INFO - PROGRESS: 1.05% 10.25KiB/s [12.10KiB/s] Errors: 0
2022-07-04 15:19:30,225 - octoprint.plugins.firmwareupdater - INFO - PROGRESS: 1.10% 10.26KiB/s [12.12KiB/s] Errors: 0
2022-07-04 15:19:30,233 - octoprint.plugins.firmwareupdater - INFO - PROGRESS: 1.16% 10.28KiB/s [12.13KiB/s] Errors: 0
2022-07-04 15:19:30,242 - octoprint.plugins.firmwareupdater - INFO - PROGRESS: 1.21% 10.29KiB/s [12.15KiB/s] Errors: 0
2022-07-04 15:19:30,251 - octoprint.plugins.firmwareupdater - INFO - PROGRESS: 1.26% 10.30KiB/s [12.16KiB/s] Errors: 0
2022-07-04 15:19:30,260 - octoprint.plugins.firmwareupdater - INFO - PROGRESS: 1.31% 10.31KiB/s [12.17KiB/s] Errors: 0
2022-07-04 15:19:30,269 - octoprint.plugins.firmwareupdater - INFO - PROGRESS: 1.37% 10.32KiB/s [12.18KiB/s] Errors: 0
2022-07-04 15:19:30,278 - octoprint.plugins.firmwareupdater - INFO - PROGRESS: 1.42% 10.33KiB/s [12.19KiB/s] Errors: 0
2022-07-04 15:19:30,287 - octoprint.plugins.firmwareupdater - INFO - PROGRESS: 1.47% 10.33KiB/s [12.19KiB/s] Errors: 0
2022-07-04 15:19:30,296 - octoprint.plugins.firmwareupdater - INFO - PROGRESS: 1.52% 10.31KiB/s [12.17KiB/s] Errors: 0
2022-07-04 15:19:30,306 - octoprint.plugins.firmwareupdater - INFO - PROGRESS: 1.58% 10.30KiB/s [12.16KiB/s] Errors: 0
2022-07-04 15:19:30,315 - octoprint.plugins.firmwareupdater - INFO - PROGRESS: 1.63% 10.31KiB/s [12.18KiB/s] Errors: 0
2022-07-04 15:19:30,323 - octoprint.plugins.firmwareupdater - INFO - PROGRESS: 1.68% 10.32KiB/s [12.19KiB/s] Errors: 0
2022-07-04 15:19:30,332 - octoprint.plugins.firmwareupdater - INFO - PROGRESS: 1.73% 10.33KiB/s [12.19KiB/s] Errors: 0
2022-07-04 15:19:30,341 - octoprint.plugins.firmwareupdater - INFO - PROGRESS: 1.79% 10.33KiB/s [12.20KiB/s] Errors: 0
2022-07-04 15:19:30,350 - octoprint.plugins.firmwareupdater - INFO - PROGRESS: 1.84% 10.34KiB/s [12.21KiB/s] Errors: 0
[... cut ...]
    self.protocol._send(FileTransferProtocol.protocol_id, FileTransferProtocol.Packet.WRITE, data)
  File "/home/pi/oprint/lib/python3.7/site-packages/binproto2/protocols.py", line 223, in _send
    self._await_response()
  File "/home/pi/oprint/lib/python3.7/site-packages/binproto2/protocols.py", line 243, in _await_response
    switch[token](data)
  File "/home/pi/oprint/lib/python3.7/site-packages/binproto2/protocols.py", line 292, in _response_ok
    raise SynchronizationError()
binproto2.exceptions.SynchronizationError
```

benlye, 2022-07-04:
That's still a BFT protocol error. This plugin only uses the protocol, it is maintained by someone else.

closed 2022-07-04 by benlye

benlye, 2022-07-04:
For what it's worth, I just updated my OctoPrint installation with with `sudo apt upgrade` and the BFT protocol is still working fine for me.  Whatever problem you have, I don't think the root cause is in the Firmware Updater plugin.

mriscoc, 2022-07-05:
I did a test and init a transfer by hand:
```bash
 python ./transfer.py -p /dev/ttyUSB0 -b250000 -d 512 -c F.bin
```
and was successful, then, I'm lost.
[image]

mriscoc, 2022-07-05:
For reference, I increase "Wait after connect" and "Communication timeout" values:

[image]

And the transfer is ok again.