OctoPrint/OctoPrint-FirmwareUpdater #318: [marlinBFT] Flashing failed, Unable to transfer firmware file to printer

Simonlebucheron, 2022-07-05:
**Hardware Setup**
Creality V4.2.2, Ender3V2, octoprint, raspberrypi 3b1.2

**Software setup**
OctoPrint 1.8.1, octopi_version: 0.18.0
Experimented with different firmwares : marlin release 2.1,  Marlin bugfix-2.1.x
VS code : 
default_envs = STM32F103RE_creality_xfer or STM32F103RE_creality
Tested with commented and uncommented #define CUSTOM_FIRMWARE_UPLOAD

**Describe the problem**
Message "Flashing failed Unable to transfer firmware file to printer" when flashing with marlin BFT method.
The octoprint.log tell us : `Exception: Cannot open file on client`
What does it mean ?

**Log Files**
octoprint.log:
[octoprint(5).log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/9045612/octoprint.5.log)

plugin_firmwareupdater_console.log
Nothing in it (why ?)

**Additional context**
I tried different configurations over two days, different firmwares, and i finally make a clean install of octoprint 1.8.1 on a new sdcard but non results.
Since it is the only flashing method which seams to be working with de V4.2.2 creality board, I would love to have this feature so I would be happy to help figuring out a solution.

benlye, 2022-07-05:
Can you try another client / browser?

Simonlebucheron, 2022-07-05:
> Can you try another client / browser?

I just test with firefox, chrome and microsoft edge but with no results, that was you mean ?
 

> Also, can you enable verbose BFT protocol logging and upload the resulting octoprint.log file? The console log is empty because this method doesn't run any external commands.

How should I do that ?

Simonlebucheron, 2022-07-05:
Soory, I forgot I have seen verbose BTP protocol logging in the documentation and I already activate it earlier. 
I attached a new full log after it clean the old one from octoprint :
[octoprint(7).log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/9047415/octoprint.7.log)

What is : `2022-07-05 16:56:26,512 - octoprint.plugins.firmwareupdater - INFO - Setting maximum upload size for /flash to 5243904` at the beginning of the log?

Simonlebucheron, 2022-07-05:
> Do you have a separate computer you can try from?

I tried from an another computer (windows 10 too), same thing in the log "Cannot open file on client"

benlye, 2022-07-05:
It's as if the file really doesn't exist on your Pi, or something (like permissions) is stopping OctoPi from reading it.  I've never seen this before.

`2022-07-05 16:56:26,512 - octoprint.plugins.firmwareupdater - INFO - Setting maximum upload size for /flash to 5243904`

Is saying that they largest file the server will accept from the browser is 5,243,904 Bytes, or about 5MB.  We can increase that 5MB should be plenty.  What is the size of the firmware file you're trying to upload?

benlye, 2022-07-05:
Actually the error is misleading / badly worded.  `Cannot open file on client` really seems to mean that the protocol cannot open the target file (I thought it meant the source file).

I can only guess that there is something in your running firmware which is causing the BFT protocol to fail.

benlye, 2022-07-05:
Another thought - I've got a Creality v4.2.2 board for testing.  Do you have the bin file for the firmware that your printer is running?  If you upload the file I could try loading it on my board to see what happens when I try to use BFT to load a new file to it.

Simonlebucheron, 2022-07-05:
Here is my last firmware 
[firmware-20220705-120042.zip](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/9048062/firmware-20220705-120042.zip)

In parallel, Ill try with different location of the file, like directly one the raspberry.

Simonlebucheron, 2022-07-05:
> In parallel, Ill try with different location of the file, like directly one the raspberry.

I do not know how to select a local file with the plugin, maybe it is not even possible ?
I tried several names, with no results.

Sould we suspect the network between the file and the raspberry ?

benlye, 2022-07-06:
> Here is my last firmware [firmware-20220705-120042.zip](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/9048062/firmware-20220705-120042.zip)
> 

I can confirm that with this FW installed on the board I cannot use BFT to transfer any further FW updates, so the problem seems to be something in this image which is rendering the BFT protocol unusable.

What's interesting is that the code which is supposed to delete the previous FW file is also not working - the file remains on the SD card.  Furthermore, no files can be removed from the SD card with the `M30` command.  It's as if the SD card is read-only.

```
end: M20
Recv: Begin file list
Recv: FOO.GCO 0
Recv: BOO~1.GCO 0
Recv: End file list
Recv: ok
[...]
Send: M30 FOO.GCO
Recv: Deletion failed, File: FOO.GCO.
Recv: ok
```

Compared with the same command on a firmware I compiled from the bugfix-2.0.x branch this morning:
```
Send: M20
Recv: Begin file list
Recv: FOO.GCO 0
Recv: BOO~1.GCO 0
Recv: End file list
Recv: ok
[...]
Send: M30 FOO.GCO
Recv: File deleted:FOO.GCO
Recv: ok
```

So I think your problem is in your firmware, but I don't know what is causing it.  I noticed you have `CUSTOM_FIRMWARE_UPLOAD` enabled so I tried that too but it does not cause the same problem (at least not on its own).

If you compiled this firmware yourself you need to take a look at the settings you have changed and see if you can figure out where the problem is.

Simonlebucheron, 2022-07-06:
Indeed, that is interesting, thank you for your investigation.

> So I think your problem is in your firmware, but I don't know what is causing it. I noticed you have `CUSTOM_FIRMWARE_UPLOAD` enabled so I tried that too but it does not cause the same problem (at least not on its own).

It was just a test after reading some threads on the binary file transfer issues, I will disable it from now.

> If you compiled this firmware yourself you need to take a look at the settings you have changed and see if you can figure out where the problem is.

Yes I have, with VScode but nothing fancy I think, I will look further in this way and try to compile the bugfix 2.0.x, but my primary goal was to update my printer from and old `bugfix-2.0.x` to `2.1` release.

By the way, in the documentation for [marlin BFT flashing](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/blob/master/doc/marlinbft.md) it refers only to `bugfix-2.0.x` branch, can you try compiling from your side with `bugfix-2.1.x` which is the new default branch, or `2.1.x` ?

benlye, 2022-07-06:
> By the way, in the documentation for [marlin BFT flashing](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/blob/master/doc/marlinbft.md) it refers only to bugfix-2.0.x branch, can you try compiling from your side with bugfix-2.1.x which is the new default branch, or 2.1.x ?

Yes, I'm running bugfix-2.1.x on my own printer and BFT works fine.  My board is a BTT SKR Turbo, but I'll try 2.1.x on the Creality board too when I get a chance.

benlye, 2022-07-06:
I can confirm that firmware I just compiled from the current bugfix-2.1.x branch for my Creality v4.2.2 board also has this problem (the SD card seems to be read only - files cannot be deleted).

Firmware I just compiled from today's bugfix-2.1.x branch for my SKR Turbo does *not* have the problem - I can read and write the SD card.

So this looks like a bug or configuration issue in Marlin's bugfix-2.1.x branch which affects the Creality board (maybe others).

Simonlebucheron, 2022-07-06:
I have just recompiled a fresh new FW from latest `bugfix-2.0.x` branch without any changes in configuration.h and configuration_adv.h except `#define SDCARD_CONNECTION ONBOARD` and `#define BINARY_FILE_TRANSFER` and BFT is not working (same error).

> Compared with the same command on a firmware I compiled from the bugfix-2.0.x branch this morning:

Since you tested it this morning with the same board, I do not understand why it is not working for me. Any ideas ?
I will test an "old" 2.0.x bugfix branch.

benlye, 2022-07-06:
My initial tests were with an older bugfix-2.0.x branch.  Reading through the Marlin issue, it sounds like there is a 'new' bugfix-2.0.x since 2022-02-22, and builds up to this date might be OK.  

I haven't been following Marlin development too closely so don't know what the story is with 'new' and 'old' bugfix-2.0.x but I did see when I tried to update one of my branches that there were a *lot* of changes at some point in the past few months.  It sounds like there was a complete SD IO rewrite somewhere in there.

Simonlebucheron, 2022-07-06:
Done [here](https://github.com/MarlinFirmware/Marlin/issues/24422#issuecomment-1176487903) ;)
Waiting to find a solution maybe you could update the documentation with a warning for the creality V4.2.2 owners to use only [Marlin 2.0.9.3](https://github.com/MarlinFirmware/Marlin/releases/tag/2.0.9.3) ?

mriscoc, 2022-07-08:
I arrived here later, I also had the same error with the latest Marlin Bugfix 2.1, maybe because of the refactoring of the SDIO module.

Simonlebucheron, 2022-07-09:
Hum, I tried the again with the new bugfix 2.1.x but BFT failed with the same error, no file written on the sd card... I will try later after getting some sleep...

mriscoc, 2022-07-10:
For an Ender3V2 with 422 (STM32F1) latest SDIO commit in Bugfix solves the issue.

Simonlebucheron, 2022-07-10:
It is working, I think it was a mess in my VStudio files. Thank you for all @benlye , this plugin is very useful and time saving.

closed 2022-07-10 by Simonlebucheron