OctoPrint/OctoPrint-FirmwareUpdater #319: Updating firmware on Ender 7 fails

Mech131, 2022-07-08:
**Hardware Setup**
Ender 7

**Describe the problem**
Programmer timeout.

**Log Files**
[octoprint.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/9069039/octoprint.log)
[plugin_firmwareupdater_console.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/9069040/plugin_firmwareupdater_console.log)
[octoprint-systeminfo-20220708000635.zip](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/9069041/octoprint-systeminfo-20220708000635.zip)

**Additional context**
I tired a couple of other configurations and it also failed.

benlye, 2022-07-08:
I don't know the board at all, but a quick Google search says that it uses an STM32 MCU on a board which is similar to the Creality v4.2.7 board in the Ender 3.  Presumably, like other Ender boards, it is updated by copying a firmware file to the SD card. 

The avrdude method you've been trying is not going to work for you.  Most likely you need to use the MarlinBFT flash method, which means you need to install firmware with the Binary File Transfer protocol enabled _before_ you can use this plugin.  Instructions for using this method are [here](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/blob/master/doc/marlinbft.md).

If you need further help, I suggest you find an Ender user group and ask there - I don't have the printer or the board so am very limited in what I can do for you.

Mech131, 2022-07-08:
@benlye So my main problem is that I installed InsanityAutomations firmware, but it broke the mSD Card slot. The SD Card fails to initiate on boot. So i am unable to flash my firmware by simply dropping the firmware on the mSD card and updating from there. They recommended using this plugin to flash my firmware, but didn't provide any instructions on how to flash it. I'm fairly tech savy, so i'm able to install whatever dependencies are needed and work my way though the update. But i cannot make any changes to the firmware.

benlye, 2022-07-08:
If the Binary File Transfer protocol is not enabled in the firmware you are running, and you cannot change the firmware then this plugin cannot help you. Sorry.

I don't know anything about the board so can't say whether or not there is any way to update it via the USB port.

closed 2022-07-28 by github-actions[bot]