OctoPrint/OctoPrint-FirmwareUpdater #321: Instructions need updating for Python 3.9

salfter, 2022-08-03:
**Hardware Setup**
OctoPrint host: Raspberry Pi Compute Module 4 on a Waveshare CM4-NANO-A
printer: AM8 with SKR 1.4 Turbo

**Describe the problem**
Python 3.9 breaks the version of Heatshrink that the instructions call for installing.

**Resolution**

https://github.com/p3p/pyheatshrink hasn't been updated yet, but https://github.com/Thynix/pyheatshrink has.  It has some submodules within, though, and has no releases, so we can't just grab a tarball from GitHub.  You can make an installable tarball with this:

```
(cd /tmp && docker run "-v${PWD}:/a" devsisters/tarballize https://github.com/Thynix/pyheatshrink master) && sudo chown `whoami` /tmp/master.tar.xz && mv /tmp/master.tar.xz pyheatshrink-master.tar.xz
```

If you'd rather not do the above, I've created a tarball and hosted it on my website at https://alfter.us/wp-content/uploads/2022/08/pyheatshrink-92ffa4e.tar.xz.

Once you have the tarball, move it over to your OctoPrint host (if it isn't already there), install some prerequisites, and install the updated Heatshrink in your virtual environment:

```
sudo apt install python3-dev && octoprint/bin/pip install pyheatshrink-master.tar.xz
```

You should then be able to continue as usual: install marlin-binary-protocol, etc.

draeath, 2022-09-30:
What occurs if you are using Python 3.9 and this hasn't been fixed?

I ask as I am able to install the plugin on 3.9.13 (aarch64) without any apparent issue. I have not, however, attempted to utilize it as my printer's firmware is current already.

Slurpgeit, 2022-10-01:
On 64-bit raspbian it will fail to compile heatshrink and fail the marlin-binary-protocol package.

@salfter thank you for this workaround!

+1 for fixing this issue ;)

SANSd20, 2022-12-13:
using 3.10 i get this error:

```
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> Heatshrink
```

JWSmythe, 2023-02-18:
+1 for adding this to the instructions.  I've been chasing this problem around since yesterday, and constantly running into the problem where heatshrink wouldn't install, so marlin-binary-protocol wouldn't install!  This worked! 

These were the commands I did.  Results are irrelevant, since they worked as expected.
```
su pi 
cd ~
wget https://alfter.us/wp-content/uploads/2022/08/pyheatshrink-92ffa4e.tar.xz
~/oprint/bin/pip install pyheatshrink-92ffa4e.tar.xz
~/oprint/bin/pip install marlin-binary-protocol
```
I opted to use his tar, rather than making my own.  Least effort required.    I tried with a fresh tar from github, but it fails just like the normal install.  Maybe you should save his tar here, and give the alt instructions with that link.  That would have saved me a day of trying to figure it out.

This board is a new install from 3 days ago, on new hardware.  This is the only thing I've run into problems with.   I have two other boards running OctoPrint and Firmware updater, but I did those in the past.  They run on RPi/Raspbian and Rock64/Slackware(Slarm64).

System info:
```
OctoPrint 1.8.6
Firmware Updater 1.13.3
~pi/oprint/bin/python 3.9.2
~pi/oprint/bin/pip 22.3.1

Hardware: Orange Pi Zero2
OS: Debian 11 - Bullseye.
uname -a: Linux octoprint-ender3 5.16.17-sun50iw9 #3.0.6 SMP Tue Aug 9 13:51:16 CST 2022 aarch64 GNU/Linux
/etc/os-release: PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"

OS Img: Obico firmware, which includes Octoprint preinstalled.

Image install info: https://www.obico.io/docs/user-guides/install-octoprint-orange-pi-zero-2-kit/#1-download-the-octoprint-for-orange-pi-software
```

And, thanks for the cool plugin.  It has saved me a lot of walking to/from the machines.

jneilliii, 2023-04-24:
Curious if anyone's tried this module fork?

https://github.com/jaketri/marlin-binary-protocol

NeoMatrixJR, 2023-07-06:
created PR for this: https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/pull/349

PythonesqueSpam, 2024-01-24:
Could this be the same issue trying to install on my BTT SKR 3 EZ?

The instructions say I have to install heatshrink from within the virtual environment for octoprint but doesn't explain what that means or how to do it.  I assumed it means run it from within the hidden .octoprint directory so I ran this command:
 ~/.octoprint $ sudo ../venv/bin/pip3 install https://github.com/p3p/pyheatshrink/releases/download/0.3.3
/pyheatshrink-pip.zip

Any idea what this resulting load of nonsense means?

Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/p3p/pyheatshrink/releases/download/0.3.3/pyheatshrink-pip.zip
  Downloading https://github.com/p3p/pyheatshrink/releases/download/0.3.3/pyheatshrink-pip.zip (2.5 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 2.5/2.5 MB 720.4 kB/s eta 0:00:00
  Preparing metadata (setup.py) ... done
Building wheels for collected packages: Heatshrink
  Building wheel for Heatshrink (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [21 lines of output]
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build/lib.linux-aarch64-cpython-311
      creating build/lib.linux-aarch64-cpython-311/heatshrink
      copying heatshrink/__init__.py -> build/lib.linux-aarch64-cpython-311/heatshrink
      copying heatshrink/streams.py -> build/lib.linux-aarch64-cpython-311/heatshrink
      running build_ext
      building 'heatshrink.core' extension
      creating build/temp.linux-aarch64-cpython-311
      creating build/temp.linux-aarch64-cpython-311/heatshrink
      creating build/temp.linux-aarch64-cpython-311/heatshrink/_heatshrink
      aarch64-linux-gnu-gcc -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I. -I./heatshrink/_heatshrink -I/home/pytho/venv/include -I/usr/include/python3.11 -c heatshrink/_heatshrink/heatshrink_decoder.c -o build/temp.linux-aarch64-cpython-311/heatshrink/_heatshrink/heatshrink_decoder.o -std=c99
      aarch64-linux-gnu-gcc -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I. -I./heatshrink/_heatshrink -I/home/pytho/venv/include -I/usr/include/python3.11 -c heatshrink/_heatshrink/heatshrink_encoder.c -o build/temp.linux-aarch64-cpython-311/heatshrink/_heatshrink/heatshrink_encoder.o -std=c99
      aarch64-linux-gnu-gcc -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I. -I./heatshrink/_heatshrink -I/home/pytho/venv/include -I/usr/include/python3.11 -c heatshrink/core.c -o build/temp.linux-aarch64-cpython-311/heatshrink/core.o -std=c99
      heatshrink/core.c:196:12: fatal error: longintrepr.h: No such file or directory
        196 |   #include "longintrepr.h"
            |            ^~~~~~~~~~~~~~~
      compilation terminated.
      error: command '/usr/bin/aarch64-linux-gnu-gcc' failed with exit code 1
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for Heatshrink
  Running setup.py clean for Heatshrink
Failed to build Heatshrink
ERROR: Could not build wheels for Heatshrink, which is required to install pyproject.toml-based projects

draeath, 2024-01-24:
You are on the right track but that's unrelated to this issue.

That looks like an incompatibility with Python 3.11 (that include used to be available in older releases of python afaik)

stablestud, 2024-05-13:
Hmm sadly MBT flashing is still broken.    
When running `pip3 install marlin-binary-protocol` it installs just fine, but is not detected by `FirmwareUpdater`,    
and when using the Python 3 method from the documention it fails due to incompatability of `heatshrink` with Python 3.9

Is there any offical solution to this?

jneilliii, 2024-05-14:
> When running `pip3 install marlin-binary-protocol` it installs just fine, but is not detected by `FirmwareUpdater`,

Did you activate OctoPrint's venv before running this command? 

`source ~/oprint/bin/activate`

I think @The-EG may have a solution, I vaguely remember seeing a post on discord.

The-EG, 2024-05-16:
`heatshrink` is not compatible with newer versions of Python, but `heatshrink2` is and functions as a drop in replacement. `marlin-binary-protocol` erroneously depends on both `heatshrink` and `heatshrink2`. Removing the dependency to `heatshrink` will allow `marlin-binary-protocol` to be installed and function properly.

You can do that yourself in a local copy, or you can install a fixed up version of `marlin-binary-protocol` from my fork: 
```terminal
 $ pip install https://github.com/The-EG/marlin-binary-protocol/archive/refs/heads/master.zip
```

(Of course, make sure that you do have your OctoPrint venv activated as jneilliii said)

benlye, 2024-06-22:
I've _finally_ updated the doc for this. Big thanks to @The-EG for supplying the patched package.

closed 2024-06-22 by benlye