OctoPrint/OctoPrint-FirmwareUpdater #323: incompatibility with werkzeug 2.1+ and room for improvement in catching exceptions

arekm, 2022-08-16:
**Hardware Setup**
prusa mk3s+

**Describe the problem**
Plugin worked fine until I tried it on fresh system using octoprint from maintenance branch.

* Clicking on "flash from file"
* Progress bar shows "Flashing firmware..."
* Nothing happens

Browser inspector shows me that POST call to /flash plugin endpoint is being made and server returns: 
```400 Bad Request: The browser (or proxy) sent a request that this server could not understand.```

So that's the **first problem** - javascript part of flash plugin doesn't react correctly if endpoint returns any error. It should display some error to user in such case.

* Nothing in octoprint log. Turns out that any exception in ```flash_firmware(self)``` (that's not catched inside) is not logged by octoprint. Don't know why.

To figure out what's happening I'm putting entire flash_firmware content in 

```
try:
...
        except Exception as e:
            self._logger.exception(e)
```

block and now we have something:

```
2022-08-15 22:20:34,659 - octoprint.plugins.firmwareupdater - ERROR - 400 Bad Request: The browser (or proxy) sent a request that this server could not understand.
Traceback (most recent call last):
  File "/home/pi/devel/OctoPrint/venv/lib/python3.9/site-packages/flask/wrappers.py", line 128, in on_json_loading_failed
    return super().on_json_loading_failed(e)
  File "/home/pi/devel/OctoPrint/venv/lib/python3.9/site-packages/werkzeug/wrappers/request.py", line 611, in on_json_loading_failed
    raise BadRequest(
werkzeug.exceptions.BadRequest: 400 Bad Request: Did not attempt to load JSON data because the request Content-Type was not 'application/json'.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/pi/devel/OctoPrint/venv/lib/python3.9/site-packages/octoprint_firmwareupdater/__init__.py", line 100, in flash_firmware
    value_source = flask.request.json if flask.request.json else flask.request.values
  File "/home/pi/devel/OctoPrint/venv/lib/python3.9/site-packages/werkzeug/wrappers/request.py", line 540, in json
    return self.get_json()
  File "/home/pi/devel/OctoPrint/venv/lib/python3.9/site-packages/werkzeug/wrappers/request.py", line 571, in get_json
    return self.on_json_loading_failed(None)
  File "/home/pi/devel/OctoPrint/venv/lib/python3.9/site-packages/flask/wrappers.py", line 133, in on_json_loading_failed
    raise BadRequest() from e
werkzeug.exceptions.BadRequest: 400 Bad Request: The browser (or proxy) sent a request that this server could not understand.
2022-08-15 22:20:34,661 - octoprint - ERROR - Exception on /plugin/firmwareupdater/flash [POST]
Traceback (most recent call last):
  File "/home/pi/devel/OctoPrint/venv/lib/python3.9/site-packages/flask/app.py", line 2525, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/devel/OctoPrint/venv/lib/python3.9/site-packages/flask/app.py", line 1823, in full_dispatch_request
    return self.finalize_request(rv)
  File "/home/pi/devel/OctoPrint/venv/lib/python3.9/site-packages/flask/app.py", line 1842, in finalize_request
    response = self.make_response(rv)
  File "/home/pi/devel/OctoPrint/venv/lib/python3.9/site-packages/flask/app.py", line 2134, in make_response
    raise TypeError(
TypeError: The view function for 'plugin.firmwareupdater.flash_firmware' did not return a valid response. The function either returned None or ended without a return statement.
```

werkzeug 2.1+ has a change which will raise BadRequest exception if there are json calls while Content-Type is not application/json. For firmwareupdater part Content-Type is multipart/form-data if I remember correctly.

https://github.com/pallets/werkzeug/issues/2339

Solution for this is doing

```
import werkzeug

[...]

            try:
                value_source = flask.request.json if flask.request.json else flask.request.values
            except werkzeug.exceptions.BadRequest:
                value_source = flask.request.values
```

instead of current

```
          value_source = flask.request.json if flask.request.json else flask.request.values
```

So that was **second problem** - incompatibility with werkzeug 2.1+. With above change flashing succeeds.

And **last thing**, entire _flash_worker() should also sit in try except block since any exception there (that's not already catched inside) will be silently ignored without any trace in octoprint.log (as this is run in Thread and octoprint doesn't get exceptions from separate thread).

About octoprint not catching and logging exception in flash_firmware I'm not sure. IMO it should catch it and log it since it is not run in separate thread (I think).

benlye, 2022-08-16:
Feel free to submit pull requests for any improvements.

arekm, 2022-08-30:
The problem is still there but issue can be closed. There is a open pull request fixing it.

closed 2022-08-30 by arekm