OctoPrint/OctoPrint-FirmwareUpdater #331: Already Flashing

jaymedavis, 2023-01-15:
Hello there!  Thanks for the tool, very handy.

I started flashing my prusa mk3, forgetting I had a slice engineering mosquito magnum.  The flash did not complete succesfully.  Trying to flash again gives me the message

**```
Flashing failed
Already flashing.
```**

can you tell me how to cancel this and start over so I can send the correct firmware from slice engineering?

Here is my log:
```
2023-01-15 16:37:43,510 
2023-01-15 16:37:43,521 /usr/bin/avrdude -v -q -p m2560 -c wiring -P /dev/ttyS0 -D -U flash:w:/tmp/tmp3tg0ofrz:i
2023-01-15 16:37:44,212 > 
2023-01-15 16:37:44,213 > avrdude: Version 6.3-20171130
2023-01-15 16:37:44,213 >          Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
2023-01-15 16:37:44,213 >          Copyright (c) 2007-2014 Joerg Wunsch
2023-01-15 16:37:44,213 > 
2023-01-15 16:37:44,214 >          System wide configuration file is "/etc/avrdude.conf"
2023-01-15 16:37:44,214 >          User configuration file is "/home/pi/.avrduderc"
2023-01-15 16:37:44,214 >          User configuration file does not exist or is not a regular file, skipping
2023-01-15 16:37:44,215 > 
2023-01-15 16:37:44,215 >          Using Port                    : /dev/ttyS0
2023-01-15 16:37:44,215 >          Using Programmer              : wiring
2023-01-15 16:37:44,215 > 
2023-01-15 18:39:14,827 
2023-01-15 18:39:14,829 /usr/bin/avrdude -v -q -p m2560 -c wiring -P /dev/ttyS0 -D -U flash:w:/tmp/tmpaipvo3d2:i
2023-01-15 18:39:15,510 > 
2023-01-15 18:39:15,511 > avrdude: Version 6.3-20171130
2023-01-15 18:39:15,511 >          Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
2023-01-15 18:39:15,511 >          Copyright (c) 2007-2014 Joerg Wunsch
2023-01-15 18:39:15,512 > 
2023-01-15 18:39:15,512 >          System wide configuration file is "/etc/avrdude.conf"
2023-01-15 18:39:15,512 >          User configuration file is "/home/pi/.avrduderc"
2023-01-15 18:39:15,512 >          User configuration file does not exist or is not a regular file, skipping
2023-01-15 18:39:15,513 > 
2023-01-15 18:39:15,513 >          Using Port                    : /dev/ttyS0
2023-01-15 18:39:15,513 >          Using Programmer              : wiring
2023-01-15 18:39:15,513 > 
2023-01-15 20:56:02,994 
2023-01-15 20:56:02,996 /usr/bin/avrdude -v -q -p m2560 -c wiring -P /dev/ttyAMA0 -D -U flash:w:/tmp/tmpvd2fu7ca:i
2023-01-15 20:56:03,641 > 
2023-01-15 20:56:03,642 > avrdude: Version 6.3-20171130
2023-01-15 20:56:03,642 >          Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
2023-01-15 20:56:03,643 >          Copyright (c) 2007-2014 Joerg Wunsch
2023-01-15 20:56:03,643 > 
2023-01-15 20:56:03,644 >          System wide configuration file is "/etc/avrdude.conf"
2023-01-15 20:56:03,645 >          User configuration file is "/home/pi/.avrduderc"
2023-01-15 20:56:03,645 >          User configuration file does not exist or is not a regular file, skipping
2023-01-15 20:56:03,646 > 
2023-01-15 20:56:03,646 >          Using Port                    : /dev/ttyAMA0
2023-01-15 20:56:03,647 >          Using Programmer              : wiring
2023-01-15 20:56:03,647 > 
2023-01-15 20:56:08,886 > avrdude: stk500v2_ReceiveMessage(): timeout
2023-01-15 20:56:08,887 > 
2023-01-15 20:56:13,115 > avrdude: stk500v2_ReceiveMessage(): timeout
2023-01-15 20:56:13,116 > 
2023-01-15 20:56:17,348 > avrdude: stk500v2_ReceiveMessage(): timeout
2023-01-15 20:56:17,349 > 
2023-01-15 20:56:21,572 > avrdude: stk500v2_ReceiveMessage(): timeout
2023-01-15 20:56:21,573 > 
2023-01-15 20:56:25,801 > avrdude: stk500v2_ReceiveMessage(): timeout
2023-01-15 20:56:25,802 > 
2023-01-15 20:56:30,029 > avrdude: stk500v2_ReceiveMessage(): timeout
2023-01-15 20:56:30,030 > avrdude: stk500v2_getsync(): timeout communicating with programmer
2023-01-15 20:56:30,031 > 
2023-01-15 20:56:30,031 > avrdude done.  Thank you.
2023-01-15 20:56:30,032 > 
2023-01-15 20:56:30,032 > 
2023-01-15 20:59:44,739 
2023-01-15 20:59:44,743 /usr/bin/avrdude -v -q -p m2560 -c wiring -P /dev/ttyS0 -D -U flash:w:/tmp/tmpicusswm5:i
2023-01-15 20:59:45,329 > 
2023-01-15 20:59:45,330 > avrdude: Version 6.3-20171130
2023-01-15 20:59:45,330 >          Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
2023-01-15 20:59:45,331 >          Copyright (c) 2007-2014 Joerg Wunsch
2023-01-15 20:59:45,331 > 
2023-01-15 20:59:45,331 >          System wide configuration file is "/etc/avrdude.conf"
2023-01-15 20:59:45,331 >          User configuration file is "/home/pi/.avrduderc"
2023-01-15 20:59:45,332 >          User configuration file does not exist or is not a regular file, skipping
2023-01-15 20:59:45,332 > 
2023-01-15 20:59:45,332 >          Using Port                    : /dev/ttyS0
2023-01-15 20:59:45,333 >          Using Programmer              : wiring
2023-01-15 20:59:45,333 > 
```

benlye, 2023-01-16:
You probably need to restart OctoPrint (or restart your Raspberry Pi) to clear the 'already flashing' error.

jaymedavis, 2023-01-16:
Unfortunately, that didn't help.  I tried several times.

benlye, 2023-01-16:
Can you post the octoprint log as an attachment? The log above is the firmware updater console log but there will be more info in OctoPrint's log.

benlye, 2023-01-16:
Can you add it **as an attachment**?

benlye, 2023-01-16:
Here's what I see:
1. You tried to flash a few times but the process errored with an avrdude comms error, then you tried to flash but for some reason it didn't error or end:
```
2023-01-15 20:59:44,462 - octoprint.plugins.firmwareupdater - INFO - Firmware update profile index: 0
2023-01-15 20:59:44,472 - octoprint.plugins.firmwareupdater - INFO - Firmware update profile name: Default
2023-01-15 20:59:44,473 - octoprint.plugins.firmwareupdater - INFO - Printer port: /dev/ttyS0
2023-01-15 20:59:44,491 - octoprint.plugins.firmwareupdater - INFO - Flash method: avrdude
2023-01-15 20:59:44,691 - octoprint.plugins.firmwareupdater - INFO - Firmware update started
2023-01-15 20:59:44,739 - octoprint.plugins.firmwareupdater - INFO - Running '/usr/bin/avrdude -v -q -p m2560 -c wiring -P /dev/ttyS0 -D -U flash:w:/tmp/tmpicusswm5:i' in /usr/bin
```

Now every time you try the process stops but it's unclear why.  Since the attempt logged above I don't see any restarts of OctoPrint in the log files.  So, can you try to restart OctoPrint once more, please - maybe just reboot the Pi that it's running on.

Once that's done, try to flash again.  If you get an error, attach the console log and the octoprint log and I'll see if I can figure out what's going on.

jaymedavis, 2023-01-17:
Sure, thanks for the help.

Here's the two files again.  There will be two more attempts:  first was restarting octoprint, second was rebooting the system.  Here is my settings as well:

[plugin_firmwareupdater_console.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/10430385/plugin_firmwareupdater_console.log)
[octoprint.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/10430386/octoprint.log)
[image]

benlye, 2023-01-17:
Right, so you're no longer getting the 'Already flashing' error, are you?  Now you get the error from avrdude?

```
2023-01-17 01:33:46,955 - octoprint.plugins.firmwareupdater - ERROR - Flashing failed. Timeout communicating with programmer.
```

That means that your configuration is not correct for your board.  I can't help you with that as I don't have the board/printer and can't tell you what settings will work.  Your best bet is to find a user community for your board / setup and ask there.

jaymedavis, 2023-01-17:
Yeah, it is a different error now, at least.  It's the prusa mk3.  I have followed https://mattwidmann.net/notes/updating-prusa-i3-mk3s-firmware-with-octoprint/ and https://blog.svenadolph.net/update-prusa-i3-m-k3-s-firmware-from-octo-pi-using-the-octo-print-firmware-updater-plugin/ to verify those settings. 

I think I'll have to flash it manually.  Thanks again for looking.

closed 2023-01-17 by jaymedavis