OctoPrint/OctoPrint-FirmwareUpdater #333: Unable to transfer Firmware file to printer

dlopez408, 2023-01-25:
**Hardware Setup**
Please board type and printer make and model.

**Describe the problem**
Please provide a clear and concise description of what the problem is, including details of any error messages you receive (screenshots are helpful).

**Log Files**
Attach the OctoPrint log and the Firmware Updater log Log files, which can be accessed under **Settings** -> **Logging** in OctoPrint, or in `~/.octoprint/logs/` on the host.  Both `octoprint.log` and `~/.octoprint/logs/plugin_firmwareupdater_console.log` should be included.

**Additional context**
Add any other context about the problem here.

w7boz, 2023-01-26:
This may be totally wrong. But have you tried a different baud rate? Maybe 115200?

benlye, 2023-01-27:
```
2023-01-25 14:40:05,776 - octoprint.plugins.firmwareupdater - INFO - Transfering file to printer using Marlin BFT '/tmp/tmpuzcyoxo2' -> /firmware.bin
2023-01-25 14:40:05,818 - octoprint.plugins.firmwareupdater - ERROR - Flashing failed. Unable to transfer file.
Traceback (most recent call last):
  File "/home/dlo/OctoPrint/lib/python3.8/site-packages/octoprint_firmwareupdater/methods/marlinbft.py", line 105, in _flash_marlinbft
    filetransfer.copy(firmware, target, True, False)
  File "/home/dlo/OctoPrint/lib/python3.8/site-packages/binproto2/protocols.py", line 407, in copy
    self.open(dest_filename, compression_support, dummy)
  File "/home/dlo/OctoPrint/lib/python3.8/site-packages/binproto2/protocols.py", line 372, in open
    raise Exception("Cannot open file on client")
Exception: Cannot open file on client
```

I don't think it's a baud rate problem - the BFT protocol is connected and synced. `Cannot open file on client` means that the protocol was not able to create the target file on your printer.  

Try going into the plugins options and enabling the `Verbose file transfer progress logging` option (it's on the 'Advanced 2' tab.  Try to flash again and then upload the log.  It may not tell us exactly what's going on, but it's worth trying.

Otherwise, it's possibly a combination of the SD card settings in your Marlin config.

dlopez408, 2023-02-10:
Thanks for the replies! I will update with the new log soon

dlopez408, 2023-02-10:
Here is the log with the `Verbose file transfer progress logging` option enabled

[octoprint verbose.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/10710401/octoprint.verbose.log)

benlye, 2023-02-19:
Not much I can do to help with this - the protocol can't open the destination file on your printer. The binary file protocol is written and maintained by someone else.

closed 2023-03-11 by github-actions[bot]