OctoPrint/OctoPrint-FirmwareUpdater #334: Feature request  backup eeprom when updating mk3, restore also possible if downgrading

cnlson, 2023-01-30:
Prusa MK3 eeprom backup and restore is available via gcode commands IE backup when flashing firmware and save the backup either to the printers sd or the pi's 
save it with the prior firmware name and only allow it to be restored if flashing that firmware back on.

when trying new beta firmware it is hard to get volunteers if you could possibly lose sheet settings, live z, and calibration. adding this feature would allow for easily returning to a stable firmware after testing a beta. 

For instance on 3.11 now flashing to 3.13 beta

1. read eeprom (commands provided https://github.com/prusa3d/Prusa-Firmware/pull/3957#issuecomment-1406799876 ) 
2. save eeprom with a naming format of  "3.11 backup" 
3. if reflashing to 3.11 offer to restore backup "3.11 backup"

step 3 could be manual but would want to check that the firmware and backup match to avoid restoring settings to non referencil locations. 

@wavexx in that thread can provide the breakdown of gcode commands they have created to backup and restore

benlye, 2023-01-31:
I'm not really sure what the ask is here.  You can already easily run GCode commands from the plugin prior to flashing new firmware.

closed 2023-02-20 by github-actions[bot]

cnlson, 2023-03-07:
> I'm not really sure what the ask is here. You can already easily run GCode commands from the plugin prior to flashing new firmware.

The ask is to make this easy to do so it can be a part of the routine

benlye, 2023-03-07:
Like I said, the plugin already has ability to automatically run GCode commands both before and after flashing, so if it's a case of running a Gcode command to make the backup, you can already easily do that.