OctoPrint/OctoPrint-FirmwareUpdater #335: FEATURE REQUEST: Enable updater to find and read zip'ed files

puterboy, 2023-01-30:
Prusa at least distributes its firmware in Zip'ed files.
It would save a step if the module could recognize and read Zip files (two steps if you then want to come back and delete the uncompressed version after updating)
This shouldn't be too hard as Zip is a well-known standard with plenty of tools to read.

Thanks

benlye, 2023-01-31:
Can you upload (or point me at) an example of a zip file? Is the file inside always named the same? Is there only one file inside, or are there multiple files? If it's a zip file containing one bin file (or if the firmware bin file always has the same name) then it shouldn't be too hard.

puterboy, 2023-01-31:
It's just a compressed Zip of a folder that contains a single (uncompressed) folder with the same name that in turn contains one .hex firmware file plus a couple of pdfs.

```
$ unzip -l prusa3d_fw_3_12_1_MK3S.zip
Archive:  prusa3d_fw_3_12_1_MK3S.zip
  Length      Date    Time    Name
---------  ---------- -----   ----
        0  2023-01-23 08:21   prusa3d_fw_3_12_1_MK3S/
   935909  2023-01-16 05:53   prusa3d_fw_3_12_1_MK3S/Firmware_upgrade_guide_CS.pdf
   947775  2023-01-16 05:53   prusa3d_fw_3_12_1_MK3S/Firmware_upgrade_guide_DE.pdf
   951528  2023-01-16 05:54   prusa3d_fw_3_12_1_MK3S/Firmware_upgrade_guide_ENG.pdf
   947239  2023-01-16 05:54   prusa3d_fw_3_12_1_MK3S/Firmware_upgrade_guide_ES.pdf
   947675  2023-01-16 05:54   prusa3d_fw_3_12_1_MK3S/Firmware_upgrade_guide_FR.pdf
   946434  2023-01-16 05:54   prusa3d_fw_3_12_1_MK3S/Firmware_upgrade_guide_IT.pdf
  1528209  2023-01-16 05:54   prusa3d_fw_3_12_1_MK3S/Firmware_upgrade_guide_PL.pdf
  1048002  2023-01-23 08:19   prusa3d_fw_3_12_1_MK3S/prusa3d_fw_MK3S_MK3S+_3_12_1_5686.hex
---------                     -------
  8252771                     9 files
```

Here is the download site: https://help.prusa3d.com/downloads

puterboy, 2023-02-01:
Couldn't you just do the equivalent of unzip -l to identify the hex
file and then just unzip & unload that file?

Ben Lye wrote at about 02:33:37 -0800 on Wednesday, February 1, 2023:
 > Hmm, I don't think I'll implement this. 
 > 
 > I can see how it would be slightly more convenient for some users, but uploading a ~7MB file to the Pi just to extract and use a 1MB hex file seems wasteful. 
 > 
 > It's not like it's a big burden for the end user to extract the firmware file out of the ZIP file, and doing that on a system which is (hopefully) running antivirus and anti-malware software, and should have much more power for the decompression, just seems like a better idea.
 > 
 > -- 
 > https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/issues/335#issuecomment-1411834572
 >

benlye, 2023-02-01:
No, that's not how it works when the browser uploads a file. There's really no way to do it on the client side or in the browser.

closed 2023-02-21 by github-actions[bot]