OctoPrint/OctoPrint-FirmwareUpdater #337: Sovol SV06 Support

luccassa, 2023-02-17:
Sovol support and specifically SV06 firmware support would be amazing. the printer has a lot of firmware leveling issues and is constantly being updated. Thanks!

benlye, 2023-02-19:
How is the board normally updated?

closed 2023-03-11 by github-actions[bot]

Dustmuffins, 2023-04-01:
I'd like to throw my interest in for this one too.

benlye, 2023-04-01:
> I'd like to throw my interest in for this one too.

Then maybe you could answer the question I asked - how does this board get updated?

Dustmuffins, 2023-04-06:
[Here's a video detailing the process.](https://www.youtube.com/watch?v=p6l253OJa34)

mekyleobrien91, 2023-04-06:
@benlye - basically, one typically adds a .bin file to the SD card and restarts the board whilst it's plugged in. I don't believe it matters what the file name is.

However, the mainboard doesn't seem to expose the SD card as a drive to a host connected via the USB cable.

benlye, 2023-04-06:
If the board/firmware doesnt expose the SD card then it doesn't seem like there will be much that the plugin can do.

Have you tried the Binary File Transfer protocol?