OctoPrint/OctoPrint-FirmwareUpdater #338: Octoprint unable to detect "marlin-binary-protocol"

Lord-Memester, 2023-02-19:
> **Hardware Setup**
> Please board type and printer make and model.

OctoPi: Raspberry Pi 3b 
Printer: Voxelab Aquila (G32 chip)

> **Describe the problem**
> Please provide a clear and concise description of what the problem is, including details of any error messages you receive (screenshots are helpful).

Octoprint seems unable to detect the presence of `marlin-binary-protocol`.
[image]

> **Log Files**
> Attach the OctoPrint log and the Firmware Updater log Log files, which can be accessed under **Settings** -> **Logging** in OctoPrint, or in `~/.octoprint/logs/` on the host.  Both `octoprint.log` and `~/.octoprint/logs/plugin_firmwareupdater_console.log` should be included.

[octoprint.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/10777091/octoprint.5.log)
[plugin_firmwareupdater_console.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/10777092/plugin_firmwareupdater_console.log)

> **Additional context**
> Add any other context about the problem here.

Yeah, `octoprint.log` is a bit confusing because it retained data over restarts of OctoPrint. But I promise you, it couldn't find the protocol. I installed `heatshrink` followed by `marlin-binary-protocol` as suggested in the documentation, but the plugin still showed it as lacking. Yes, I refreshed the page. I can't imagine the printer would affect what the plugin believes about the state of installation of the protocol, but I'm running `Marlin 2.1.3*`.

###### *Since I uploaded my build of Marlin to GitHub, you should be able to see exactly what I used in my firmware by looking at the repositories listed on my profile page.

benlye, 2023-02-19:
Most common cause of this problem is not installing the Python packages in the OctoPrint virtual environment.

Exactly what commands did you run to install heatshrink and the binary protocol?

Note this section of the documentation...

> NB: If you are running OctoPrint in a VirtualEnv (as recommended) you need to run the appropriate pip commands below inside that environment. For OctoPi users, this is ~/oprint/bin/pip anywhere it says pip or pip3.

Lord-Memester, 2023-02-19:
Well, I navigated to the appropriate folder before executing the commands. Does that count? I tried using `pip` in the folder, but that didn't work. However, `pip3` did. Is that possibly why?

benlye, 2023-02-19:
No, just navigating to the folder doesn't count. You either have to use the pip commands that are in the venv (as per the instructions) or you you have to activate the venv.

Lord-Memester, 2023-02-19:
Yep. Just tried it now. Seems I'm just too used to windows!

Lord-Memester, 2023-02-19:
Although, on a separate note, I've also encountered an issue with the lv something protocol for uploading with SD card. I'll check to see if the issue is the same as was presented here. If it isn't, I'll make a new issue.

Lord-Memester, 2023-02-20:
the LSP protocol (don't remember the full name) doesn't seem to work with my printer. Might be too similar to the Ender 3 v2 for it to work. Additionally, with the correct installation of marlin BFT and enabling it in the firmware, flashing was unsuccessful.

benlye, 2023-02-21:
How is your board normally updated?  I've never seen one so don't know what it needs.

Lord-Memester, 2023-02-21:
Through a firmware.bin file on the SD card.

benlye, 2023-02-21:
If the file is getting written to the SD card then there's not a lot I can do. If the file isn't getting to the SD card we can look at that.  Either way I need logs from unsuccessful attempts to see what's going on.

Lord-Memester, 2023-02-22:
These should contain the information you need. I was very busy today, so I'll get back to you at a later date if you need something I didn't provide here.
[plugin_firmwareupdater_console (2).log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/10808313/plugin_firmwareupdater_console.2.log)

[octoprint (6).log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/10808311/octoprint.6.log)

benlye, 2023-02-23:
There was no attempt at a firmware update in those logs.

We've solved the original issue so I'm going to close this now.  Feel free to open a new issue when you have time to try again and gather the logs.

closed 2023-02-23 by benlye