OctoPrint/OctoPrint-FirmwareUpdater #341: MarlinBFT and probably all methods broken with OctoPrint 1.9.0rc2

The-EG, 2023-03-10:
**Hardware Setup**
Ender 3 Pro with an BTT Octopus v1.1

**Describe the problem**
Flashing is broken when using MarlinBFT method on OctoPrint 1.9.0rc2. After selecting the bin file and pressing 'Flash From File' the progress bar displays "Flashing Firmware" and nothing happens.

The browser console shows a 400 (BAD REQUEST) error for the `/plugin/firmwareupdater/flash` url.

There are no errors in the octoprint.log (attached anyway).

**Log Files**

[octoprint.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/10945114/octoprint.log)

**Additional context**
Issue caused by changes in latest OctoPrint release candidate version.

benlye, 2023-03-10:
Just want to set expectations - I'm travelling for work and probably won't be able to look into this for at least a week.

benlye, 2023-03-10:
Could you try the devel version of the plugin?

The-EG, 2023-03-11:
> Could you try the devel version of the plugin?

Same result with version 1.14.0b2.

benlye, 2023-03-17:
I just got to give this a try.  Interesting results...

My installation already had plugin v1.14.0b2 and after updating to OctoPrint v1.9.0rc2 I had no problems flashing the firmware.  I downgraded my plugin to v1.13.3 and got the 400 error, I then upgraded to v1.14.0b2 and I still get the 400 error.

This is going to be fun.

The-EG, 2023-03-17:
You can reproduce it at least, I'll take it!

I did poke around a bit, it doesn't appear to be even making it to the `/flash` route.

benlye, 2023-03-17:
I think I have the fix for this - please test v1.14.0b3 from the plugin's Development branch.

benlye, 2023-03-17:
The solution was actually provided in #323 - all I had to do was merge the PR.  Thanks, @arekm.

The-EG, 2023-03-17:
Great, I'll take a look this evening.

The-EG, 2023-03-18:
Confirmed, 1.14.0b3 resolves this for me on 1.9.0rc2.

closed 2023-03-19 by benlye