OctoPrint/OctoPrint-FirmwareUpdater #342: Avrdude flashing method not functioning

guaddy, 2023-03-16:
**Hardware Setup**
Prusa MK3S+ with Einsy RAMBo

**Describe the problem**
Flashing is not functional using the avrdude method. After selecting the appropriate .hex file, and selecting "Flash from File", the progress bar changes but nothing happens.

**Log Files**
[octoprint.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/10985834/octoprint.log)

**Additional context**
 This recently changed after an update to Octopi 1.0.0 and OctoPrint 1.9.2rc2

benlye, 2023-03-17:
I think I have the fix for this - please test v1.14.0b3 from the plugin's Development branch.

guaddy, 2023-03-18:
It looks like 1.14.0b3 fixed it, flashing went as expected. Thank you for the help!

blt3dp, 2023-03-18:
I'm using updated octopi and 1.14.0b3 and I'm not able to get it to flash.  Logs show timeout communicating with programmer.  What settings did you use to get it to work?

counttec, 2023-03-19:
I am having a similar issue flashing my i3 mk3s+, but Avrdude is returning an error code. "Avrdude returned code 127"

My log file:
[octoprint.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/11010463/octoprint.log)

benlye, 2023-03-19:
> I am having a similar issue flashing my i3 mk3s+, but Avrdude is returning an error code. "Avrdude returned code 127" Already attempted to move to dev build v1.14.0b3 to no avail.
> 
> My log file: [octoprint.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/11010463/octoprint.log)

This problem is resolved and you have a different issue.  Please open a new issue and provide the firmwareupdater log file (in addition to the octoprint log file.  Most likely you are using the wrong settings for Avrdude, which is not something I can really help you with as I don't have or know your printer.

closed 2023-03-19 by benlye