OctoPrint/OctoPrint-FirmwareUpdater #343: Avrdude firmware flash fails and returns code 127

counttec, 2023-03-19:
ATmega2560 board, Prusa i3 Mk3s+ printer

**Describe the problem**
While attempting to flash the latest stable firmware from Prusa I am receiving an error code 127.
This behavior persisted when I updated Firmware Updater to the dev build.
The settings I am using have not changed since the last successful flash.

**Log Files**
[octoprint.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/11011270/octoprint.log)
[plugin_firmwareupdater_console.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/11011271/plugin_firmwareupdater_console.log)

benlye, 2023-03-19:
Log says this:
```
2023-03-19 01:18:27,413 > /usr/bin/avrdude: error while loading shared libraries: libftdi.so.1: cannot open shared object file: No such file or directory
```

So it looks like something is wrong with your avrdude installation - at least one shared library is missing and needs to be reinstalled.

counttec, 2023-03-21:
Was able to reinstall libftdi1, however I first had to delete '/var/lib/dpkg/info/libftdi1:armhf.triggers' due to an error during reinstall of the SO.
 `Too-long line or missing newline in '/var/lib/dpkg/info/libftdi1:armhf.triggers`

After fixing the library, and updating Firmware Updater (coincidentally probably), I was able to flash my printer without issue

closed 2023-03-21 by counttec