OctoPrint/OctoPrint-FirmwareUpdater #345: Eryone Thinker Series 2560

sdh1991, 2023-05-05:
**Hardware Setup**
Eryone Thinker S V2, SE, 2560 board 

**Describe the problem**
Any chance getting support for this board?

**Log Files**
Attach the OctoPrint log and the Firmware Updater log Log files, which can be accessed under **Settings** -> **Logging** in OctoPrint, or in `~/.octoprint/logs/` on the host.  Both `octoprint.log` and `~/.octoprint/logs/plugin_firmwareupdater_console.log` should be included.

**Additional context**
Add any other context about the problem here.

benlye, 2023-05-05:
It has an Atmega 2560 so you should be able to update it using the [avrdude method](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/blob/master/doc/avrdude.md).  Have you tried?  Do you have any other process for updating it?  If so, the logs might show the command being used, which would make it easier to configure the plugin.

sdh1991, 2023-05-06:
Im trying but what do I put as the path to avrdude? Sorry im kind of new to this...

benlye, 2023-05-21:
> Im trying but what do I put as the path to avrdude? Sorry im kind of new to this...

Not trying to be awkward, but it will be wherever it got installed to on your system, which will depend on how you installed it.  

If you followed the [instructions](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/blob/master/doc/avrdude.md#raspberry-pi), it's probably `/usr/bin/avrdude`.  You can check by running the command `which avrdude` on your Pi.

closed 2023-05-26 by github-actions[bot]