OctoPrint/OctoPrint-FirmwareUpdater #347: Board reset failed on Marlin 2.1.2.1

bmos, 2023-06-10:
**Hardware Setup**
Creality Ender-5 Pro / Creality V4.2.7

**Describe the problem**
I am able to upload a firmware file to the plugin and it is being copied to the SD card (I can see it on the card afterwards) and I can manually call M997 from the terminal to reset the printer successfully. HOWEVER, when I try to flash via this plugin it gives me the error "Board reset failed."

**Log Files**
plugin_firmwareupdater_console.log is empty
octoprint.log:
```2023-06-10 12:28:28,335 - octoprint.plugins.firmwareupdater - INFO - Firmware update profile index: 0
2023-06-10 12:28:28,345 - octoprint.plugins.firmwareupdater - INFO - Firmware update profile name: Default
2023-06-10 12:28:28,346 - octoprint.plugins.firmwareupdater - INFO - Printer port: /dev/ttyUSB0
2023-06-10 12:28:28,354 - octoprint.plugins.firmwareupdater - INFO - Flash method: marlinbft
2023-06-10 12:28:28,359 - octoprint.plugins.firmwareupdater - INFO - Python package 'marlin-binary-protocol' is installed: True
2023-06-10 12:28:28,362 - octoprint.plugins.firmwareupdater - INFO - Marlin BINARY_FILE_TRANSFER capability is enabled: True
2023-06-10 12:28:28,382 - octoprint.plugins.firmwareupdater - INFO - Firmware update started
2023-06-10 12:28:28,383 - octoprint.plugins.firmwareupdater - INFO - Disconnecting from printer
2023-06-10 12:28:28,498 - octoprint.plugins.firmwareupdater - INFO - Current Baud: 115200
2023-06-10 12:28:28,498 - octoprint.plugins.firmwareupdater - INFO - Initializing Marlin BFT protocol
2023-06-10 12:28:28,500 - octoprint.plugins.firmwareupdater - INFO - pySerial Version: 3.5
2023-06-10 12:28:28,561 - octoprint.plugins.firmwareupdater - INFO - Connecting to printer at '/dev/ttyUSB0' using Marlin BFT protocol
2023-06-10 12:28:28,563 - octoprint.plugins.firmwareupdater - INFO - Connecting: Switching Marlin to binary protocol...
2023-06-10 12:28:28,645 - octoprint.plugins.firmwareupdater - INFO - Connection synced [0], binary protocol version 0.1.0, 96 byte payload buffer
2023-06-10 12:28:28,645 - octoprint.plugins.firmwareupdater - INFO - Transfering file to printer using Marlin BFT '/tmp/tmpfhxkp_kl' -> /firmware.bin
2023-06-10 12:28:51,599 - octoprint.plugins.firmwareupdater - INFO - Binary file transfer complete
2023-06-10 12:28:52,617 - octoprint.plugins.firmwareupdater - INFO - Firmware update reset: attempting to reset the board
2023-06-10 12:28:52,624 - octoprint.plugins.firmwareupdater - INFO - Resetting printer at '/dev/ttyUSB0'
2023-06-10 12:28:52,642 - octoprint.plugins.firmwareupdater - INFO - Sending out-of-band reset command to '/dev/ttyUSB0'
2023-06-10 12:28:52,652 - octoprint.plugins.firmwareupdater - INFO - Waiting 10s for printer at '/dev/ttyUSB0' to reset```

**Additional context**
Add any other context about the problem here.

benlye, 2023-06-10:
Does either reset option cause the printer to do anything at all? I assume it's not just failing to detect the reset, the printer isn't even resetting?

Can you post the log from the alternative reset attempt?

bmos, 2023-06-10:
After removing all the files on the SD card and testing again on alternate, it worked perfectly. Not sure if it was the bin files of previous attempts or something else, but thanks for your reply.

closed 2023-06-10 by bmos