OctoPrint/OctoPrint-FirmwareUpdater #350: Flashing failed - Cannot enter bootloader, check your `BOOT0/Reset` pins settings and try again

jkmathew, 2023-07-23:
**Hardware Setup**
Ender 3 / Creatlity 4.2.2

**Describe the problem**
I tried [this guide](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/blob/master/doc/stm32flash.md) to flash the firmware, But the flashing process is failing with error
```
Flashing failed 
Cannot enter bootloader, check your BOOT0/Reset pins settings and try again
```
How can I update firmware on Creality 4.2.2 board with this plugin?

**Log Files**
[octoprint.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/12140146/octoprint.log)
[plugin_firmwareupdater_console.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/12140147/plugin_firmwareupdater_console.log)

benlye, 2023-07-24:
Not sure why you picked that method - Creality boards use avrdude.
https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/blob/master/doc/avrdude.md

jkmathew, 2023-07-24:
I'm new to 3d printing and firmware programming, I saw `STM32F103RE_creality` when building Marlin. So I thought I should use `STM32Flash`. Let me try avrdude,

jkmathew, 2023-07-24:
Avrdude also failed 😢 
it shows error message
```

Flashing failed
Avrdude says: 'not in sync: resp=0x20
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x54
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x3a
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x32
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x33
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x2e
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x35
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x35
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x20
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x2f

avrdude done. Thank you.'
````

[plugin_firmwareupdater_console.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/12152423/plugin_firmwareupdater_console.log)

benlye, 2023-07-24:
Oops, I take it back, it's not avrdude for the 4.2.2, it's either the lpc176x or marlinBFT method.

https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/blob/master/doc/lpc176x.md
https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/blob/master/doc/marlinbft.md

Both are a bit of a pain to set up, but work equally well once you get there.  The dependencies for MarlinBFT can be a particluar pain for Python 3.9 (see #349).

jkmathew, 2023-07-25:
Hmm...
I think both these won't work in my case, The reason I tried firmware updater instead of flashing with conventional sdcard reboot is because my printer going into a boot loop when I boot the printer with sd card inserted.

 I tried [lpc176x.md](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/blob/master/doc/lpc176x.md), but I could not get the sdcard mounted on the pi(I think). When I test `Path to firmware folder` It gives the error `The path is not writeable`

And for MarlinBFT, I have to flash Marlin once, I am not able to do this because of the issue I mentioned earlier.

jkmathew, 2023-07-26:
Thank you for the reply. Closing this issue since we cannot go forward more, unfortunately.

As a last note, I am formating the sd card from a MacBook, I suspect this case the problem of boot loop (not sure here as well). I will try to format it from the pi and flash.

closed 2023-07-26 by jkmathew

benlye, 2023-07-26:
You might like to try this program to help clean up the SD card before you remove it from your Mac:
https://www.javawa.nl/cleaneject_en.html