OctoPrint/OctoPrint-FirmwareUpdater #351: {disableverify} does not get replaced by space or -v if it is the last thing on the command line

cdsmith69, 2023-07-28:
The {disableverify} parameter should be replaced by either -v or by a space, depending on the check box for "Disable write verification."  But if it is the last thing on the command line the text {disableverify} is left unchanged in the command line that is called.

I believe this may be because of the trailing spaces in the replace commands in bossac.py:

    if bossac_disableverify:
        bossac_command = bossac_command.replace(" {disableverify} ", " ")
    else:
        bossac_command = bossac_command.replace(" {disableverify} ", " -v ")

If {disableverify} is the last thing on the command line the trailing space does not exist and the replace doesn't happen.

The context here is that I have a device that requires a more complicated flashing process than just one call to bossac, so I thought I would just put a shell script on the Raspberry Pi called proflash.sh that does the multiple steps I need, and run that script with the command line  /home/pi/proflash.sh {bossac} {port} {firmware} {disableverify} instead of the usual bossac command line.  But my script was getting "{disableverify}" instead of a space or -v.  Flipping the order to /home/pi/proflash.sh {bossac} {port} {disableverify} {firmware} fixes the issue and the script gets the proper space or -v in place of {disableverify}.

benlye, 2023-08-03:
Good catch, I'll take care of it.

bhuliu, 2024-06-15:
Good catch！！！
——————————————————
[Steppermotor](https://www.steppermotor.fr) provides a variety of different types of gear reducers including planetary reducers, parallel shaft reducers, and worm reducers.

benlye, 2024-06-22:
Resolved in 1.14.1

closed 2024-06-22 by benlye