OctoPrint/OctoPrint-FirmwareUpdater #354: Ender 3 v3 SE support

KaiStarkk, 2023-12-24:
Is this printer currently supported?:
Creality Ender 3 v3 SE -- GD32F303RET6 ARM board

benlye, 2024-01-02:
How do you currently update the board?

KaiStarkk, 2024-01-03:
This one is updated with MicroSD card for the screen (load firmware onto SD card, insert into printer while powered off, then start) -- then with SD card for the main board in the same way.

benlye, 2024-01-03:
Then you should be able to use the MarlinBFT method. Setting it up is a pain due to the Python dependencies. There is another open issue that should explain the workarounds - I need to get around to updating the docs.

KaiStarkk, 2024-01-05:
Managed to install heatpack and marlin-binary-protocol without issue, but I might be out of luck. The reason for asking about support for this printer is that the SD slot is currently broken, seems to be a common issue.

It appears I would need to first install firmware with BINARY_FILE_TRANSFER:1 set, which I can't do without the slot being fixed. I assume the other methods like avrdude aren't applicable or would also need that flag set?

benlye, 2024-01-05:
You might be able to use the lpc176x method. Your current firmware will have to expose the SD card to the host though. If it doesn't, you are in the same spot with needing to flash a new firmware to enable the required options.