OctoPrint/OctoPrint-FirmwareUpdater #356: Support for BTT SKR Mini E3 v3.0

cbho, 2024-01-19:
Hi,
In a now closed
[topic
](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/issues/326) it is mentioned, that support for the BTT SKR Mini E3 v3.0 might be added at some point.

Is there any news about that?

Or is it just me, that haven't figured out the right settings (on a Ender 5 pro)?

Regards,
Chr.

benlye, 2024-01-19:
How do you update the board normally?

cbho, 2024-01-19:
At present by using the SD-card.
Have tried different combos of settings, but also a bit cautious due to the comment about possibly damaging the board...

benlye, 2024-01-19:
If you update using the SD card then the Marlin BFT method should work. I can't imagine how it could damage the board.

It's a bit of a pain to set up due to some dependencies but there are a few pointers for doing it.

Take a look at #349, especially https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/pull/349#issuecomment-1883573730

I need to do some doc updates.

cbho, 2024-01-19:
I'll give it a shot within the next days and report back.
Thanks for the quick reply :-)

cbho, 2024-02-03:
Hi again,
sorry for the late reply.
Tried to follow the instuctions, but can't install _Heatshrink_ :-(

According to the instructions:
```
pi@octopi:~ $ ~/oprint/bin/pip install https://github.com/p3p/pyheatshrink/releases/download/0.3.3/pyheatshrink-pip.zip

Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/p3p/pyheatshrink/releases/download/0.3.3/pyheatshrink-pip.zip
  Downloading https://github.com/p3p/pyheatshrink/releases/download/0.3.3/pyheatshrink-pip.zip (2.5 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 2.5/2.5 MB 2.1 MB/s eta 0:00:00
  Preparing metadata (setup.py) ... done
Building wheels for collected packages: Heatshrink
  Building wheel for Heatshrink (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [140 lines of output]
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build/lib.linux-armv7l-3.9
      creating build/lib.linux-armv7l-3.9/heatshrink
      copying heatshrink/streams.py -> build/lib.linux-armv7l-3.9/heatshrink
      copying heatshrink/__init__.py -> build/lib.linux-armv7l-3.9/heatshrink
      running build_ext
      building 'heatshrink.core' extension
      creating build/temp.linux-armv7l-3.9
      creating build/temp.linux-armv7l-3.9/heatshrink
      creating build/temp.linux-armv7l-3.9/heatshrink/_heatshrink
      arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I. -I./heatshrink/_heatshrink -I/home/pi/oprint/include -I/usr/include/python3.9 -c heatshrink/_heatshrink/heatshrink_decoder.c -o build/temp.linux-armv7l-3.9/heatshrink/_heatshrink/heatshrink_decoder.o -std=c99
      arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I. -I./heatshrink/_heatshrink -I/home/pi/oprint/include -I/usr/include/python3.9 -c heatshrink/_heatshrink/heatshrink_encoder.c -o build/temp.linux-armv7l-3.9/heatshrink/_heatshrink/heatshrink_encoder.o -std=c99
      arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I. -I./heatshrink/_heatshrink -I/home/pi/oprint/include -I/usr/include/python3.9 -c heatshrink/core.c -o build/temp.linux-armv7l-3.9/heatshrink/core.o -std=c99
      heatshrink/core.c: In function ‘__pyx_pf_10heatshrink_4core_7Encoder_7fill’:
      heatshrink/core.c:5226:38: warning: comparison of integer expressions of different signedness: ‘size_t’ {aka ‘unsigned int’} and ‘Py_ssize_t’ {aka ‘int’} [-Wsign-compare]
       5226 |     __pyx_t_5 = ((__pyx_v_total_sunk < __pyx_t_8) != 0);
            |                                      ^
      heatshrink/core.c: In function ‘__Pyx_modinit_type_init_code’:
      heatshrink/core.c:7929:39: error: ‘PyTypeObject’ {aka ‘struct _typeobject’} has no member named ‘tp_print’
       7929 |   __pyx_type_10heatshrink_4core_Writer.tp_print = 0;
            |                                       ^
      heatshrink/core.c:7937:39: error: ‘PyTypeObject’ {aka ‘struct _typeobject’} has no member named ‘tp_print’
       7937 |   __pyx_type_10heatshrink_4core_Reader.tp_print = 0;
            |                                       ^
      heatshrink/core.c:7945:59: error: ‘PyTypeObject’ {aka ‘struct _typeobject’} has no member named ‘tp_print’
       7945 |   __pyx_type_10heatshrink_4core___pyx_scope_struct___drain.tp_print = 0;
            |                                                           ^
      heatshrink/core.c: In function ‘__Pyx_ParseOptionalKeywords’:
      heatshrink/core.c:8621:21: warning: ‘_PyUnicode_get_wstr_length’ is deprecated [-Wdeprecated-declarations]
       8621 |                     (PyUnicode_GET_SIZE(**name) != PyUnicode_GET_SIZE(key)) ? 1 :
            |                     ^
      In file included from /usr/include/python3.9/unicodeobject.h:1026,
                       from /usr/include/python3.9/Python.h:97,
                       from heatshrink/core.c:4:
      /usr/include/python3.9/cpython/unicodeobject.h:446:26: note: declared here
        446 | static inline Py_ssize_t _PyUnicode_get_wstr_length(PyObject *op) {
            |                          ^~~~~~~~~~~~~~~~~~~~~~~~~~
      heatshrink/core.c:8621:21: warning: ‘PyUnicode_AsUnicode’ is deprecated [-Wdeprecated-declarations]
       8621 |                     (PyUnicode_GET_SIZE(**name) != PyUnicode_GET_SIZE(key)) ? 1 :
            |                     ^
      In file included from /usr/include/python3.9/unicodeobject.h:1026,
                       from /usr/include/python3.9/Python.h:97,
                       from heatshrink/core.c:4:
[... cut ...]
I've tried flashing a new version of the OctoPi image, restoring my instance from a backup and tried again (system info attached), but the result stays the same (doesn't matter if I add sudo or not).

Unfortunately, I've got absolutely no knowledge about Python, so it's difficult for me to figure out what I should do to solve the problem...

Hopefully, it makes more sense to you?

Regards,
Chr.
[octoprint-systeminfo-20240203202243.zip](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/14152645/octoprint-systeminfo-20240203202243.zip)

cbho, 2024-02-05:
Found [this post](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/issues/321#issuecomment-1435733820) and did as mentioned - now prerequisites are fulfilled, at least...

benlye, 2024-02-06:
> And now: firmware update possible...
> 
> (After I figured out, that the two last paragraphs on the [Flashing an LPC176x or STM32 board using Binary File Transfer](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/blob/devel/doc/marlinbft.md#flashing-an-lpc176x-or-stm32-board-using-binary-file-transfer) was not only related to Creality Ender 3 V2, but are general...)

Great news!

To be fair, I thought that only the Creality boards were fussy about old .bin files, but I have moved that section up to apply generally as it won't hurt other boards.

closed 2024-02-06 by benlye