OctoPrint/OctoPrint-FirmwareUpdater #358: Support needed for Prusa Mk4

dcorbin, 2024-06-10:
**Hardware Setup**
Prusa xBuddy on the Prusa Mk4

**Describe the problem**
Prusa firmware files are published as .bbf, which are clearly not supported (I can't browse for one).  Beyond that, I do not know if the upgrade process is different on the Mk4.

**Log Files**
N/A

**Additional context**
Add any other context about the problem here.

jneilliii, 2024-06-10:
I suspect the MK4 is running 32bit main board and probably doesn't have marlin binary file transfer as an option. How would you normally update your board, put the bbf file on the sd card and power cycle the printer?

benlye, 2024-06-12:
Can your OctoPrint host mount the USB drive? (Take a look at the [LPC176x method](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/blob/master/doc/lpc176x.md)) 

If not, and you can't get Marlin Binary File Transfer enabled, there's not really much the plugin can do as it needs some way to write to the board's storage.

dcorbin, 2024-06-22:
I'll have to get back on the mounting the USB drive.

I'm told that Prusa Connect can push the firmware and trigger the update.  Seems like it ought to be possible from Octoprint, though perhaps that requires an entirely different API and should be a different plugin.

closed 2024-06-23 by dcorbin