OctoPrint/OctoPrint-FirmwareUpdater #359: Board reset fails after successful BFT

ali1234, 2024-06-12:
**Hardware Setup**
Ender 3 with BigTreeTech SKR E3 Mini DIP V1.0
OctoPrint version : 1.10.1
OctoPi version    : 0.18.0
Marlin version: 2.1.2.3

**Describe the problem**
After successful Marlin BFT, the board crashes instead of resetting.

**Log Files**
[plugin_firmwareupdater_console.log](https://github.com/user-attachments/files/15798120/plugin_firmwareupdater_console.log)
[octoprint.log](https://github.com/user-attachments/files/15798112/octoprint.log)

**Additional context**
The firmware file is successfully transferred to the SD card. Only resetting does not work, with or without "alternate reset" selected. The printer's display says "Resetting..." but then it does not reset. It just stays like that forever and does not respond to controls.

After manually power cycling the printer, it will correctly install the firmware from the SD card, but then Octoprint cannot reconnect to it. It stays stuck at "State: Opening serial connection" forever. This is not a Linux kernel issue: restarting the Octoprint service with `sudo service octoprint restart` is enough to make it connect properly to the printer.

benlye, 2024-06-12:
I'm wondering if some other plugin is interfering with the port. Can you disable your other OctoPrint plugins and then try a firmware update? PortLister is probably the one I would start with.

Post the OctoPrint log.

ali1234, 2024-06-12:
There is no change with all other plugins disabled.

[octoprint.log](https://github.com/user-attachments/files/15805308/octoprint.log)

benlye, 2024-06-12:
OK. Was worth a try. 

What happens if you issue `M997` in the OctoPrint terminal? The board _should_ reset...

ali1234, 2024-06-12:
The same thing: the printer freezes with "Resetting..." on the display.

benlye, 2024-06-12:
Then it seems like there's something else going on with your setup. If you can solve the problem with `M997` outside of the updater plugin, the plugin should begin to work.

My assumption is that something else on your Pi is blocking the port and you'll have to do some digging to try to find out what.

ali1234, 2024-06-12:
So failing to reset seems to be a bug in Marlin, and getting stuck at "State: Opening serial connection" seems to be an Octoprint bug. I worked out that just clicking "disconnect" and then "reconnect" is enough, after power cycling the printer to un-crash it, at least when portlister is disabled. The port is not blocked, Octoprint just gets stuck because the printer doesn't respond to any commands when it is crashed. The Pi is running OctoPi and the only other software on it is webcamd.

closed 2024-06-12 by ali1234

ali1234, 2024-06-14:
It turned out to be the SD card - ironically the one supplied with the E3 Mini. The SD card supplied with the Ender 3 works fine.