OctoPrint/OctoPrint-FirmwareUpdater #361: Flashing Prusa MMU2 Firmware does not start

novski, 2024-06-25:
**Hardware Setup**
Octoprint 1.10.2, Python 3.9.2, Octopi 1.0.0cam
Two USB Cables connected. One runing to the Prusa MK3S+ and one to the MMU2.

**Describe the problem**
I updated my MK3S+ firmware sucessfully through FirmewareUpdater by selecting the /dev/ttyACM0 Serial interface. Then selected /dev/ttyACM1 and tried to update the Firmware of the MMU2. But that times out.

[image]
[image]
logs:
```
2024-06-22 18:17:39,984 > 
2024-06-25 11:47:42,075 
2024-06-25 11:47:42,080 /usr/bin/avrdude -v -q -p m2560 -c wiring -P /dev/ttyACM1 -D -U flash:w:/tmp/tmp43p1yt5s:i
2024-06-25 11:47:42,713 > 
2024-06-25 11:47:42,713 > avrdude: Version 6.3-20171130
2024-06-25 11:47:42,713 >          Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
2024-06-25 11:47:42,714 >          Copyright (c) 2007-2014 Joerg Wunsch
2024-06-25 11:47:42,714 > 
2024-06-25 11:47:42,714 >          System wide configuration file is "/etc/avrdude.conf"
2024-06-25 11:47:42,714 >          User configuration file is "/home/novski/.avrduderc"
2024-06-25 11:47:42,714 >          User configuration file does not exist or is not a regular file, skipping
2024-06-25 11:47:42,714 > 
2024-06-25 11:47:42,714 >          Using Port                    : /dev/ttyACM1
2024-06-25 11:47:42,715 >          Using Programmer              : wiring
2024-06-25 11:47:42,715 > 
2024-06-25 11:47:48,063 > avrdude: stk500v2_ReceiveMessage(): timeout
2024-06-25 11:47:48,064 > 
2024-06-25 11:47:53,067 > avrdude: stk500v2_ReceiveMessage(): timeout
2024-06-25 11:47:53,068 > 
2024-06-25 11:47:58,071 > avrdude: stk500v2_ReceiveMessage(): timeout
2024-06-25 11:47:58,072 > 
2024-06-25 11:48:03,077 > avrdude: stk500v2_ReceiveMessage(): timeout
2024-06-25 11:48:03,077 > 
2024-06-25 11:48:08,081 > avrdude: stk500v2_ReceiveMessage(): timeout
2024-06-25 11:48:08,097 > 
2024-06-25 11:48:13,087 > avrdude: stk500v2_ReceiveMessage(): timeout
2024-06-25 11:48:13,094 > avrdude: stk500v2_getsync(): timeout communicating with programmer
2024-06-25 11:48:13,095 > 
2024-06-25 11:48:13,095 > avrdude done.  Thank you.
2024-06-25 11:48:13,096 > 
2024-06-25 11:48:13,096 > 
```
Here the same part when Flashing the Prusa MK3S+ Printer it self:
```
2024-06-22 18:12:26,622 
2024-06-22 18:12:26,627 /usr/bin/avrdude -v -q -p m2560 -c wiring -P /dev/ttyACM0 -D -U flash:w:/tmp/tmpq4rsh08w:i
2024-06-22 18:12:27,624 > 
2024-06-22 18:12:27,625 > avrdude: Version 6.3-20171130
2024-06-22 18:12:27,625 >          Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
2024-06-22 18:12:27,625 >          Copyright (c) 2007-2014 Joerg Wunsch
2024-06-22 18:12:27,626 > 
2024-06-22 18:12:27,626 >          System wide configuration file is "/etc/avrdude.conf"
2024-06-22 18:12:27,626 >          User configuration file is "/home/novski/.avrduderc"
2024-06-22 18:12:27,627 >          User configuration file does not exist or is not a regular file, skipping
2024-06-22 18:12:27,627 > 
2024-06-22 18:12:27,627 >          Using Port                    : /dev/ttyACM0
2024-06-22 18:12:27,628 >          Using Programmer              : wiring
2024-06-22 18:12:27,628 >          AVR Part                      : ATmega2560
2024-06-22 18:12:27,628 >          Chip Erase delay              : 9000 us
2024-06-22 18:12:27,629 >          PAGEL                         : PD7
2024-06-22 18:12:27,629 >          BS2                           : PA0
2024-06-22 18:12:27,629 >          RESET disposition             : dedicated
2024-06-22 18:12:27,629 >          RETRY pulse                   : SCK
2024-06-22 18:12:27,630 >          serial program mode           : yes
2024-06-22 18:12:27,630 >          parallel program mode         : yes
2024-06-22 18:12:27,630 >          Timeout                       : 200
2024-06-22 18:12:27,631 >          StabDelay                     : 100
2024-06-22 18:12:27,632 >          CmdexeDelay                   : 25
2024-06-22 18:12:27,632 >          SyncLoops                     : 32
2024-06-22 18:12:27,632 >          ByteDelay                     : 0
2024-06-22 18:12:27,633 >          PollIndex                     : 3
2024-06-22 18:12:27,633 >          PollValue                     : 0x53
2024-06-22 18:12:27,633 >          Memory Detail                 :
2024-06-22 18:12:27,634 > 
2024-06-22 18:12:27,634 >                                   Block Poll               Page                       Polled
2024-06-22 18:12:27,634 >            Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
2024-06-22 18:12:27,635 >            ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
2024-06-22 18:12:27,635 >            eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
2024-06-22 18:12:27,635 >            flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
2024-06-22 18:12:27,635 >            lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
2024-06-22 18:12:27,636 >            hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
2024-06-22 18:12:27,636 >            efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
2024-06-22 18:12:27,636 >            lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
2024-06-22 18:12:27,637 >            calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
2024-06-22 18:12:27,637 >            signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
2024-06-22 18:12:27,637 > 
2024-06-22 18:12:27,638 >          Programmer Type : Wiring
2024-06-22 18:12:27,638 >          Description     : Wiring
2024-06-22 18:12:27,638 >          Programmer Model: AVRISP
2024-06-22 18:12:27,638 >          Hardware Version: 15
2024-06-22 18:12:27,639 >          Firmware Version Master : 2.10
2024-06-22 18:12:27,639 >          Vtarget         : 0.0 V
2024-06-22 18:12:27,639 >          SCK period      : 0.1 us
2024-06-22 18:12:27,640 > 
2024-06-22 18:12:27,640 > avrdude: AVR device initialized and ready to accept instructions
2024-06-22 18:12:27,640 > avrdude: Device signature = 0x1e9801 (probably m2560)
2024-06-22 18:12:27,640 > avrdude: reading input file "/tmp/tmpq4rsh08w"
2024-06-22 18:12:27,641 > avrdude: writing flash (246430 bytes):
2024-06-22 18:12:27,641 > 
2024-06-22 18:13:07,130 > avrdude: 246430 bytes of flash written
2024-06-22 18:13:07,131 > avrdude: verifying flash memory against /tmp/tmpq4rsh08w:
2024-06-22 18:13:07,131 > avrdude: load data flash data from input file /tmp/tmpq4rsh08w:
2024-06-22 18:13:07,131 > avrdude: input file /tmp/tmpq4rsh08w contains 246430 bytes
2024-06-22 18:13:07,132 > avrdude: reading on-chip flash data:
2024-06-22 18:13:07,132 > 
2024-06-22 18:13:38,730 > avrdude: verifying ...
2024-06-22 18:13:38,730 > avrdude: 246430 bytes of flash verified
2024-06-22 18:13:38,730 > 
2024-06-22 18:13:38,730 > avrdude done.  Thank you.
2024-06-22 18:13:38,731 > 
2024-06-22 18:13:38,731 > 
2024-06-22 18:13:38,735 Flashing successful.
2024-06-22 18:17:09,033 
```

Is the MMU not suported?

benlye, 2024-06-25:
It is supported, but you have to create a separate profile with different settings for the MMU, which are here:
https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/blob/master/doc/avrdude.md#plugin-settings-for-prusa-mmu

novski, 2024-06-25:
I bet I will forget that kind of obstacle until the next time... 
A prepared list of profiles with a dropdown would help.

closed 2024-06-25 by benlye