OctoPrint/OctoPrint-FirmwareUpdater #362: I can't use the Firmware Updater plugin

luisr320, 2024-07-08:
**Hardware Setup**
I have a Prusa iMK3S, with a Raspberry Pi 4B, which works fine. I can upload the Gcode from Prusa Slicer by Wifi to the Octoprint Server and the printer works fine.

**Describe the problem**

- I can upload the Gcode from Prusa Slicer by Wifi to the Octoprint Server and the printer works fine.
- I'm using the TX/TX connection between the Raspberry Pi and the Prusa board. On the RPi the serial port is identified as ttyAMA0.
- If I connect to the Rpi with putty, I can send a echo message to this port and I can see it displayed in the Octoprint console.
- I have the Baud rate set to Default, but I have tried all the individual baud rates set manually.
- I have uninstalled and reinstalled avrdude, and made the updates and upgrades.
- I'm running Octoprint 1.10.2 and the Firmware updater is version 1.14.1.
- I have downloaded the latest firmware for this printer directly from the Prusa site, with the name MK3_FW_3.14.0_MULTILANG.hex.
- The command line sent by the plugin is this: /usr/bin/avrdude -v -q -p m2560 -c wiring -P /dev/ttyAMA0 -D -U flash:w:/tmp/tmpt96e0c2x:i
- I tried disconnecting the octoprint server before trying to send the firmware, but it doesn't work either.

No matter what I do, I just can't flash the new firmware into the printer. I get the message "Starting flash...", it hangs for about 30 seconds and then i get a "Flashing failed; Timeout communicating with the programmer".

Please HELP!.
Thank you

**Log Files**

**Firmware updater log:**

2024-07-08 14:12:29,295 /usr/bin/avrdude -v -q -p m2560 -c wiring -P /dev/ttyAMA0 -D -U flash:w:/tmp/tmpnhg61777:i
2024-07-08 14:12:29,864 > 
2024-07-08 14:12:29,865 > avrdude: Version 6.3-20171130
2024-07-08 14:12:29,865 >          Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
2024-07-08 14:12:29,865 >          Copyright (c) 2007-2014 Joerg Wunsch
2024-07-08 14:12:29,866 > 
2024-07-08 14:12:29,866 >          System wide configuration file is "/etc/avrdude.conf"
2024-07-08 14:12:29,866 >          User configuration file is "/home/pi/.avrduderc"
2024-07-08 14:12:29,866 >          User configuration file does not exist or is not a regular file, skipping
2024-07-08 14:12:29,867 > 
2024-07-08 14:12:29,867 >          Using Port                    : /dev/ttyAMA0
2024-07-08 14:12:29,867 >          Using Programmer              : wiring
2024-07-08 14:12:29,867 > 
2024-07-08 14:12:32,785 > avrdude: stk500v2_ReceiveMessage(): timeout
2024-07-08 14:12:32,786 > 
2024-07-08 14:12:36,794 > avrdude: stk500v2_ReceiveMessage(): timeout
2024-07-08 14:12:36,795 > 
2024-07-08 14:12:40,803 > avrdude: stk500v2_ReceiveMessage(): timeout
2024-07-08 14:12:40,805 > 
2024-07-08 14:12:44,812 > avrdude: stk500v2_ReceiveMessage(): timeout
2024-07-08 14:12:44,813 > 
2024-07-08 14:12:48,820 > avrdude: stk500v2_ReceiveMessage(): timeout
2024-07-08 14:12:48,821 > 
2024-07-08 14:12:52,830 > avrdude: stk500v2_ReceiveMessage(): timeout
2024-07-08 14:12:52,830 > avrdude: stk500v2_getsync(): timeout communicating with programmer
2024-07-08 14:12:52,831 > 
2024-07-08 14:12:52,832 > avrdude done.  Thank you.
2024-07-08 14:12:52,832 > 

**Octopi log:**

2024-07-08 12:48:27,506 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2024-07-08 12:48:27,527 - octoprint.plugins.octoeverywhere - INFO - Server con wss://lon.octoeverywhere.com/octoclientws[1] IS past it's RunFor time 2 days, 22:19:37.241884 and IS past it's time since last user activity 2 days, 22:19:34.540668
2024-07-08 12:48:27,527 - octoprint.plugins.octoeverywhere - INFO - Server con wss://lon.octoeverywhere.com/octoclientws[1] RunFor is complete and will be disconnected.
2024-07-08 12:48:27,528 - octoprint.plugins.octoeverywhere - INFO - Closing all open web stream sockets (0)
2024-07-08 12:48:27,528 - octoprint.plugins.octoeverywhere - INFO - OctoServerCon websocket close start. IsPrimary?:True; wsId:2530712144
2024-07-08 12:48:27,531 - octoprint.server - INFO - --- Log roll over detected ---------------------------------------------------
2024-07-08 12:48:27,531 - octoprint.server - INFO - OctoPrint 1.10.2
2024-07-08 12:48:27,533 - octoprint.plugin.core - INFO - 34 plugin(s) registered with the system:
|  Achievements Plugin (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/achievements
|  Action Command Notification Support (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/action_command_notification
|  Action Command Prompt Support (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/announcements
|  Anonymous Usage Tracking (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/tracking
|  Application Keys Plugin (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/appkeys
|  Backup & Restore (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/backup
[... 110 lines cut ...]
    raise e
  File "/home/pi/oprint/lib/python3.7/site-packages/flask_limiter/extension.py", line 1147, in _check_request_limit
    in_middleware,
  File "/home/pi/oprint/lib/python3.7/site-packages/flask_limiter/extension.py", line 1022, in __filter_limits
    marked_for_limiting,
  File "/home/pi/oprint/lib/python3.7/site-packages/flask_limiter/manager.py", line 102, in resolve_limits
    decorated_limits.extend(self.blueprint_limits(app, blueprint))
  File "/home/pi/oprint/lib/python3.7/site-packages/flask_limiter/manager.py", line 185, in blueprint_limits
    app, blueprint
  File "/home/pi/oprint/lib/python3.7/site-packages/flask_limiter/manager.py", line 250, in _blueprint_exemption_scope
    for k, f in self._blueprint_exemptions.items()
  File "/home/pi/oprint/lib/python3.7/site-packages/flask_limiter/manager.py", line 248, in <setcomp>
    ancestor_exemption = {
RuntimeError: dictionary changed size during iteration
2024-07-08 12:54:04,346 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from https://plugins.octoprint.org/plugins.json
2024-07-08 12:54:37,026 - octoprint.plugins.firmwareupdater - INFO - Firmware update profile index: 0
2024-07-08 12:54:37,036 - octoprint.plugins.firmwareupdater - INFO - Firmware update profile name: Default
2024-07-08 12:54:37,037 - octoprint.plugins.firmwareupdater - INFO - Printer port: /dev/ttyAMA0
2024-07-08 12:54:37,044 - octoprint.plugins.firmwareupdater - INFO - Flash method: avrdude
2024-07-08 12:54:37,067 - octoprint.plugins.firmwareupdater - INFO - Firmware update started
2024-07-08 12:54:37,092 - octoprint.plugins.firmwareupdater - INFO - Running '/usr/bin/avrdude -v -q -p m2560 -c wiring -P /dev/ttyAMA0 -D -U flash:w:/tmp/tmpr2_hqn_w:i' in /usr/bin
2024-07-08 12:55:02,986 - octoprint.plugins.firmwareupdater - ERROR - Flashing failed. Timeout communicating with programmer.

**Additional context**
I keep getting the Udervoltage warning on Octoprint, but it works fine

benlye, 2024-07-08:
avrdude isn't getting a response when it tries to connect to the port you've told it to use. 

I don't have this board but from what others have commented it looks like you're using the right avrdude settings, so the problem is probably in your config / wiring.

There's not really anything I can do to help - you're better off looking for a user group where someone might have a similar setup.

benlye, 2024-07-08:
You might find something useful in https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/issues/307. If you're using the RPi header you're probably going to have to wire up more pins and use `linuxgpio` instead of `wiring`. It seems like the RX/TX ports in that header can't be used to flash the MCU the way you're trying to do it.

luisr320, 2024-07-08:
Thank you. I think that the problem I have is due to the connection between the Prusa board and the Rpi4 board being done trough the RX/TX Uart port. I think that the connection for the firmware updater to work must be done via USB

flo269, 2024-07-08:
Hey, I am experiencing the same symptoms, with a different setup. I have the same versions installed as you, but I am connected via USB. I'm about to head out, but will upload logs etc. later or tomorrow. Just a quick heads up.

MK3s+ and no hardware or octopi changes since the last, flawlessly working, updates.

luisr320, 2024-07-08:
So you are connecting the Rpi4 with the Prusa board using the USB ports?

flo269, 2024-07-08:
Yes, I print with this setup all time and even today between the failed updates attempts. I haven't changed a thing (apart from Filament) in ages.
[image]

luisr320, 2024-07-08:
Ok, so finally i figured it out. I was trying to use the TX/RX ports of both the Rpi4 and the Einsy Board, and even though it works for transferring the gcode from PrusaSlicer to the printer via Wifi, it wont work for the firmware updater. For that, I connected the Einsy Board USB port directly to one of the the Rpi4 USB ports, which created a new tty device named ACM0, and by leaving all the setups options as default, I managed to update the new firmware remotely.  Hurray!!!
Try leaving the baudrate in AUTO, and use the "wiring" connection

flo269, 2024-07-09:
Great to hear that it works for you.
I tried auto baudrate and "wiring" connection is set and used as before. I'm still stuck.

@anyone Anything else I could do that might be helpful to find out what's going on? Thanks! :-)

flo269, 2024-07-09:
I just remembered that I had installed "[Octoprint-PrusaResetMode](https://github.com/jacopotediosi/Octoprint-PrusaResetMode)" a while back. That was what was blocking the connection. I disabled it and then everything worked flawlessly...

Sorry for the noise!!

closed 2024-07-24 by benlye